tttapa/Control-Surface #467: Mapping Feature on Multiple Pentiometers

nickskeba, 2021-05-12:
I've been able to use the mapping on one pentiometer and it works great. Without it, I can't get multiple to work well.
Here is my code which really only works for one pentiometer and not two. 

CCPotentiometer potentiometers[] = { {
  35, {MIDI_CC::Channel_Volume, CHANNEL_1}
},

{
  32, {MIDI_CC::Channel_Volume, CHANNEL_2}
}};

// The filtered value read when potentiometer is at the 0% position
constexpr analog_t minimumValue = 255;
// The filtered value read when potentiometer is at the 100% position
constexpr analog_t maximumValue = 16383 - 255;

// A mapping function to eliminate the dead zones of the potentiometer:
// Some potentiometers don't output a perfect zero signal when you move them to
// the zero position, they will still output a value of 1 or 2, and the same
// goes for the maximum position.
analog_t mappingFunction(analog_t raw) {
    // make sure that the analog value is between the minimum and maximum
    raw = constrain(raw, minimumValue, maximumValue);
    // map the value from [minimumValue, maximumValue] to [0, 16383]
    return map(raw, minimumValue, maximumValue, 0, 16383);
    // Note: 16383 = 2¹⁴ - 1 (the maximum value that can be represented by
    // a 14-bit unsigned number
}

tttapa, 2021-05-13:
It's not clear to me what the issue is. Please post your full code and the complete error message you're getting.

I think your question might be answered by https://github.com/tttapa/Control-Surface/issues/348#issuecomment-738144784.

Zlraul, 2021-07-10:
i think i got the same problems, when i try to use multiple potentiometer using array it affect all CC that on that array.

tttapa, 2021-07-10:
> when i try to use multiple potentiometer using array it affect all CC that on that array.

This is a hardware issue, usually caused by using high-resistance potentiometers, the Arduino's ADC can only handle up to 10kΩ of output impedance, which corresponds to 20kΩ potentiometers. If your potentiometers don't meet this specification, you get cross-talk between ADC channels. You could add some capacitors between the potentiometer wipers and ground to limit the effect.

Another cause could be power or wiring issues.

Zlraul, 2021-07-10:
> This is a hardware issue, usually caused by using high-resistance potentiometers, the Arduino's ADC can only handle up to 10kΩ of output impedance, which corresponds to 20kΩ potentiometers. If your [...]

oh yeah that may be it, i use 100K potentio bc i just have that for a moment, will try some capacitor thanks