tttapa/Control-Surface #487: CC Toggle

Wallirec, 2021-05-26:
Hello! How can I modify this sketch so that CC Toggle will be transmitted?
--------------------------------------------------------------------------------------
#include <Control_Surface.h>

USBMIDI_Interface midi;

// Ignore the name “increment” button, it's simply a button
// class that can detect long and short presses.
IncrementButton btn { 4 }; // pin number of button

void setup() {
  Control_Surface.begin();
  btn.begin();
}

void loop() {
  Control_Surface.loop();
  switch (btn.update()) {
    case IncrementButton::ReleasedShort: // If button is released after short press
      Control_Surface.sendCC(28, 127); break;
    case IncrementButton::IncrementLong: // If button is still pressed after some time
      Control_Surface.sendCC(29, 127); break;
  }
}
-------------------------------------------------------------------------------------------
Now I can only turn on the effects on the connected device, I can't turn it off, because the next time I press it, the CC is transmitted again with a value of 127

Wallirec, 2021-05-29:
Or tell me how to transfer messages by double clicking.

tttapa, 2021-05-29:
> CC Toggle will be transmitted?

Declare a boolean flag, and toggle the state of that flag whenever the button event of interest occurs. Then send the right MIDI message based on the state of the flag.

You can find some inspiration here: https://tttapa.github.io/Control-Surface-doc/Doxygen/d8/dba/Abstract_2MIDIButtonLatched_8hpp_source.html

> Or tell me how to transfer messages by double clicking.

Save the times when the button was pressed/released, if you get two presses less than x ms apart, that means there was a double click.

I don't have time to help with code right now, and these questions are not really Control Surface-specific, so you might find more help on the Arduino forum.

Wallirec, 2021-05-30:
Thanks, all worked well!

closed 2021-05-30 by Wallirec