tttapa/Control-Surface #509: Incomplete reading of a Fatar keyboard

anoat, 2021-06-29:
Hello Peter,

Thanks to your help, the manufacture of my midi controller is progressing well, however I (still) need your help ...

I want to set up a 37 key keyboard with velocity, with my switch and my potentiometer.

For switches and potentiometers I use your library (works perfectly).
For the keyboard I use this code (https://bitbucket.org/ulph/fatar88lux/src/master/teensy_fatar_88.ino
)(works perfectly ).
Both (Your library and the code for the keyboard) work perfectly independently of each other.
However when I merge the two the keyboard reading is "bad", the velocity cannot exceed 34/127.

On the hardware level, I use a teensy4.1, the potentiometers (27), the selectors (3) and the switches (18) are connected to 4067 multiplexers, the keyboard is connected directly to the pins of the Teensy.

How can I make the keyboard read correctly?

Thank you so much ;)

Here is my code:
**Code**

```cpp
// 
#define PRINT_SERIAL_DEBUG 0
#define MIDI_SEMI_PANIC 0 // should not have to :/

#include <errno.h>
#include <Control_Surface.h>

USBMIDI_Interface midi;

const byte adressCC[3] = {112 , 90, 92 };
const byte tlc4[4] = {32 , 33, 34, 35 };
const byte tlc2[2][2] = { {20 , 21 }, {18 , 17 }};
const byte CCsel[2] = { 0 , 2 };
const byte TlcSel[2] = { 1 , 2 };

char tlcCC[3];
int Selec3 ;
constexpr size_t numbuttons = 13;
uint8_t tlcpins[numbuttons] = {5,  24, 25, 47,
                               16, 15, 22, 19,
                               6, 4, 3, 23, 0
                              };
/*
  CONFIGURATION
*/

const unsigned int switch_rest_period_micros = 25; // how long wait for debouncing
const unsigned int scan_period_micros = 5; // how often increment select pin. too often yeilds false readings

/*
  SETTINGS
*/

struct Settings_t {
  float vel_t0; // period for hardest strike
  float vel_t1; // period for softest recognisable
  byte vel_pow; // power law for mapping
  byte vel_min; // lowest velocity value to send
  byte midi_channel; // which
  byte root_key; // which is the lowest key on the keybed

};

Settings_t g_settings = {
  2200,
  500000,
  7,
  0,
[... 323 lines cut ...]
}

unsigned long last_scan_micros = 0;
unsigned long last_send_usb = 0;

void loop() {

  Control_Surface.loop();

  unsigned long now = micros();

  if ((now - last_scan_micros) < scan_period_micros) {
    return;
  }
  last_scan_micros = now;

  for (int i = 0; i < c_numSwitchPairs; i++) {
    int state_idx = g_select + 8 * i;
    FatarKeyState *keyState = &g_KeyStates[state_idx];
    keyState->update(now);
  }
  increment_select();
}

```

tttapa, 2021-06-29:
This might not be that easy. It looks like the original code expects to update every 50 µs, which is very fast. I suspect that it's not possible to read 27 potentiometers this quickly. Most time will probably be spent reading the analog inputs, given that the processor of the Teensy 4.1 should be plenty fast. You could try measuring how long that takes, and you could try increasing the ADC clock rate or the number of averaged samples if it turns out to be a bottleneck.  

There's also a delay in the multiplexer implementation to prevent starting a reading before the changes to the address lines have propagated. This is definitely necessary on fast processors like the T 4.1, but you might get away with a shorter delay:

https://github.com/tttapa/Control-Surface/blob/feee931d44d5f20fb7913b171895316e5819549d/src/AH/Hardware/ExtendedInputOutput/AnalogMultiplex.hpp#L246

You could compare the duration of `Control_Surface.loop()` to the duration of 27×2 analogReads (times two because every time the multiplexer channel is changed, Control Surface does a dummy read):

https://github.com/tttapa/Control-Surface/blob/feee931d44d5f20fb7913b171895316e5819549d/src/AH/Hardware/ExtendedInputOutput/AnalogMultiplex.hpp#L217

anoat, 2021-06-30:
Good evening Peter, thank you for this precious information (if you have any others, I'm a taker;) ).

Good concerning my problem, after a few ... minutes ... hours ... I found a solution!
(I tried a lot of things without success).

In the end I use IntervalTimer to call the functions at specific times.

Regarding the execution time of control surface (with my parameters) it is 667 microseconds, for the keyboard it is 1 microsecond, however I left 50 microseconds in my code (I thought that maybe left more idle to processor).

Do you have any other tip to optimize the execution of your code?

PS: All the tips you gave me worked with a huge gain in speed of calculation (I cannot say precisely because I did not measure it)

```cpp
#define PRINT_SERIAL_DEBUG 0
#define MIDI_SEMI_PANIC 0 // should not have to :/

#include <errno.h>
#include <Control_Surface.h>

IntervalTimer myTimer;
USBMIDI_Interface midi;

const byte adressCC[3] = {112 , 90, 92 };
const byte tlc4[4] = {32 , 33, 34, 35 };
const byte tlc2[2][2] = { {20 , 21 }, {18 , 17 }};
const byte CCsel[2] = { 0 , 2 };
const byte TlcSel[2] = { 1 , 2 };

char tlcCC[3];
int Selec3 ;
constexpr size_t numbuttons = 13;
uint8_t tlcpins[numbuttons] = {5,  24, 25, 47,
                               16, 15, 22, 19,
                               6, 4, 3, 23, 0
                              };
/*
  CONFIGURATION
*/

const unsigned int switch_rest_period_micros = 25; // how long wait for debouncing
unsigned int scan_period_micros = 50; // how often increment select pin. too often yeilds false readings

/*
  SETTINGS
*/

struct Settings_t {
  float vel_t0; // period for hardest strike
  float vel_t1; // period for softest recognisable
  byte vel_pow; // power law for mapping
  byte vel_min; // lowest velocity value to send
  byte midi_channel; // which
  byte root_key; // which is the lowest key on the keybed

};

Settings_t g_settings = {
  2200,
  500000,
  7,
  0,
  0,
  21,
};

enum kb_pins_t {

  // break / make pins
  kb_MK0  = 32,
[... 316 lines cut ...]
}

unsigned long last_scan_micros = 0;
unsigned long first_scan_micros = 0;
unsigned long last_send_usb = 0;
unsigned long new_scan ;

void keys() {

  unsigned long now = micros();
  last_scan_micros = now;

  for (int i = 0; i < c_numSwitchPairs; i++) {
    int state_idx = g_select + 8 * i;
    FatarKeyState *keyState = &g_KeyStates[state_idx];
    keyState->update(now);
  }
  increment_select();
}

void loop() {
  Control_Surface.loop();
}

```

tttapa, 2021-06-30:
Glad to hear you got it to work!

> Regarding the execution time of control surface (with my parameters) it is 667 microseconds

That's a pretty good number, is that with the default settings or after changes to the library? I did a quick test, and a single `analogRead()` on the Teensy 4.1 seems to take about 20 µs, which would give a total run time of at least 27×2×20 µs ≃ 1 ms.

anoat, 2021-07-01:
Pierre,

I have 667 µs with the modifications.
 I deleted :
- the delay in the multiplexer `

```cpp 
Control-Surface/src/AH/Hardware/ExtendedInputOutput/AnalogMultiplex.hpp  
Line 246 in feee931
 delayMicroseconds(5);
 ```

- the double reading of multiplexers
```cpp 
Control-Surface/src/AH/Hardware/ExtendedInputOutput/AnalogMultiplex.hpp
Line 217 in feee931
 ExtIO::analogRead(analogPin); // Discard first reading 
 ```
Is it possible to optimize multiplexer reading by "analogReadFast" or DigitalReadFast "?

I take this opportunity to ask you another question, I use 6-position selector, they allow me to select values ​​(127/6). Currently I put them as potentiometer, it works however when I turn the selcteur I receive a multitude of values ​​between the old position and the new one.
I suspect that this is not the most suitable. What is the best way to implement this component in code?

tttapa, 2021-07-09:
> Is it possible to optimize multiplexer reading by "analogReadFast" or DigitalReadFast "?

Not really, digitalReadFast only works if the pin numbers are constants, which is not the case here. You could gain some speedup by writing everything manually of course, without using `ExtIO`, but that requires you to write your own MIDI output elements for this purpose, or do everything in the main `loop` function.

> I take this opportunity to ask you another question, I use 6-position selector, they allow me to select values ​​(127/6). Currently I put them as potentiometer, it works however when I turn the [...]

The following is based on the [Custom-MIDI-Output-Element](https://tttapa.github.io/Control-Surface-doc/Doxygen/d4/d3b/Custom-MIDI-Output-Element_8ino-example.html) example:
```cpp
#include <Control_Surface.h>

template <uint8_t N>
class NPositionSelector : public MIDIOutputElement {
 public:
  NPositionSelector(pin_t pin, MIDIAddress address)
    : analog(pin), address(address) {}
 
 public:
  // This method is called once by `Control_Surface.begin()`.
  void begin() final override { /* Nothing to do here */ }
 
  // This method is called continuously by `Control_Surface.loop()`.
  void update() final override {
    if (analog.update()) {
      uint8_t selection = analogValueToSelection(analog.getValue());
      if (selection != previousSelection) {
        Control_Surface.sendCC(address, selectionToMIDI(selection));
        previousSelection = selection;
      }
    }
  }

  constexpr static uint8_t resolution = 7; // bits

  static uint8_t analogValueToSelection(analog_t v) {
    return N * v / (1u << resolution);
  }
  static uint8_t selectionToMIDI(uint8_t s) {
    return (128 * s + 64) / N;
  }
 
 private:
  AH::FilteredAnalog<resolution> analog;
  MIDIAddress address;
  uint8_t previousSelection = 0xFF;
};
  
// :::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::: //

USBMIDI_Interface midi;
 
NPositionSelector<6> sel {
  A0,              // Analog input pin
  {7, CHANNEL_1},  // MIDI address
};
 
void setup() {
  Control_Surface.begin(); // Initialize Control Surface (calls button.begin())
}
 
void loop() {
  Control_Surface.loop(); // Update the Control Surface (calls button.update())
}
```