tttapa/Control-Surface #515: NoteButtonMatrix with Teensy 4.0

JRS-VPO, 2021-07-06:
Hi,

I have posted this on the Teensy forum, but have had no replies yet (posted on 27th June 2021). Apologies if this is not a bug, but I would appreciate some assistance.

I'm new to the world of Arduino/Teensy and am embarking on a project to build a physical organ console to work with the Hauptwerk Virtual Pipe Organ (VPO) software which relies on sending and receiving MIDI messages between the console and Hauptwerk. After searching around (having little programming experience), I discovered the Control Surface library which has the functions I need to develop the code required for the VPO console build (handling note on/off, CC messages, sysex messages, multiplexers, shift registers, etc.).

One of the core requirements for VPO consoles is reading keyboards (a typical organ has 2 x 61 note manual keyboards and a 32 note pedalboard). It is common practice for the physical keyboards to be wired in a matrix to reduce the number of wires (and of course the GPIO pin connections) and my target is to use the NoteButtonMatrix functionality (maybe together with a multiplexer on the rows to further reduce the GPIO pin numbers).

I am using a Teensy 4.0 and have set the USB type to MIDI.

I have tried the NoteButtonMatrix example sketch and even when only one button at a time is pressed, I get multiple note on/off messages being sent. I have amended the example code to reflect that I am using a 4x4 matrix keypad for testing and have successfully tested this with an Arduino UNO and the Hairless MIDI interface. Below is a MIDI-OX screenshot of the output from pressing button 1 in the matrix (this should send note on with note value '1' on channel 1) :

[image]

[code]
#include <Control_Surface.h>

USBMIDI_Interface midi;

// The note numbers corresponding to the buttons in the matrix
const AddressMatrix<4, 4> addresses = {{
  {1, 2, 3, 4},
  {5, 6, 7, 8},
  {9, 10, 11, 12},
  {13, 14, 15, 16},
}};

NoteButtonMatrix<4, 4> buttonmatrix = {
  {2, 3, 4, 5}, // row pins
  {6, 7, 8, 9},    // column pins
  addresses,    // address matrix
  CHANNEL_1,    // channel and cable number
};

void setup() {
  Control_Surface.begin();
  Serial.begin(115200);
  }

void loop() {
  delay(250);
  Control_Surface.loop();
}
[/code]

Note on/off messages are continually sent whilst the button is pressed.

I have checked the wiring of the keypad (and as mentioned above) successfully tested the sketch with Arduino UNO. There are no diodes in the keyboard matrix, but I don't believe these are necessary at this stage of testing as I am only pressing one button at a time.

Has anyone experienced a similar issue and found a resolution please?

Regards,
Richard

JRS-VPO, 2021-07-06:
When I upload the NoteButtonMatrix example to the Teensy 4.0 with the CPU speed set to 24 MHz it works as expected. If I upload at any higher CPU speed I get the behaviour reported in my original post.

tttapa, 2021-07-06:
The Teensy is simply too fast, it can read/write its pins faster than the time it takes for these changes to propagate to the buttons.

You can add a small delay as suggested here: https://github.com/tttapa/Control-Surface/issues/97  
I'll see if I can push a fix later this week.

JRS-VPO, 2021-07-06:
Thanks Pieter and sorry I hadn't found this amongst the GitHub issues. I will test out adding the delay suggested in issue #97.

Your library is fantastic and a real help to me in abstracting the difficult stuff around MIDI transports, handling MIDI messages in and out and adding things like the keyboard matrix, multiplexers and shift registers (I have just successfully got a sketch working with an Arduino Pro Micro combining mux's and the note button matrix so I could read a full 64 note keyboard with the Pro Micro with its relatively low number of GPIO pins).

There is an increasing interest in the Hauptwerk forum around Arduino projects to connect old organ consoles to Hauptwerk, but most of these are based on the Arduino Mega  (and people writing code from scratch) due to the number of pins available and are then of course restricted to 5 pin MIDI (without resorting to flashing firmware to get it to be MIDI class compliant).

I think that with a bit more experimenting (e.g. Hauptwerk sends a lot of information to be displayed on LCD/oled/TFT displays as sysex messages which doesn't make it straightforward to parse the required text from the sysex buffer) I will do some educating around the benefits of the Control Surface library with Teensy and some extra hardware (e.g. CH4607 MUX's) to enable a 'modular' build of the encoders/decoders required to turn old organ consoles (or build new ones) to work with Hauptwerk.

Thanks again for your prompt response and ongoing endeavours in maintaining and improving the library.

Regards,
Richard

JRS-VPO, 2021-07-07:
Adding the DelayMilliseconds(5) as per issue #97 has fixed the issue with the Teensy 4.0 as well.

tttapa, 2021-07-09:
I'm glad to hear you find the library useful!

> (e.g. Hauptwerk sends a lot of information to be displayed on LCD/oled/TFT displays as sysex messages which doesn't make it straightforward to parse the required text from the sysex buffer)

This might be interesting for inspiration: https://github.com/tttapa/Control-Surface/blob/master/src/MIDI_Inputs/MCU/LCD.hpp

JRS-VPO, 2021-07-17:
Thanks for the inspiration. I had to re-learn about pointers (I last did any C programming at Uni in 1984!) and have now managed to adapt the SysEx-Send-Receive example to copy only the portion of the sysex buffer data that I need using the memcpy function. It's a bit crude with array positions and copy length 'hardcoded', but Hauptwerk has been using the same format for sysex messages over various software versions and I can't see it changing in the near future.

I'm sure I will have more questions in the future, but will close this issue as resolved.

closed 2021-07-17 by JRS-VPO