tttapa/Control-Surface #543: Bankable::CCAbsoluteEncoder wont compile - "CS::Bankable' does not name a type"

Lemonmeth, 2021-08-10:
**Describe the bug**
Compiler returns:
sketch_aug10a:15:11: error: 'CCAbsoluteEncoder' in namespace 'CS::Bankable' does not name a type
 Bankable::CCAbsoluteEncoder enc[] {
           ^~~~~~~~~~~~~~~~~
exit status 1
'CCAbsoluteEncoder' in namespace 'CS::Bankable' does not name a type

**To Reproduce**
Compile/Validate code.

**Expected behavior**
Documentations says CCAbsoluteEncoder is bankable, but does not seem to accept it.
https://tttapa.github.io/Control-Surface-doc/Doxygen/d7/d2c/classBankable_1_1CCAbsoluteEncoder.html#a8b4a784e03e643c42d8a9a1c4e309473

Bankable::CCRotaryEncoder works without problem.

**Code**
```cpp
#include <Encoder.h>
#include <Control_Surface.h>

USBMIDI_Interface midi; // MIDI Interface to use
 
Bank<2> bank(2);

IncrementSelectorLEDs<2> bankSelector {
  bank,
  {3},         // button pins
  {4, 5}, // LED pins
};
 
using namespace MIDI_CC;
Bankable::CCAbsoluteEncoder<2> enc[] {
  {{bank, BankType::CHANGE_ADDRESS}, {16, 8}, {Sound_Controller_1, CHANNEL_1}},
  {{bank, BankType::CHANGE_ADDRESS}, {10, 9}, {Sound_Controller_2, CHANNEL_1}},
};
 
void setup() {
  Control_Surface.begin();
}
 
void loop() {
  Control_Surface.loop();
}
```

Goal:
- Bankable set of two encoders with relative input (+1), workaround for jumping values when switching banks. CCSmartPots works but still have to "realign" pot with last value (before switching) to continue sending midi.

tttapa, 2021-08-10:
What version of Control Surface do you have installed? The documentation you linked to is for the master version, previous releases do not have `Bankable::CCAbsoluteEncoder` yet.

Lemonmeth, 2021-08-10:
I have the 1.2.0 version. That would explain the problem. Im not well versed in git unfortunately, how would i obtain latest version?
Thanks for your quick reply

Lemonmeth, 2021-08-10:
Oh i see, ofc! i feel foolish.. Got the master and it compiles now! :)

Thanks for your time, and sorry to trouble you.
Take care :)

closed 2021-08-10 by Lemonmeth