tttapa/Control-Surface #552: ESP32 and BLE constant reboots with minimal code

sail2themoon, 2021-08-19:
I have an "Espressif ESP32 Dev Kit Board WROOM 32", and as soon as I use the bluetooth function of this library, I get reboots all the time. Also the board makes a little noise every second it reboots, like some hardware was shorted...Which I can hardly imagine. With every other code, the board runs fine. Also bluetooth is working with the "ESP32 BLE" library.

This is my test-code, based on "MIDI-monitor" here. The line "MIDI_Interface::beginAll();" causes the reboots. As you can see, nothing else is happening here:

// Instantiate the two MIDI interfaces
BluetoothMIDI_Interface midible;
//USBDebugMIDI_Interface mididbg;
 
// Instantiate a MIDI pipe to connect the two interfaces
//BidirectionalMIDI_Pipe mpipe;
 
void setup() {
  // Create a bidirectional route between the interfaces: 
  // all input to one interface is sent to the output of the other
//  midible | mpipe | mididbg;
  // Initialize all MIDI interfaces
  MIDI_Interface::beginAll();
}
 
void loop() {
  // Continuously handle MIDI input
//  MIDI_Interface::updateAll();
}
 
```
// Instantiate the two MIDI interfaces
BluetoothMIDI_Interface midible;
//USBDebugMIDI_Interface mididbg;
 
// Instantiate a MIDI pipe to connect the two interfaces
//BidirectionalMIDI_Pipe mpipe;
 
void setup() {
  // Create a bidirectional route between the interfaces: 
  // all input to one interface is sent to the output of the other
//  midible | mpipe | mididbg;
  // Initialize all MIDI interfaces
  MIDI_Interface::beginAll();
}
 
void loop() {
  // Continuously handle MIDI input
//  MIDI_Interface::updateAll();
}
```

tttapa, 2021-08-20:
Thanks for letting me know! 

This issue seems to be caused by the addition of new fields in the `esp_pthread_cfg_t` struct in newer versions of the ESP-IDF and the ESP32 Arduino Core. Since they didn't exist when I wrote the code, the were not initialized, with the uninitialized variables causing issues when creating a thread.

This should be fixed in 399ad2372c844ef0b7866a6c513cd1b80d06e755. 

The following code works for me with version 1.0.6 of the ESP32 Arduino Core:
```cpp
#include <Control_Surface.h> // Include the Control Surface library
 
// Instantiate the two MIDI interfaces
BluetoothMIDI_Interface midible;
USBDebugMIDI_Interface mididbg;
 
// Instantiate a MIDI pipe to connect the two interfaces
BidirectionalMIDI_Pipe mpipe;
 
void setup() {
  // Create a bidirectional route between the interfaces: 
  // all input to one interface is sent to the output of the other
  midible | mpipe | mididbg;
  // Initialize all MIDI interfaces
  MIDI_Interface::beginAll();
}
 
void loop() {
  // Continuously handle MIDI input
  MIDI_Interface::updateAll();
}
```

sail2themoon, 2021-08-20:
You're welcome, thank you very much for that fast response and fix. No reboots any more with the updated library, running fine now.