tttapa/Control-Surface #555: Change device name

Lemonmeth, 2021-08-21:
Hi Pieter!

Im having an issue with changing name of my teensy lc

**Describe the bug**
probably not a bug.

**To Reproduce**
create name.c file:
´´´cpp
#include <usb_names.h>

#define MANUFACTURER_NAME  {'m','y','N','a','m','e'}
#define MANUFACTURER_NAME_LEN 6
#define PRODUCT_NAME    {'S','o','m','e','t','h','i','n','g'}
#define PRODUCT_NAME_LEN  9

struct usb_string_descriptor_struct usb_string_manufacturer_name = {
  2 + MANUFACTURER_NAME_LEN * 2,
  3,
  MANUFACTURER_NAME
};
struct usb_string_descriptor_struct usb_string_product_name = {
  2 + PRODUCT_NAME_LEN * 2,
  3,
  PRODUCT_NAME
};
´´´

add file name.c to sketch and compile.

**Expected behavior**
name.c file should change device name.

**Code**

```cpp
#include <Control_Surface.h>

USBMIDI_Interface midi; // MIDI Interface to use
 
Bank<2> bank(2);

IncrementSelectorLEDs<2> bankSelector {
  bank,
  {3},         // button pins
  {4, 5}, // LED pins
};
 
using namespace MIDI_CC;
Bankable::CCButtonLatched<2> lat[] {
  {{bank, BankType::CHANGE_ADDRESS}, 23, {Sound_Controller_1, CHANNEL_1}},
  {{bank, BankType::CHANGE_ADDRESS}, 20, {Sound_Controller_2, CHANNEL_1}},
  {{bank, BankType::CHANGE_ADDRESS}, 18, {Sound_Controller_3, CHANNEL_1}},
  {{bank, BankType::CHANGE_ADDRESS}, 15, {Sound_Controller_4, CHANNEL_1}},
};

Bankable::CCPotentiometer Potentiometer [] {
  {{bank, BankType::CHANGE_CHANNEL}, A0, {Sound_Controller_5, CHANNEL_1}},
};

void setup() {
  Control_Surface.begin();
}
 
void loop() {
  Control_Surface.loop();
}
```

**Error**
´´´cpp
C:\Users\B\AppData\Local\Temp\arduino_build_315404\libraries\Control-Surface-1.2.0\MIDI_Interfaces\Teensy-USB-MIDI-name.c.o:(.data.usb_string_product_name+0x0): multiple definition of `usb_string_product_name'
C:\Users\B\AppData\Local\Temp\arduino_build_315404\sketch\name.c.o:(.data.usb_string_product_name+0x0): first defined here
c:/program files (x86)/arduino/hardware/tools/arm/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld.exe: Disabling relaxation: it will not work with multiple definitions
c:/program files (x86)/arduino/hardware/tools/arm/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld.exe: Warning: size of symbol `usb_string_product_name' changed from 20 in C:\Users\B\AppData\Local\Temp\arduino_build_315404\sketch\name.c.o to 32 in C:\Users\B\AppData\Local\Temp\arduino_build_315404\libraries\Control-Surface-1.2.0\MIDI_Interfaces\Teensy-USB-MIDI-name.c.o
collect2.exe: error: ld returned 1 exit status
Error compiling for board Teensy LC.
´´´

tttapa, 2021-08-21:
The idea is that you change the names in `Teensy-USB-MIDI-name.py` (the first 20 lines), and then run the file manually.

If you don't have Python installed, you can edit the `Teensy-USB-MIDI-name.c` file if you want, it won't break anything. Or you can delete `Teensy-USB-MIDI-name.c` altogether if you want to be able to set the name in each sketch individually.

Lemonmeth, 2021-08-22:
Ah i see, is it perhaps for when using a teensy with multiple virtual cables? Removing the Teensy-USB-MIDI-name.c did the trick!

Thank you for your help! Is there a way i give back, while not being a seasoned programmer?

Take care