tttapa/Control-Surface #566: Last state smart potentiometer get value?

Nivolp1976, 2021-09-03:
Hi there. I was able to get the current value not activated from a smart potentiometer. is it possible to get the latest status on a scale of up to 117? is to work on a display.

tttapa, 2021-09-03:
Like [`CCSmartPotentiometer::getPreviousValue()`](https://tttapa.github.io/Control-Surface-doc/Doxygen/d3/dfe/classBankable_1_1CCSmartPotentiometer.html#ab955e2466aa4e6ee31bdddaea9379799)?

> on a scale of up to 117

127

Nivolp1976, 2021-09-03:
ok sorry, my learning is slow.

Nivolp1976, 2021-09-21:
When I move a control from the daw and I get the value with a callback, can I update the last value in the smart potentiometer?
something like potentiometer.set_lastvalue () = callback_lastvalue;

tttapa, 2021-09-23:
You should be able to add a `setPreviousValue(setting_t bank, analog_t value)` function to the `SmartMIDIFilteredAnalog` class: https://tttapa.github.io/Control-Surface-doc/Doxygen/db/dbc/SmartMIDIFilteredAnalog_8hpp_source.html