tttapa/Control-Surface #569: Custom selector callback question

WeaponX15, 2021-09-06:
Hi and thanks for your work and gentleness.

I´m using custom_selector_callback example with RGBs led, ¿how can use it with 3 normal leds?

I have 3 normal leds; RED (pin 10), GREEN (pin 11), YELLOW (pin 12), when i use it the example´s sketch, ocurring this:

push bank button channel 1: RED LED ON
push bank button channel 2: RED LED ON & GREEN LED ON
push bank button channel 3: GREEN LED ON

Thanks for help me.

tttapa, 2021-09-06:
> ¿how can use it with 3 normal leds?

What do you mean? What should be displayed by the LEDs ?

> when i use it the example´s sketch, ocurring this:
> 
> push bank button channel 1: RED LED ON
> push bank button channel 2: RED LED ON & GREEN LED ON
> push bank button channel 3: GREEN LED ON

Yes, that is the expected behavior, the colors are defined in the `getColor()` function:
```cpp
    // Convert the given setting to a 3-bit RGB color value.
    static uint8_t getColor(setting_t setting) {
      switch (setting) {
        case 0: return 0b001; // Red on, Green off, Blue off (first bank)
        case 1: return 0b011; // Red on, Green on, Blue off (second bank)
        case 2: return 0b010; // Red off, Green on, Blue off (third bank)
        ...
    }
```

> Second question; Each bank just can have 8 encoders (V_POT_1....V_POT_8)?

If your DAW supports it, you can use any controller number between from 0 to 120 on any of the 16 MIDI channels. See https://tttapa.github.io/Control-Surface-doc/Doxygen/d3/df7/midi-tutorial.html#midi_md-midi-addresses for more information about MIDI addresses.  
If you're using the Mackie Control Universal protocol you're limited to the eight `V_POT_n` constants.

WeaponX15, 2021-09-08:
Thanks for your answers.

How can i do to CCAbsoluteEncoder have no limit? or loop valours? (1,2,3,...127,0,1,2,3...). I use a rotary encoder as "potentiometer" (limitations of vst) and i need that encoder have no limit or loop functionality.

tttapa, 2021-09-08:
> How can i do to CCAbsoluteEncoder have no limit? or loop valours? (1,2,3,...127,0,1,2,3...).

You would have to implement this yourself, see the call to `constrain` in the `MIDIAbsoluteEncoder` implementation:

https://github.com/tttapa/Control-Surface/blob/d59741dba1686de75016562bfe3d0d04cac24ca1/src/MIDI_Outputs/Abstract/MIDIAbsoluteEncoder.hpp#L35-L46