tttapa/Control-Surface #578: Arduino nano 33 IOT unstable potenziometer

alfredolo72, 2021-09-22:
Hi,
i am making a little stream deck. I use 3 potenziometer 10K linear.
Arduino nano 33 IOT - connected beetween 3.3v and gnd.
The exist value of potenziometer are very unstable. I tried both USBMIDI_Interface and HairlessMIDI_Interface to see any difference.
I have not problems if i connect my little stream deck to arduino uno. It works perfect.
Arduino uno - connected beetween 5v and gnd.
Any idea ?

Veroledez, 2021-09-23:
Si tienes un multimetro prueba midiendo el voltaje entre 3.3v y gnd, y conecta los potenciometros, si hay variacion  de voltaje usa otro regulador de 3.3v para los potenciometros  gnd comun entre el arduino  el nuevo regulador

tttapa, 2021-09-23:
Do you see the same issues using the AnalogReadSerial Arduino example? (You can use the Serial Plotter in the IDE to test this.)

This is most likely a hardware problem.

alfredolo72, 2021-09-23:
Thanks for reply.
There is a hardware issue on nano.
https://forum.arduino.cc/t/extreme-noisy-board-voltage-3-3v-on-nano-33-iot/623016
There is a lot of noise. Here is some result reading with Serial.println(potentiometer1.getRawValue());
776
804
772
676
804
748
756
756
736
816
792
772
768
760

And here is the same code with Arduino uno and the same potenziometer.
704
704
704
720
704
704
720
704
704
720
720
720
704
704
704
720
720
704

I will send it back and i will get a different board.
Thank you all and dont buy it...

Veroledez, 2021-09-23:
Prueba a poner un capacitor  de 1uf o menos, un pin a gnd y el otro  pin del capacitor al pin analogico

alfredolo72, 2021-09-23:
I tried it. Dont work, there is no difference with or without capacitor

Veroledez, 2021-09-24:
Es problema del regulador, usa uno lineal de 3.3v y prueba nuevamente

alfredolo72, 2021-09-24:
Thanks for reply.
I dont like idea to buy a defective product and add extra hardware to fix it.
I sended it back and buyed teensy 4.1 . I have just to wait few days.
Thank you