tttapa/Control-Surface #583: ESP32 Flawless under Mac, buggy under Windows 10

tiziano149, 2021-10-07:
Hello community,

I have been experimenting with ESP32. 
Under Mac the Bluetooth implementation works flawlessly without connectivity issues.

On windows 10, although always recognized as a native BLE Midi device (but still buggy crashing every DAW if the amount of input data is too high) it behaves very unexpectedly.

When I use midiberry, most of the times it is connected properly from windows settings but midiberry just reads "Device information found!" but with no input.

After installing a million Windows updates, restarting my pc many times and installing some optional Updates from here I managed to get it working: http://newbodyfresher.linclip.com/ "KB4586853 on Windows 10, Version 2004 / 20H2."

Now, I was quite sure it was windows fault but I tried some other ESP32 MidiBLE libraries and they work flawlessly, so it must be a bug in control surface.

I tried:

-https://github.com/max22-/ESP32-BLE-MIDI

-https://github.com/lathoub/Arduino-BLE-MIDI

Both work without issues.

I also sometimes get that when I pair the ESP32 it shows up as "connected" and then drops connection after around 20secs, so it is very inconsistent.

After a full day of it perfectly working under Windows 10, now it is not recognized anymore. The other libraries still work perfectly.

Has anyone experienced a similiar issue and found a solution?
Here I leave the code I am trying to run.

Best wishes
Tiziano

```

#include <Control_Surface.h> // Include the Control Surface library

// Instantiate a MIDI interface to use for the Control Surface.
BluetoothMIDI_Interface midi;

// Instantiate a CCPotentiometer object
CCPotentiometer potentiometer {
  A0,                                   // Analog pin connected to potentiometer
  {MIDI_CC::General_Purpose_Controller_4, CHANNEL_1}, // Channel volume of channel 1
};

void setup() {
  Control_Surface.begin(); // Initialize Control Surface
}

void loop() {
  Control_Surface.loop(); // Update the Control Surface
}
```

Veroledez, 2021-10-08:
Hola, probe un pas  veces ble  windows y en todas me tiraba mucha latencia por lo que no lo use, sin embargo para emparejarlo correctamente instale el driver del korg,
https://www.korg.com/us/support/download/driver/0/530/2886/
Instalalo y nos cuentas como te fue.

Veroledez, 2021-10-08:
Hola, el ble lo probe hace mas de 6 meses, use la rama master, y lo ultimo disponible en ese entonces, solo tuve problemas de latencia, tenia una version vieja de win10, ahora actualizado no se pase. De codigo use 16 potenciometros  y 37 botones, nunca se desconecto y conecto a la primera, el codigo ya no lo tengo, lo borre luego de intentar solucionar la latencia, era de 1 segundo o más a veces. 
El problema parece ser de control surface y windows, ya que si funciona en mac y con otras librerias ble, el problema es ambas.

Esperemos si Peter puede darnos otro diagnostico de lo que sucede

tttapa, 2021-10-10:
This is quite odd, I've never had issues on Linux. While I only did a brief test on Windows (several months ago), I didn't notice any problems there either. (Granted, I didn't pay too much attention to Windows, because contrary to Linux and OSX/iOS, it had no decent MIDI BLE support, at least out of the box, I'm not a Windows user myself, so I don't know how the situation changed since then.)

One of the differences is the connection interval, Control Surface sets this to a short interval, as per the MIDI over BLE standard, whereas the other libraries use the default setting. Control Surface does this to minimize latency.  
Another difference is the MTU, some other libraries I've tested use an MTU of 23 bytes, which is tiny. Control Surface uses an MTU of 517, which results in much higher bandwidth.

Maybe Windows doesn't really like these settings? I don't rule out other differences or even bugs, but this might be an easy thing to check before diving deeper.

You can find these settings here:
https://github.com/tttapa/Control-Surface/blob/7a3b189776bb01bcbd24da38b58703a6923716e0/src/MIDI_Interfaces/BLEMIDI/ESP32/advertising.c#L12-L14

https://github.com/tttapa/Control-Surface/blob/7a3b189776bb01bcbd24da38b58703a6923716e0/src/MIDI_Interfaces/BLEMIDI/ESP32/midi-init.c#L47

Compare them to the parameters used by the ESP32 Arduino BLE library, for instance:

https://github.com/espressif/arduino-esp32/blob/7c3a82a52591894bf31726c64badf68e31817df8/libraries/BLE/src/BLEAdvertising.cpp#L37-L38

The two MIDI BLE libraries you mentioned never seem to change the MTU, so I think you can assume it to be 23.

Another thing that might be worth looking into is the connection parameter update that happens here:

It is derived from the Espressif examples, e.g. https://github.com/espressif/esp-idf/blob/1cb31e50943bb757966ca91ed7f4852692a5b0ed/examples/bluetooth/bluedroid/ble/gatt_server/main/gatts_demo.c#L467-L478, but I'm unsure how it relates to the other connection parameters that were set for advertising and I couldn't find any information online.

The ESP-IDF that's included with the Arduino ESP32 core has been updated since when I was developing BLE MIDI support, and the ESP-IDF BLE API might have changed since then.

I'll definitely look into this, but I don't have much free time at the moment.

tttapa, 2021-10-10:
> I didn't understand how to set the MTU to 23 sadly

You should be able to just comment out these lines:
https://github.com/tttapa/Control-Surface/blob/7a3b189776bb01bcbd24da38b58703a6923716e0/src/MIDI_Interfaces/BLEMIDI/ESP32/midi-init.c#L47-L51

tiziano149, 2021-10-10:
nope, still crashes sadly :/

Veroledez, 2021-10-10:
Prueba con bome midi translator en lugar de loopmidi, es mas eficiente

josehdx, 2022-08-27:
> Hello community,
> 
> I have been experimenting with ESP32. Under Mac the Bluetooth implementation works flawlessly without connectivity issues.
> 
> On windows 10, although always recognized as a native BLE Midi device (but still buggy crashing every DAW if the amount of input data is too high) it behaves very unexpectedly.
> 
> When I use midiberry, most of the times it is connected properly from windows settings but midiberry just reads "Device information found!" but with no input.
> 
> After installing a million Windows updates, restarting my pc many times and installing some optional Updates from here I managed to get it working: http://newbodyfresher.linclip.com/ "KB4586853 on Windows 10, Version 2004 / 20H2."
> 
> Now, I was quite sure it was windows fault but I tried some other ESP32 MidiBLE libraries and they work flawlessly, so it must be a bug in control surface.
> 
> I tried:
> 
> -https://github.com/max22-/ESP32-BLE-MIDI
> 
> -https://github.com/lathoub/Arduino-BLE-MIDI
> 
> Both work without issues.
> 
> I also sometimes get that when I pair the ESP32 it shows up as "connected" and then drops connection after around 20secs, so it is very inconsistent.
> 
> After a full day of it perfectly working under Windows 10, now it is not recognized anymore. The other libraries still work perfectly.
> 
> Has anyone experienced a similar issue and found a solution? Here I leave the code I am trying to run, which works under MacOS but struggles to run properly on windows.
> 
> Best wishes Tiziano
> 
> ```
> 
> #include <Control_Surface.h> // Include the Control Surface library
> 
> // Instantiate a MIDI interface to use for the Control Surface.
> BluetoothMIDI_Interface midi;
> 
> // Instantiate a CCPotentiometer object
> CCPotentiometer potentiometer {
>   A0,                                   // Analog pin connected to potentiometer
>   {MIDI_CC::General_Purpose_Controller_4, CHANNEL_1}, // Channel volume of channel 1
> };
> 
> void setup() {
>   Control_Surface.begin(); // Initialize Control Surface
> }
> 
> void loop() {
>   Control_Surface.loop(); // Update the Control Surface
> }
> ```

Hola

Disculpa, podras compartir codigos sencillos de botones de las dos librerias adicionales que comentaste?

Es que me pasa algo muy raro, no consigo enviar nada de datos desde mi ESP32 a mi laptop, a pesar de tener emparejado y conectado de manera correcta via BLE. En midi berry no veo nada de datos MIDI en el debugger y ya he intentado con varios ejemplos de la libreria de control surface

Lo raro es que si ni uso BLE, si hago una conexion con el BT clasico que le llaman por puerto serie, si puede enviar datos cuando presiono los botones en los pines seleccionados en un codigo q copie de internet.

Me parece que he descargado con anterioridad alguna de las librerias que menciones, pero me gustaria algun codigo por favor

Saludos!