tttapa/Control-Surface #584: Calibration potentiometre

e36910, 2021-10-09:
Hello! I would like calibration potentiometre. F.E: When I use value in pot 30% I get 0%, when I use value in pot 70% I get 100%. Another word 0 = from 0 to 30, and 70 = from 70 to 100(max)
Which I need use code? Thank.

e36910, 2021-10-09:
I using this code, but nothing changes.

`#include <Control_Surface.h> // Include the Control Surface library

// Instantiate a MIDI over USB interface.
HairlessMIDI_Interface midi;
//USBMIDI_Interface midi;

//leds
const pin_t ledPins[] =  {2, 3, 4, 5, 6, 7, 8, 9, 10, 11};

using namespace MIDI_Notes;

//массив кнопок
CCButton button[] = {
  {2,{note(C, 4), CHANNEL_1}}, //кнопка подключена на пин 2 и посылает ноту C4 на канал 1
  {3,{note(D, 4), CHANNEL_1}},
  {4,{note(E, 4), CHANNEL_1}},
  {5,{note(F_, 4), CHANNEL_1}},
  {6,{note(G, 4), CHANNEL_1}},
  {7,{note(Gb, 4), CHANNEL_1}},
  {8,{note(A, 4), CHANNEL_1}},
  {9,{note(Ab, 4), CHANNEL_1}},  
  {10,{note(B, 4), CHANNEL_1}},
  {11,{note(Bb, 4), CHANNEL_1}},
};

//button array
PCButton buttons[] = {
  {8, {35, CHANNEL_2}}, //button on pin 2 and send MIDI CC 46
  {9, {36, CHANNEL_2}}, 
  {10, {37, CHANNEL_2}},
  {11, {38, CHANNEL_2}},
 };

//массив потенциометров
CCPotentiometer potentiometers[] = {
  {A0, { 26, CHANNEL_1}}, //потенциометр подключен к пину А0, посылает MIDI CC 26, на  канал 1
};

// The value to send when the potentiometer is in the 0% position
constexpr analog_t minimumValue = 5000;
// The value to send when the potentiometer is in the 100% position
constexpr analog_t maximumValue = 15000;
 
analog_t mappingFunction(analog_t raw) {
    // map the value from [0, 16383] to [minimumValue, maximumValue]
    return map(raw, 0, 16383, minimumValue, maximumValue);
    // Note: 16383 = 2¹⁴ - 1 (the maximum value that can be represented by
    // a 14-bit unsigned number
}

void setup() {
  Control_Surface.begin(); // Initialize Control Surface
}

void loop() {
  Control_Surface.loop(); // Update the Control Surface
}`

tiziano149, 2021-10-09:
did you try this example?
```

// Include the Control Surface library
#include <Control_Surface.h>
 
// Instantiate a MIDI over USB interface.
USBMIDI_Interface midi;
 
// Create a new instance of the class `CCPotentiometer`, called `potentiometer`,
// on pin A0, that sends MIDI messages with controller 7 (channel volume)
// on channel 1.
CCPotentiometer potentiometer {
  A0, {MIDI_CC::Channel_Volume, CHANNEL_1}
};
 
// The filtered value read when potentiometer is at the 0% position
constexpr analog_t minimumValue = 255;
// The filtered value read when potentiometer is at the 100% position
constexpr analog_t maximumValue = 16383 - 255;
 
// A mapping function to eliminate the dead zones of the potentiometer:
// Some potentiometers don't output a perfect zero signal when you move them to
// the zero position, they will still output a value of 1 or 2, and the same
// goes for the maximum position.
analog_t mappingFunction(analog_t raw) {
    // make sure that the analog value is between the minimum and maximum
    raw = constrain(raw, minimumValue, maximumValue);
    // map the value from [minimumValue, maximumValue] to [0, 16383]
    return map(raw, minimumValue, maximumValue, 0, 16383);
    // Note: 16383 = 2¹⁴ - 1 (the maximum value that can be represented by
    // a 14-bit unsigned number
}
 
void setup() {
    // Add the mapping function to the potentiometer
    potentiometer.map(mappingFunction);
    // Initialize everything
    Control_Surface.begin();
    Serial.begin(115200);
}
 
void loop() {
    // Update the Control Surface (check whether the potentiometer's
    // input has changed since last time, if so, send the new value over MIDI).
    Control_Surface.loop();
    // Use this to find minimumValue and maximumValue: it prints the raw value
    // of the potentiometer, without the mapping function
    Serial.println(potentiometer.getRawValue());
}
```

e36910, 2021-10-09:
I'm using arduino nano atmega 328p.

On this line I am getting an error
`// Use this to find minimumValue and maximumValue: it prints the raw value
    // of the potentiometer, without the mapping function
    Serial.println(potentiometer.getRawValue());`

closed 2021-10-10 by e36910