tttapa/Control-Surface #627: One press only commands 127. I would like 1 press and 1 command only 127 velocity. How to do that?

e36910, 2021-11-25:
I need working with abelton track. It understand only 1 CC command only 127 velocity. Thank you!

Veroledez, 2021-11-26:
Hola, no entiendo bien tu pregunta, si necesitas enviar mensajes cc en lugar de notas, con valor de 127, usa esto

https://github.com/tttapa/Control-Surface/blob/master/examples/1.%20MIDI%20Output/2.%20Buttons%20%26%20Switches/1.%20Momentary%20Push%20Buttons/CCButton/CCButton.ino

e36910, 2021-11-26:
Ccbutton works for one push vel 127, next push 0 vel.

Veroledez, 2021-11-26:
Parece que quieres solo enviar el cc y 127,  para omitir el 0 que es como el apagado por asi decirlo, debes crear tu propia clase, el ejemplo custom midi output te servira, debes borrar la parte en la que se verifica el estado rising del boton

e36910, 2021-11-26:
Yes. How to write code for setup max vel 127?

tttapa, 2021-12-05:
First define a custom CC sender:

https://github.com/tttapa/Control-Surface/blob/e6542665da8997fdf33aee5ff1e0b7383c7ebe00/src/MIDI_Senders/DigitalCCSender.hpp#L13-L37

Just leave the `sendOff()` function empty.

Then create a MIDI button with it:

https://github.com/tttapa/Control-Surface/blob/e6542665da8997fdf33aee5ff1e0b7383c7ebe00/src/MIDI_Outputs/CCButton.hpp#L20-L38