tttapa/Control-Surface #645: Bankable::CCButtonLatched disables the whole interface

FotisDovas, 2021-12-17:
**Describe the bug**
After creating an array of bankable ccButtonLatched, the whole interface bleeds to nothingness :P 
And by this, i mean that 
1) Leds showing the current bank selection don't light up when the encoderSelectorLeds is rotating.
2) Serial monitor prints nothing at all, ever.
I tried both muxed pin for the button, as well as Analog pins straight on the microcontroller.
All other (unbanked) buttons work nicely. Even custom ones.
There is no syntax error of any sort, and the program compiles and uploads fine.
Strange..

**To Reproduce**
Create a similar array, with at least one CCButtonLatched, and a bank<5> to combine with the items.
Teensy 3.5

**Expected behavior**
I would expect the software to keep sending data through the teensy 3.5 after instantiation of the latchedButton array

**Code**

```cpp
#include <Control_Surface.h>
USBMIDI_Interface midi;

Bank<5> bankMain;
                                                 //--- FUNCTION BUTTONS
 Bankable::CCButtonLatched<5> functionButtons[] = {
   {{bankMain, BankType::CHANGE_CHANNEL},  
    A3,
   {0x0D ,CHANNEL_2},
   }
  };  

EncoderSelectorLEDs<5> bankSelectorEncoder {
  bankMain,
  {24,25}, //controlSignalPins
  {33, 34, 35, 36 ,37}, //LedPins
  8, Wrap::Wrap }; // wanted the encoder to be less sensitive, and also wrap

void setup() {
Serial.begin(38400);

  pinMode (24, INPUT_PULLUP);
  pinMode (25, INPUT_PULLUP);

  pinMode(37,OUTPUT);    
  pinMode(36,OUTPUT);
  pinMode(35,OUTPUT);
  pinMode(34,OUTPUT);
  pinMode(33,OUTPUT);
}

void loop() {
   Control_Surface.loop();
Serial.println(bankSelectorEncoder.get());
}
```

FotisDovas, 2021-12-17:
It is a mux issue, and pullup resistor specific.. closing issue, bankable latched works fine

Veroledez, 2021-12-17:
Hola, en el void setup falta llamar a:
Control_Surface.begin();

Ademas no es necesario declarar si los pines son entrada o salida, todo lo hace la biblioteca automaticamente cuando usas sus instancias.

FotisDovas, 2021-12-17:
Yeah, i forgot to past that on the snipet above, but had it on the main code. And you are also correct about the pinmodes, they are optional.
The mistake i had made was that when using a mux pin for one button I missplaced a number, and had muxes[0].pin(16), which is impossible for a 0-15 mux hehe sorry
Thanks for the reply though