tttapa/Control-Surface #650: Can I use wifi connection with node mcu v3 (esp 8266)?

e36910, 2021-12-23:
Scetch a good working with usb connection with node mcu v3 (esp 8266). Can I use rtpMidi via wifi?

tttapa, 2021-12-23:
Here's an example for the ESP32: https://tttapa.github.io/Control-Surface-doc/Doxygen/dc/d7d/AppleMIDI_8ino-example.html

The code for the ESP8266 is similar, please refer to the AppleMIDI library documentation for details.

e36910, 2022-01-06:
I don't understand this, could you help me edit this code? Substitute AppleMIDI instead of hairless
`#include <Control_Surface.h> 

HairlessMIDI_Interface midi;

using namespace MIDI_Notes;

CCButtonLatched button[] = {
  {5,{61, CHANNEL_1}}, 
  {4,{62, CHANNEL_1}},
  {2,{64, CHANNEL_1}},
  {14,{65, CHANNEL_1}},
  {12,{66, CHANNEL_1}},
  {13,{67, CHANNEL_1}},
  {3,{68, CHANNEL_1}},
  {0,{63, CHANNEL_1}},
   
};

//button array
PCButton buttons[] = {
  {5, {35, CHANNEL_2}}, //D1
  {4, {36, CHANNEL_2}}, //D2
  {2, {37, CHANNEL_2}}, //D4
  {14, {38, CHANNEL_2}}, //D5
  {12,{39, CHANNEL_2}}, //D6
  {13,{40, CHANNEL_2}}, //D7
  {3,{34, CHANNEL_2}}, //RX
  {0,{33, CHANNEL_2}}, //D3
 }; 

CCPotentiometer potentiometers[] = {
{A0, { 26, CHANNEL_1}}, 
};

constexpr analog_t minimumValue = 0; //30% 16383 * 0,3 = 4915
constexpr analog_t maximumValue = 16383; //70% 16383 * 0,7 = 11468

analog_t mappingFunction(analog_t raw) {
return map(raw, 1000, 13000, minimumValue, maximumValue);
}

void setup() {
potentiometers[0].map(mappingFunction);
Control_Surface.begin(); // Initialize Control Surface
}

void loop() {
Control_Surface.loop(); // Update the Control Surface
}

`

Veroledez, 2022-02-09:
Hola, tal vez esto te sirva, no volvi a probar, si consigues lecturas analogicas postea tu codigo. Graciad.

 https://github.com/tttapa/Control-Surface/discussions/476#discussion-3368492