tttapa/Control-Surface #653: Websocket not working

alfredolo72, 2021-12-24:
Hello,
i have maked a ethernet + websocket script, it works.
Adding control_surfare i get errors.
I am using Teensy 4.1
To reproduce the error make a new sketch with 👍 
```
#include <ArduinoWebsockets.h>
#include <Control_Surface.h>
```
Here is error :

c:/users/edo/downloads/arduino-1.8.15/hardware/tools/arm/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/lib/armv7e-m/fpu/fpv5-d16\libstdc++.a(functexcept.o): In function `std::__throw_bad_alloc()':

functexcept.cc:(.text._ZSt17__throw_bad_allocv+0x0): multiple definition of `std::__throw_bad_alloc()'

C:\Users\Edo\AppData\Local\Temp\arduino_build_503595\libraries\Control-Surface-master\AH\STL\objs.a(vector.cpp.o):C:\Users\Edo\Documents\Arduino\libraries\Control-Surface-master\src\AH\STL/vector.cpp:6: first defined here

c:/users/edo/downloads/arduino-1.8.15/hardware/tools/arm/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld.exe: Disabling relaxation: it will not work with multiple definitions

c:/users/edo/downloads/arduino-1.8.15/hardware/tools/arm/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/lib/armv7e-m/fpu/fpv5-d16\libstdc++.a(functexcept.o): In function `std::__throw_length_error(char const*)':

functexcept.cc:(.text._ZSt20__throw_length_errorPKc+0x0): multiple definition of `std::__throw_length_error(char const*)'

C:\Users\Edo\AppData\Local\Temp\arduino_build_503595\libraries\Control-Surface-master\AH\STL\objs.a(vector.cpp.o):C:\Users\Edo\Documents\Arduino\libraries\Control-Surface-master\src\AH\STL/vector.cpp:6: first defined here

collect2.exe: error: ld returned 1 exit status

Any idea ?
thank you

tttapa, 2021-12-24:
It looks like the bug in Teensyduino that the vector.cpp file was meant to work around has been fixed in Teensyduino 1.53, so I think you can just delete vector.cpp altogether.

alfredolo72, 2021-12-24:
Thank you for reply.
I installed now latest Arduino 1.8.19 and Teensyduino 1.56 and the problem persist.
Where is the location of vector.cpp to delete ?

tttapa, 2021-12-24:
C:\Users\Edo\Documents\Arduino\libraries\Control-Surface-master\src\AH\STL\vector.cpp

Teensyduino <1.53 didn't define `std::__throw_length_error`, so Control Surface defined it. Now Teensyduino >=1.53 has been fixed and defines it, but you'll have to delete Control Surface's definition.

alfredolo72, 2021-12-24:
it works 
thank you !

commit 9a7f07d4 (2022-04-17): Support latest Teensyduino
Fixes #653
closed 2022-04-17 by tttapa