tttapa/Control-Surface #659: Need help for fader midi

cuuhodrc, 2021-12-28:
Hi,

Could you help me to create the example code follow this function on this website of arduino pro micro?

https://nuancescontroller.fr

Best regards,

tttapa, 2021-12-28:
Did you try this example?  
https://tttapa.github.io/Control-Surface-doc/Doxygen/d3/d8d/Multiple-Control-Change-Potentiometers_8ino-example.html

cuuhodrc, 2021-12-28:
Hi,
I tried but i don't know how to add these function:
- You can press the switch to change each Midi controller number 
   your custom MIDI CC choice is stored in ROM
- A tiny oled 0.91 screen displays the midi controller number on the top and the value on the bottom. ( when press button can change the CC value of each faders)
- The display goes to sleep after 2 minutes of inactivity in order to increase life span.
- The first row is CC value, the second row show the number of CC when moving fader
[image]