tttapa/Control-Surface #690: Bank on Teensy LC not working IncrementDecrementSelector

gringow831, 2022-02-02:
Hi Pieter,

I am having a few issues with my Teensy LC. 
IncrementDecrementSelector doesn't work. I have tried the same code with a Leonardo and everything works there.
I have tried many combinations of the USB type: midi, midix4, serial+midi, serial+midix4. The following code is the simplest I tried to make the banks work before I try making CABLENB work

The second thing which I can't get my head around is why if I want to use virtual cables midix4 doesn't work but I when selecting serial+midix4 it does.

```cpp
#include <Control_Surface.h>

USBMIDI_Interface midi;

Bank<2> bank(8);
IncrementDecrementSelector<2> bankSelector = {bank, {2, 3}, Wrap::Wrap};

Bankable::PBPotentiometer knobs[] = {
  { bank, A0, {CHANNEL_1}},
};

void setup() {
  RelativeCCSender::setMode(MACKIE_CONTROL_RELATIVE);
  Control_Surface.begin();
}

void loop() {
  Control_Surface.loop();
}
```

closed 2022-02-04 by gringow831

tttapa, 2022-02-04:
There shouldn't really be any difference between the Teensy and the Leonardo. Did you get it to work?

gringow831, 2022-02-04:
I still don't understand what's going on with my LC. It does work on my Teensy 4.0

dchstudios, 2023-02-28:
It does not work on my Teensy 4.1.  Then again, for some reason, I cannot get any Bank implementation working on Any of my  boards.. I'm on Arduino 2.0.4  I've tried on Teensy 4.1, Adafruit Feather S3, Adafruit QtPy C3, Unexpected Maker (UM) Tiny Pico, UM Tiny S3, UM Feather S3, Adafruit Feather Huzzah32, a couple Dev boards of C3 and other ESP32 boards.... Zero success in Banks. The main thing is I'm trying to create a DAW Control Surface that's wireless, preferably bluetooth.. (I hate having the neck of my guitar hit the monitor and/or mic any time I lean over to fiddle with Play/Record/Delete/Rewind and mouse control etc on recording many takes). I can get most other aspects of Control Surface working, just not Banks, nor this same IncrementDecrementSelector.  They compile, and for e.g I see a potentiometer sending MIDI signals via Hairless, just no bank functionality. 95% of them I'm trying to work with BluetoothMIDI just because they have that and is what I'm going for.  The Teensy 4.1 was just to see if I could get Banks working with USB, just to see Bank implementation actually work on Something. Nope!  ANY help would be awesome.