tttapa/Control-Surface #715: EEPROM - Save ManyAddresses settings in the Arduino

gringow831, 2022-02-28:
Hi Pieter.

I was wondering if there is a way to save the settings of a project to next time I power up I start from that point. Not sure if EEPROM would be the way or if there is another way to do this?

Let's say I have a ManyAddresses with 4 addresses and during the last part before I come back to the project or lose power I want to continue where I left off in the 3rd address for example. Is there a way to do that?

tttapa, 2022-03-02:
In your setup, read the previous value from EEPROM, and select that bank. In your loop, check if the bank setting changed and write to EEPROM.
```cpp
uint8_t savedSetting;
void setup() {
  savedSetting = /* read from EEPROM */;
  if (savedSetting < bank.getNumberOfBanks())
    bank.select(savedSetting);
  else
    savedSetting = bank.getSelection();
  Control_Surface.begin();
}

void loop() {
  Control_Surface.loop();
  auto currSetting = bank.getSelection();
  if (savedSetting != currSetting) {
    /* write currSetting to EEPROM */
    savedSetting = currSetting;
  }
}
```

gringow831, 2022-03-02:
Wow that's amazing Pieter! Thank you so much! I will try making it work in a bit.
I am actually having a little issue with CABLE_x and BANK_RIGHT/BANK_LEFT.
It only affects the CABLE_1 and not the rest. Have been trying to work it out but can't understand why.
If I only have something like a PBPotentiometer with CABLE_2 and have a BANK_RIGHT it won't go right. Any ideas of why that could be?

tttapa, 2022-03-02:
> I am actually having a little issue with CABLE_x and BANK_RIGHT/BANK_LEFT.
> It only affects the CABLE_1 and not the rest. Have been trying to work it out but can't understand why.

Hard to say without seeing the code.

That said, BANK_RIGHT/LEFT is an MCU-specific feature, the selected bank is saved in the DAW software, not in the Arduino. I would expect that all MCU Extenders share the bank selection of the main unit, I don't think different extenders/cables can have different banks.

gringow831, 2022-03-02:
save_settings:19: error: expected primary-expression before ';' token
   savedSetting = /* read from EEPROM */;
                                        ^
save_settings.ino: At global scope:
save_settings:26: error: expected declaration before '}' token
 }
 ^
expected primary-expression before ';' token

----------------
I get the error if I leave   ` savedSetting = /* read from EEPROM */;` alone
I am assuming I need to write EEPROM.read(0); where it says read from EEPROM  and EEPROM.write at save to EEPROM?

I have tried that including adding the braces too but it doesn't read the bank.

There you can see what I have tried about EEPROM and also how the BANK_RIGHT/BANK_LEFT is written. Is there a way to specify the CABLE on the MCUBANK? MCU::BANK_RIGHT, 2?

```
#include <Control_Surface.h>
#include <EEPROM.h>

USBMIDI_Interface midi;

Bank<3> bank(2);
IncrementSelector<3> bankSelector = {bank, 5};

Bankable::ManyAddresses::PBPotentiometer<3> volumecontrol[] { 
  {bank, 14, {{{MCU::VOLUME_1, CABLE_2}, {MCU::VOLUME_3, CABLE_1}, {MCU::VOLUME_6, CABLE_1},}}},
  {bank, 15, {{{MCU::VOLUME_2, CABLE_2}, {MCU::VOLUME_4, CABLE_1}, {MCU::VOLUME_7, CABLE_1},}}},
  {bank, 16, {{{MCU::VOLUME_3, CABLE_2}, {MCU::VOLUME_5, CABLE_1}, {MCU::VOLUME_8, CABLE_1},}}},
};

  NoteButton noteButton[] = {
  { 7, MCU::BANK_LEFT,},
  { 8, MCU::BANK_RIGHT,},
  };

uint8_t savedSetting;

void setup() {
  savedSetting = EEPROM.read(1)/* read from EEPROM */;  //??
  if (savedSetting < bank.getNumberOfBanks()){
    bank.select(savedSetting);
  }else{
    savedSetting = bank.getSelection();
  };
  Control_Surface.begin();
}

void loop() {
  Control_Surface.loop();
  auto currSetting = bank.getSelection();
  if (savedSetting != currSetting) {
    /* write currSetting to EEPROM */
    EEPROM.write(savedSetting, 1); //??
    savedSetting = currSetting;
  }
}
```

tttapa, 2022-03-02:
> `savedSetting = EEPROM.read(1)/* read from EEPROM */;  //??`

You can delete the comment, it was just a placeholder: `savedSetting = EEPROM.read(1);`

> `EEPROM.write(savedSetting, 1);`

I think you meant to write: `EEPROM.write(currSetting, 1);`

> `and also how the BANK_RIGHT/BANK_LEFT is written` 

Written? Where to?

> Is there a way to specify the CABLE on the MCUBANK? MCU::BANK_RIGHT, 2?

Sure, see https://tttapa.github.io/Control-Surface-doc/Doxygen/d3/df7/midi-tutorial.html#midi_md-midi-addresses

```cpp
NoteButton noteButton[] {
  { 7, {MCU::BANK_LEFT, CABLE_2} },
  { 8, {MCU::BANK_RIGHT, CABLE_2} },
};
```

gringow831, 2022-03-02:
This is how I have it written to save to EEPROM and have checked changing bank, (not MCU BANK) and it's not being saved. If go to the second instance of many addresses that starts on VOLUME_3. disconnect the Teensy LC, connect it again and it's reverted to the VOLUME_1 of the first manyaddresses instance. I can try also with a Teensy 4.0 in case that works

On another news, I checked using CABLE on the BANK_RIGHT and in now I can change the bank. I wonder if there is a function I could use to actually change many cables at the same time withing BANK_RIGHT / BANK_LEFT

```
#include <Control_Surface.h>
#include <EEPROM.h>

USBMIDI_Interface midi;

Bank<3> bank(2);

IncrementSelector<3> bankSelector = {bank, 5};

Bankable::ManyAddresses::PBPotentiometer<3> volumecontrol[] { 
  {bank, 14, {{{MCU::VOLUME_1, CABLE_2}, {MCU::VOLUME_3, CABLE_1}, {MCU::VOLUME_6, CABLE_1},}}},
  {bank, 15, {{{MCU::VOLUME_2, CABLE_2}, {MCU::VOLUME_4, CABLE_1}, {MCU::VOLUME_7, CABLE_1},}}},
  {bank, 16, {{{MCU::VOLUME_3, CABLE_2}, {MCU::VOLUME_5, CABLE_1}, {MCU::VOLUME_8, CABLE_1},}}},
};

  NoteButton noteButton[] = {
  { 7, {MCU::BANK_RIGHT, CABLE_1}},
  { 8, {MCU::BANK_RIGHT, CABLE_2}},
  };

uint8_t savedSetting;

void setup() {
  savedSetting = EEPROM.read(1);
  if (savedSetting < bank.getNumberOfBanks()){
    bank.select(savedSetting);
  }else{
    savedSetting = bank.getSelection();
  };
  Control_Surface.begin();
}

void loop() {
  Control_Surface.loop();
  auto currSetting = bank.getSelection();
  if (savedSetting != currSetting) {
 EEPROM.write(currSetting, 1);
    savedSetting = currSetting;
  }
}

```

tttapa, 2022-03-02:
> EEPROM.write(currSetting, 1);

See https://www.arduino.cc/en/Reference/EEPROMWrite for the order of the arguments.

gringow831, 2022-03-02:
> > EEPROM.write(currSetting, 1);
> 
> See https://www.arduino.cc/en/Reference/EEPROMWrite for the order of the arguments.

So the address should be number EEPROM.write(address, value) and value the currSetting
I have tried address 0 and the value of currSetting but didn't manage to read it back on rebooting.
Do I need to use the EEPROM.update() instead of write so it saves the bank setting?

gringow831, 2022-03-02:
I see the actual problem I had with BANK_RIGHT / BANK_LEFT. 
I do use Control_Surface.sendNoteOn(MCU::BANK_RIGHT, 127); on the setup so it starts on the second bank but as sendNoteOn needs address and velocity I haven't been able to add CABLE_2  or CABLE_3 at the end yet

tttapa, 2022-03-02:
> Unless I am mistaken I just moved Control_Surface.begin(); before saveSetting = EEPROM.read(0); and it seems to have worked.

Indeed, my bad, Control_Surface.begin() initializes the banks to their default value as well.

> Do I need to use the EEPROM.update() instead of write 

No, write is fine because we check `savedSetting != currSetting` ourselves.

> is the right thing but
is there a way to write

You have to "release" the MCU bank button before you can press it again, i.e. every sendNoteOn should be followed by a sendNoteOff.

gringow831, 2022-03-02:
Thank you much Pieter! I figured that was the case

gringow831, 2022-03-04:
Is there any way to save make Custom-Selector-Callback be saved also onto EEPROM?

gringow831, 2022-03-05:
There is a strange behaviour when loading after eeprom. Let's say I have 3 ManyAddresses like in the code I posted.
If I am on the second address after loading and press the IncrementSelector button, the first press doesn't do anything. The second time starts changing.

If I have 5 bank addresses rather than 3and I am on the 3rd bank address, it goes back to the second one.

From 5 goes to 2. Do you know why is that happening?

closed 2022-03-07 by gringow831