tttapa/Control-Surface #717: IncrementSelectorLeds 4 LED, 5 Banks one Button. ON, on, on, all on, off

gringow831, 2022-03-03:
Hi Pieter, 

I was wondering if you can point me in the right direction.
Let's say I have 4 LED, 1 IncrementSelector for the Bank and 5 Addresses in a bank. 
I have been trying to make the first bank LED 1, second LED2, third LED3, fourth ALL LEDs (threeLEDs) and 5th bank, no LEDS (NOLEDs). 

I thought the IncrementSelectorLeds could be an option but I haven't managed to do all LED at the same time or no Leds. Also, as it saves the settings, even if it were only 3 LED one by one it doesn't remember the last bank selection. I am guessing I should use bank.select 

```
#include <Control_Surface.h>
#include <EEPROM.h>

USBMIDI_Interface midi;

Bank<5> bank(1);
IncrementSelector<5> bankSelector = {bank, 5};

int threeLEDs = 0 && 1 && 2;
int NOLEDs;

IncrementSelectorLEDs<5> bankSelectorLEDs {
  bank,
  {5},         // button pins
  {0, 1, 2, threeLEDs, NOLEDs}, // NOLEDS should be all off. 
};

Bankable::ManyAddresses::PBPotentiometer<5> volumecontrol[] { 
  {bank, 6, {{{MCU::VOLUME_1}, {MCU::VOLUME_2}, {MCU::VOLUME_4}, {MCU::VOLUME_7}, {MCU::VOLUME_8,}}},
};

uint8_t savedSetting;

void setup() {
    Control_Surface.begin();
  savedSetting = EEPROM.read(0);
  if (savedSetting < bank.getNumberOfBanks())
    bank.select(savedSetting);
  else
    savedSetting = bank.getSelection();

}

void loop() {
  Control_Surface.loop();
  auto currSetting = bank.getSelection();
  if (savedSetting != currSetting) {
    EEPROM.write(0,currSetting);
    savedSetting = currSetting;
  }
}
```

gringow831, 2022-03-04:
I have tried using the Custom-Selector-Callback example although even if I save this to a new address on EEPROM it doesn't boot up on the right LED choice. 
It boots as soon as setup begins at the moment. 

I know I could split the LEDs instead of using an RGB diode but I also wonder if I can have more than 3 LEDs with the option. 

How can I modify it so it brings bank the right LED on load and also have more than 3?

tttapa, 2022-03-04:
> fourth ALL LEDs (threeLEDs) and 5th bank, no LEDS (NOLEDs).

This is not supported by the `IncrementSelectorLEDs` class, so you'll indeed have to use a custom callback, or just turn on/off the LEDs in your loop based on `currSetting`.

> if I save this to a new address on EEPROM it doesn't boot up on the right LED choice.

You're only changing the setting of the bank, not the setting of the selector. You'll have to replace `bank.select(...)` by `bankSelector.set(...)`.

> I also wonder if I can have more than 3 LEDs with the option.

Sure, you can use any number of LEDs, just add more pins to the callback class in the example.

gringow831, 2022-03-04:
> This is not supported by the `IncrementSelectorLEDs` class, so you'll indeed have to use a custom callback, or just turn on/off the LEDs in your loop based on `currSetting`.
> 
> >
I didn't know I could turn on/off the LED in the loop based on `currSetting` 
Do I need to select the right pin digitalWrite(13, LOW); for example?

> You're only changing the setting of the bank, not the setting of the selector. You'll have to replace `bank.select(...)` by `bankSelector.set(...)`.

Sorry I don't follow I tried changing the "ledbank" for bank in the GenericIncrementSelector and call it with EEPROM but that of course didn't solve the problem.

This is what I get back when I try to compile 'class CS::IncrementSelector<5u>' has no member named 'select'

```
#include <Control_Surface.h>
#include <EEPROM.h>

USBMIDI_Interface midi;

class MySelectorCallback {
  public:
    // Constructor
    MySelectorCallback(pin_t Bank1LED, pin_t Bank2LED, pin_t Bank3LED, pin_t Bank4LED)
      : Bank1LED(Bank1LED), Bank2LED(Bank2LED), Bank3LED(Bank3LED), Bank4LED(Bank4LED) {}

    // Begin function is called once by Control Surface.
    // Use it to initialize everything.
 void begin() {
     pinMode(Bank1LED, OUTPUT);
     pinMode(Bank2LED, OUTPUT);
     pinMode(Bank3LED, OUTPUT);
     pinMode(Bank4LED, OUTPUT);
     
 }

    // Update function is called continuously by Control Surface.
    // Use it to implement things like fading, blinking ...

    void update() {}

    // Update function with arguments is called when the setting
    // changes.
    // Use it to update the LEDs.
    void update(setting_t oldSetting, setting_t newSetting) {
      (void) oldSetting; // unused in this example
      show(newSetting);
    }

  private:
    // Show the color of the given setting.
    void show(setting_t setting) {
      uint8_t color = getColor(setting);
      analogWrite(Bank1LED, color & 0b001 ? 8 : LOW);
      analogWrite(Bank2LED, color & 0b010 ? 8 : LOW);
      analogWrite(Bank3LED, color & 0b100 ? 8 : LOW);
      analogWrite(Bank4LED, color & 0b110 ? 8 : LOW);
    }

    // Convert the given setting to a 3-bit RGB color value.
    static uint8_t getColor(setting_t setting) {
      switch (setting) {
        case 0: return 0b0010; //red
        case 1: return 0b0101; //green
        case 2: return 0b0111; 
        case 3: return 0b1110;
        case 4: return 0b0000;
        default: return 0b000;
      }
    }

  private:
    // Member variables to remember the pin numbers of the LEDs.
    pin_t Bank1LED, Bank2LED, Bank3LED, Bank4LED;
};

Bank<5> bank(1);

Bank<5> ledbank = 1; //BANKS OF LEDS

// Create a selector that uses our custom callback, to control the bank.
GenericIncrementSelector<5, MySelectorCallback> ledselector {
  ledbank,         // bank to manage
  {0, 1, 2, 13}, // red, green, blue LED pins
  // (this is the MySelectorCallback constructor defined above)
  {3},       // incr/decr button pins
 };

IncrementSelector<5> bankSelector = {bank, 3};

Bankable::ManyAddresses::PBPotentiometer<5> volumecontrol[] { 
  {bank, 6, {{{MCU::VOLUME_1}, {MCU::VOLUME_2}, {MCU::VOLUME_4}, {MCU::VOLUME_7}, {MCU::VOLUME_8,}}}},
};

uint8_t savedSetting; //For bank
uint8_t savedSetting_1;  //For LED bank

void setup() {
Control_Surface.begin();
  //SAVING SETTINGS FOR BANK
  savedSetting = EEPROM.read(0);
  if (savedSetting < bank.getNumberOfBanks())
    bank.select(savedSetting);
  else
    savedSetting = bank.getSelection();

      //SAVING SETTINGS FOR BANK
  savedSetting_1 = EEPROM.read(0);
  if (savedSetting_1 < bank.getNumberOfBanks())
    bankSelector.select(savedSetting_1);
  else
    savedSetting_1 = bank.getSelection();

}

void loop(){
  
   //RECALLING SETTINGS BANK
  auto currSetting= ledbank.getSelection();
  if (savedSetting != currSetting) {
    EEPROM.write(1,currSetting);
    savedSetting = currSetting;

       //RECALLING SETTINGS BANK
  auto currSetting= ledbank.getSelection();
  if (savedSetting_1 != currSetting) {
    EEPROM.write(1,currSetting);
    savedSetting_1 = currSetting;
  }

  Control_Surface.loop();
}
```

gringow831, 2022-03-04:
As I believe I made a mistake earlier copying parts. this is what I have now. 

It does compile but it doesn't load the selected LED and goes back as it has never been changed.

```
Bank<5> ledbank = 1; //BANKS OF LEDS

uint8_t savedSetting_1; //for bank
void setup() {
Control_Surface.begin();

      //SAVING SETTINGS FOR BANK
  savedSetting_1 = EEPROM.read(1);
  if (savedSetting_1 < ledbank.getNumberOfBanks())
    ledbank.select(savedSetting_1);
  else
    savedSetting_1 = ledbank.getSelection();

}

void loop(){
   
   //RECALLING SETTINGS BANK
   auto currSetting_1 = ledbank.getSelection();
   if (savedSetting_1 != currSetting_1) {
   EEPROM.write(1, currSetting_1);
   savedSetting_1 = currSetting_1;
  }
Control_Surface.loop();
}
```

gringow831, 2022-03-05:
When you meant bankSelector.set(...)is bankSelector.select() right? I'm my case ledbank. 
For some reason it doesn't load back the selected light

closed 2022-03-06 by gringow831