tttapa/Control-Surface #738: MIDI over USB using ESP32-S2 or ESP32-S3?

IsThisNameGoodEnough, 2022-04-26:
Hi,
The new line of ESP32 chips support native USB.  Any plans on incorporating support for them?  Thanks!

tttapa, 2022-05-01:
I haven't had the time to check out these new chips. However, adding support yourself shouldn't be too hard*, you just have to implement four functions:

```cpp
#include <AH/Containers/Array.hpp>
#include <Settings/NamespaceSettings.hpp>

#include <MIDIUSB.h>

BEGIN_CS_NAMESPACE

struct USBDeviceMIDIBackend {
    using MIDIUSBPacket_t = AH::Array<uint8_t, 4>;
    /// Read a single 4-byte USB MIDI packet from the receive buffer.
    MIDIUSBPacket_t read() { /* ... */ }
    /// Place one 4-byte USB MIDI packet into the transmit buffer.
    void write(uint8_t cn_cin, uint8_t midi_0, uint8_t midi_1, uint8_t midi_2) { /* ... */ }
    /// Start sending out the transmit buffer.
    void sendNow() { /* ... */ }
    /// This function should return true if `sendNow()` should be called after each
    /// series of messages. It should return false if the backend can transmit the data
    /// asynchronously (in the background) without explicit calls to `sendNow()`.
    bool preferImmediateSend() { /* ... */ }
};

END_CS_NAMESPACE
```

You can find examples in https://github.com/tttapa/Control-Surface/tree/master/src/MIDI_Interfaces/USBMIDI

--- 

<sup>(*) Assuming you can find an existing MIDI USB example for these boards, or if you can easily copy the USB descriptors from another platform.</sup>

vespertine-pl, 2022-08-08:
Hi,

I'm trying to add USB support for ESP32-S3 with ESPTinyUSB,  but without success - I'm bit of noob in coding. Any chance for implementing support for ESP32-S3 in near future? It will be great to have both USB and BLE MIDI in one device. Thanks in advance!

tttapa, 2022-08-08:
This might be helpful if you didn't come across it yet: https://tttapa.github.io/Control-Surface-doc/Doxygen/d7/d5f/Custom-USB-MIDI-Backend_8ino-example.html  
It also contains a link to the TinyUSB implementation Control Surface uses on the Pi Pico (albeit using Adafruit's wrapper, but it shouldn't be too hard to reverse engineer that).

It's still on my todo list to check out some of these new ESP32 boards, but I'm afraid I simply don't have the time at the moment.

tttapa, 2022-08-09:
If you post your code and the problems/errors you encountered, I might be able to tell what's wrong, but I don't have the hardware or time to actually test it.

Spartacus1066, 2022-08-14:
I am also using the ESP32 S3, but not using the USB at all and trying to get this code to "pipe" Serial to BLE. It won't compile. Lots of errors related to USB (which I am not using)

For instance:
` /Arduino/libraries/MIDIUSB/src/MIDIUSB.h:18:2: error: #error MIDIUSB can only be used with an USB MCU.
 #error MIDIUSB can only be used with an USB MCU.`

My tiny code (mostly your great example! Thank you) is
```
#include <Control_Surface.h> // Include the Control Surface library

// Instantiate the two MIDI interfaces
BluetoothMIDI_Interface midiBLE;
HardwareSerialMIDI_Interface midi{Serial1, 31250};
 
// Instantiate a MIDI pipe to connect the two interfaces
BidirectionalMIDI_Pipe mpipe;
 
void setup() {
  // Create a bidirectional route between the interfaces: 
  // all input to one interface is sent to the output of the other
  midi | mpipe | midiBLE;
  // Initialize all MIDI interfaces
  MIDI_Interface::beginAll();
}
 
void loop() {
  // Continuously handle MIDI input
  MIDI_Interface::updateAll();
}`

tttapa, 2022-08-14:
I've just tried this using the `main` version of Control Surface, compiled for “ESP32S3 Dev Module” using version 2.0.4 of the ESP32 Arduino core, and it compiled without issues. Which settings are you using?

Control Surface only tries to include the MIDIUSB library if the `USBCON` macro is defined, which doesn't seem to be the case for the ESP32 core: https://github.com/espressif/arduino-esp32/search?q=USBCON

You can see the full logic here: https://github.com/tttapa/Control-Surface/blob/ebec32b97fd474ba21cb1e31b7a06ea7200233c1/src/MIDI_Interfaces/USBMIDI/USBMIDI.hpp#L36-L81

Spartacus1066, 2022-08-14:
> I've just tried this using the `main` version of Control Surface, compiled for “ESP32S3 Dev Module” using version 2.0.4 of the ESP32 Arduino core, and it compiled without issues. Which settings are you using?
> 

Thank you. I just got the full library again from GitHub and it is compiling.
I am using Adafruit's Feather ESP32-S3 with 2mb PSRAM. I can't explain why the library I downloaded 15 hours ago didn't work. I am sure that I put something in the wrong place or did something else causing the issue. haha

Thank you for this library!

Trickster-git, 2022-10-23:
Hi, can you help me?
I use esp32s3 and have trouble with #include <MIDIUSB.h> it sends errors

```
WARNING: library MIDIUSB claims to run on avr, sam, samd architecture(s) and may be incompatible with your current board which runs on esp32 architecture(s).
In file included from  Documents\Arduino\MCU-OLED-SSD1306-x2\MCU-OLED-SSD1306-x2.ino:9:
 Documents\Arduino\libraries\MIDIUSB\src/MIDIUSB.h:18:2: error: #error MIDIUSB can only be used with an USB MCU.
 #error MIDIUSB can only be used with an USB MCU.
  ^~~~~
In file included from  Documents\Arduino\MCU-OLED-SSD1306-x2\MCU-OLED-SSD1306-x2.ino:9:
 Documents\Arduino\libraries\MIDIUSB\src/MIDIUSB.h:78:2: error: #error "Unsupported architecture"
 #error "Unsupported architecture"
  ^~~~~
```
which library for usbmidi should i use?

tttapa, 2022-10-23:
The ESP-IDF comes with TinyUSB, there's a MIDI example that you should be able to adapt to the `USBDeviceMIDIBackend` shown above.

https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/device/tusb_midi

Trickster-git, 2022-10-23:
Oookay, right now I understand, it is impossible for my coding skill.
Can someone help?

tttapa, 2025-08-14:
For future reference: ESP32-S3 USB MIDI support was added by a2e4b957ca314f4b206421ef792df0a55a947394, which is included in the 2.0.0 release of Control Surface.