tttapa/Control-Surface #752: Raspberry pi RP2040 support

AdamBenoit, 2022-05-27:
First off, great library. :)

I am trying to compile for a RP2040 device but get stuck at what I believe is the fact that this device is not in defined anywhere.

The error is: 

.pio\libdeps\nanorp2040connect\Control-Surface-master\src/Submodules/Encoder/Encoder.h:42:30: error: 'CORE_NUM_INTERRUPT' was not declared in this scope; did you mean 'NOT_AN_INTERRUPT'?

How do I define this board correctly? I added the following to end of interrupt_pins.h for now but its not likely the correct way:

#else
  #define CORE_NUM_INTERRUPT	NUM_DIGITAL_PINS

This is currently not tested, but i was able to get it to build.

Thanks,

Adam

tttapa, 2022-05-28:
Have a look at the [`pico`](https://github.com/tttapa/Control-Surface/tree/pico) branch. It definitely works with the official Arduino core (https://github.com/arduino/ArduinoCore-mbed), but it should work with earlephilhower's core as well.

Veroledez, 2022-05-28:
> Echa un vistazo a la [`pico`](https://github.com/tttapa/Control-Surface/tree/pico)sucursal. Definitivamente funciona con el núcleo oficial de Arduino ( https://github.com/arduino/ArduinoCore-mbed ), pero también debería funcionar con el núcleo de earlephilhower.

Hola, me alegra saber que una rama de esta gran libreria ya este soportando pi pico, seria genial si se pudiera crear cables virtuales al estilo teensy.
Gracias infinitas  por todo tu trabajo.

tttapa, 2022-05-28:
The `pico` branch has been merged into `master` now.

> seria genial si se pudiera crear cables virtuales al estilo teensy.

This is just a matter of changing the descriptors:

https://github.com/tttapa/Control-Surface/blob/08b36b598fc554ff05d725def3c52808c9a680ef/src/MIDI_Interfaces/USBMIDI/mbed-m0/PluggableUSBMIDI-descr.cpp#L39-L98

The Adafruit_TinyUSB library has a `setCables()` method to change the number of cables. See https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/366da4de614cb08daf93fe8b31e94918e7077d9c/src/arduino/midi/Adafruit_USBD_MIDI.cpp#L82-L146.

You can access it from Control Surface using `USBMIDI_Interface::backend.backend.setCables()`. For example:
```cpp
USBMIDI_Interface midi;
void setup() {
  midi.backend.backend.setCables(2);
  Control_Surface.begin();
}
```
You have to set the number of cables _before_ calling `Control_Surface.begin()`.

AdamBenoit, 2022-05-28:
You are amazing. This is well above what I was hoping for. 

I am using the earlephilhower boards but have yet figured out the configuration of the one I have. 

If I have any issues I will reach out again. Thank you!

closed 2022-05-28 by AdamBenoit

Veroledez, 2022-05-30:
> La `pico`rama se ha fusionado con `master`ahora.
> 
> > seria genial si se pudiera crear cables virtuales al estilo teensy.
> 
> Esto es solo cuestión de cambiar los descriptores:
> 
> https://github.com/tttapa/Control-Surface/blob/08b36b598fc554ff05d725def3c52808c9a680ef/src/MIDI_Interfaces/USBMIDI/mbed-m0/PluggableUSBMIDI-descr.cpp#L39-L98
> 
> Si está utilizando el núcleo de earlephilhower, entonces la biblioteca Adafruit_TinyUSB tiene un `setCables()`método para cambiar la cantidad de cables. Consulte https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/366da4de614cb08daf93fe8b31e94918e7077d9c/src/arduino/midi/Adafruit_USBD_MIDI.cpp#L82-L146 .
> 
> Puede acceder a él desde Control Surface usando `USBMIDI_Interface::backend.backend.setCables()`. Por ejemplo:
> 
> ```c++
> USBMIDI_Interface midi;
> void setup() {
>   midi.backend.backend.setCables(2);
>   Control_Surface.begin();
> }
> ```
> 
> Tienes que configurar el número de cables _antes_ de llamar `Control_Surface.begin()`.

Hola, si se desea crear cables virtuales via descriptores,  que se debe cambiar en el archivo que citas en tu respuesta, hay alguna variable o se debe copiar dos veces el codigo   para dos cables por ejemplo.
Muchas gracias

tttapa, 2022-05-31:
> si se desea crear cables virtuales via descriptores, que se debe cambiar en el archivo que citas en tu respuesta

USB uses descriptors for each aspect of a device. These descriptors are just a sequence of bytes, where the first byte indicates the length of the descriptor, and the second byte indicates what type of descriptor it is. 

The full descriptor for a MIDI device consists of the configuration descriptor, followed by Interface descriptors, and finally Endpoint descriptors. Each Interface descriptor describes an interface of a certain class (e.g. Audio or MIDI), and is then followed by a class-specific header interface, and then optionally more class-specific interfaces.

For example, for MIDI, you need have an AudioControl Interface, including just the AudioControl Header.  
Then you also need a MIDIStreaming Interface, which includes a Header, followed by MIDI Jacks (these Jacks are the virtual cables).

To change the number of virtual cables, you have to add more Jacks to the MIDIStreaming Interface descriptor (and then also update the header because the size changes when you add Jacks).

Apart from the Interfaces, you also need Endpoint descriptors. Endpoints are the components that are actually responsible for sending and receiving USB data. The Jacks are just conceptual, they don't do anything unless connected to an Endpoint.

You need a descriptor for the Endpoint itself (to define the packet size etc.), and then a class-specific MIDIStreaming Endpoint Descriptor to associate Jacks with the Endpoint.  
If you change the number of Jacks, you also have to edit this class-specific descriptor to include the Jacks you added.

If you look at the array of bytes in my previous post, you'll see the different descriptors:
- L50: A configuration descriptor, which I won't go into, and which is largely ignored by Arduino's PluggableUSB framework, but you will have to update the size.
- L53: An AudioControl Interface Descriptor 
- L54: its corresponding class-specific header descriptor

---

- L55: The MIDIStreaming Interface Descriptor
- L56: the corresponding class-specific header descriptor
- L59: the first MIDI IN Jack (Virtual Cable 1)
- L60: an external MIDI IN Jack
- You would add more Jacks after this
- L63: the first MIDI OUT Jack (Virtual Cable 1), connected to the external MIDI IN Jack
- L64: an external MIDI OUT Jack, connected the the first MIDI IN Jack
- You would add more Jacks after this

---

- L67-L75: the OUT endpoint descriptor
- L77: The class-specific descriptor that connects the first MIDI OUT Jack to this endpoint, you would include the Jacks you added to it as well (don't forget to change the lengths)
- L80-L88: the IN endpoint descriptor
- L90: The class-specific descriptor that connects the first MIDI IN Jack to this endpoint, again, include the Jacks you added and update the lengths.

The descriptor used in Control Surface is almost exactly the same as the example in Appendix B of <https://www.usb.org/sites/default/files/midi10.pdf>. I've added some comments in the code below, so you can match the following to the tables in Appendix B.  
Adding the Jacks/Cables is pretty easy, you just have to remember to give each Jack its own ID and to update all sizes correctly once you're done. 

Specifically, you have to update: `wTotalLength` in the Configuration Descriptor, `wTotalLength` in the Class-specific MS Interface Header Descriptor, `bLength` and `bNumEmbMIDIJack` in the Class-specific MS Endpoint Descriptors.

In detail:
```cpp
    uint8_t config_descriptor_temp[] {
        // configuration descriptor
        0x09, 0x02, 0x65, 0x00, 0x02, 0x01, 0x00, 0xc0, 0x50,

        // The Audio Interface Collection
        0x09, 0x04, 0x00, 0x00, 0x00, 0x01, 0x01, 0x00, 0x00, // Standard AC Interface Descriptor
        0x09, 0x24, 0x01, 0x00, 0x01, 0x09, 0x00, 0x01, 0x01, // Class-specific AC Interface Descriptor
        0x09, 0x04, 0x01, 0x00, 0x02, 0x01, 0x03, 0x00, 0x00, // MIDIStreaming Interface Descriptors
        0x07, 0x24, 0x01, 0x00, 0x01, 0x41, 0x00,             // Class-Specific MS Interface Header Descriptor

        // MIDI IN JACKS
        0x06, 0x24, 0x02, // Class-specific MS MIDI IN Jack descriptor
        0x01, 0x01, 0x00, // Embedded, ID 0x01, no string descriptor

        0x06, 0x24, 0x02, // Class-specific MS MIDI IN Jack descriptor
        0x02, 0x02, 0x00, // External, ID 0x02

        // MIDI OUT JACKS
        0x09, 0x24, 0x03, // Class-specific MS MIDI OUT Jack descriptor
        0x01, 0x03, 0x01, // Embedded, ID 0x03, 1 input pin
        0x02, 0x01, 0x00, // Source Jack 0x02, source pin 0x01, no string descriptor

        0x09, 0x24, 0x03, // Class-specific MS MIDI OUT Jack descriptor
        0x02, 0x06, 0x01, // External, ID 0x06, 1 input pin
[... 9 lines cut ...]
        0x00,         // bInterval (milliseconds)
        0x00,         // bRefresh
        0x00,         // bSynchAddress

        0x05, 0x25, 0x01, // Class-specific MS Bulk OUT Endpoint descriptor
        0x01, 0x01,       // 1 embedded MIDI OUT Jack with ID 0x01

        // IN endpoint - Standard MS Bulk Data Endpoint Descriptor
        0x09,        // bLength
        0x05,        // bDescriptorType
        bulk_in_ep,  // bEndpointAddress
        0x02,        // bmAttributes
        wmaxpkt_L,   // wMaxPacketSize (LSB)
        wmaxpkt_M,   // wMaxPacketSize (MSB)
        0x00,        // bInterval (milliseconds)
        0x00,        // bRefresh
        0x00,        // bSynchAddress

        0x05, 0x25, 0x01, // Class-specific MS Bulk OUT Endpoint descriptor
        0x01, 0x03,       // 1 embedded MIDI IN Jack with ID 0x03
    };
```
In the library, you also have to increase the number of cables:

https://github.com/tttapa/Control-Surface/blob/6aa44d1045a84950f1b3df987f8719c0bee90289/src/MIDI_Parsers/USBMIDI_Parser.hpp#L16

tttapa, 2022-05-31:
Some examples:
```cpp
// 2 virtual cables
constexpr static uint8_t config_descriptor[133] {
  0x09, 0x02, 0x85, 0x00, 0x02, 0x01, 0x00, 0x80, 0x32, 0x09, 0x04, 0x00, 0x00, 0x00, 0x01, 0x01, 0x00, 0x00, 0x09, 0x24, 0x01, 0x00, 0x01, 0x09, 0x00, 0x01, 0x01, 0x09, 0x04, 0x01, 0x00, 0x02, 0x01, 0x03, 0x00, 0x00, 0x07, 0x24, 0x01, 0x00, 0x01, 0x43, 0x00, 0x06, 0x24, 0x02, 0x01, 0x01, 0x00, 0x06, 0x24, 0x02, 0x01, 0x03, 0x00, 0x06, 0x24, 0x02, 0x02, 0x02, 0x00, 0x06, 0x24, 0x02, 0x02, 0x04, 0x00, 0x09, 0x24, 0x03, 0x01, 0x05, 0x01, 0x02, 0x01, 0x00, 0x09, 0x24, 0x03, 0x01, 0x07, 0x01, 0x04, 0x01, 0x00, 0x09, 0x24, 0x03, 0x02, 0x06, 0x01, 0x01, 0x01, 0x00, 0x09, 0x24, 0x03, 0x02, 0x08, 0x01, 0x03, 0x01, 0x00, 0x09, 0x05, 0x01, 0x02, 0x40, 0x00, 0x00, 0x00, 0x00, 0x06, 0x25, 0x01, 0x02, 0x05, 0x07, 0x09, 0x05, 0x81, 0x02, 0x40, 0x00, 0x00, 0x00, 0x00, 0x06, 0x25, 0x01, 0x02, 0x01, 0x03,
};
```
```cpp
// 4 virtual cables
constexpr static uint8_t config_descriptor[197] {
  0x09, 0x02, 0xC5, 0x00, 0x02, 0x01, 0x00, 0x80, 0x32, 0x09, 0x04, 0x00, 0x00, 0x00, 0x01, 0x01, 0x00, 0x00, 0x09, 0x24, 0x01, 0x00, 0x01, 0x09, 0x00, 0x01, 0x01, 0x09, 0x04, 0x01, 0x00, 0x02, 0x01, 0x03, 0x00, 0x00, 0x07, 0x24, 0x01, 0x00, 0x01, 0x7F, 0x00, 0x06, 0x24, 0x02, 0x01, 0x01, 0x00, 0x06, 0x24, 0x02, 0x01, 0x03, 0x00, 0x06, 0x24, 0x02, 0x01, 0x05, 0x00, 0x06, 0x24, 0x02, 0x01, 0x07, 0x00, 0x06, 0x24, 0x02, 0x02, 0x02, 0x00, 0x06, 0x24, 0x02, 0x02, 0x04, 0x00, 0x06, 0x24, 0x02, 0x02, 0x06, 0x00, 0x06, 0x24, 0x02, 0x02, 0x08, 0x00, 0x09, 0x24, 0x03, 0x01, 0x09, 0x01, 0x02, 0x01, 0x00, 0x09, 0x24, 0x03, 0x01, 0x0B, 0x01, 0x04, 0x01, 0x00, 0x09, 0x24, 0x03, 0x01, 0x0D, 0x01, 0x06, 0x01, 0x00, 0x09, 0x24, 0x03, 0x01, 0x0F, 0x01, 0x08, 0x01, 0x00, 0x09, 0x24, 0x03, 0x02, 0x0A, 0x01, 0x01, 0x01, 0x00, 0x09, 0x24, 0x03, 0x02, 0x0C, 0x01, 0x03, 0x01, 0x00, 0x09, 0x24, 0x03, 0x02, 0x0E, 0x01, 0x05, 0x01, 0x00, 0x09, 0x24, 0x03, 0x02, 0x10, 0x01, 0x07, 0x01, 0x00, 0x09, 0x05, 0x01, 0x02, 0x40, 0x00, 0x00, 0x00, 0x00, 0x08, 0x25, 0x01, 0x04, 0x09, 0x0B, 0x0D, 0x0F, 0x09, 0x05, 0x81, 0x02, 0x40, 0x00, 0x00, 0x00, 0x00, 0x08, 0x25, 0x01, 0x04, 0x01, 0x03, 0x05, 0x07,
};
```
```cpp
// 8 virtual cables
constexpr static uint8_t config_descriptor[325] {
  0x09, 0x02, 0x45, 0x01, 0x02, 0x01, 0x00, 0x80, 0x32, 0x09, 0x04, 0x00, 0x00, 0x00, 0x01, 0x01, 0x00, 0x00, 0x09, 0x24, 0x01, 0x00, 0x01, 0x09, 0x00, 0x01, 0x01, 0x09, 0x04, 0x01, 0x00, 0x02, 0x01, 0x03, 0x00, 0x00, 0x07, 0x24, 0x01, 0x00, 0x01, 0xF7, 0x00, 0x06, 0x24, 0x02, 0x01, 0x01, 0x00, 0x06, 0x24, 0x02, 0x01, 0x03, 0x00, 0x06, 0x24, 0x02, 0x01, 0x05, 0x00, 0x06, 0x24, 0x02, 0x01, 0x07, 0x00, 0x06, 0x24, 0x02, 0x01, 0x09, 0x00, 0x06, 0x24, 0x02, 0x01, 0x0B, 0x00, 0x06, 0x24, 0x02, 0x01, 0x0D, 0x00, 0x06, 0x24, 0x02, 0x01, 0x0F, 0x00, 0x06, 0x24, 0x02, 0x02, 0x02, 0x00, 0x06, 0x24, 0x02, 0x02, 0x04, 0x00, 0x06, 0x24, 0x02, 0x02, 0x06, 0x00, 0x06, 0x24, 0x02, 0x02, 0x08, 0x00, 0x06, 0x24, 0x02, 0x02, 0x0A, 0x00, 0x06, 0x24, 0x02, 0x02, 0x0C, 0x00, 0x06, 0x24, 0x02, 0x02, 0x0E, 0x00, 0x06, 0x24, 0x02, 0x02, 0x10, 0x00, 0x09, 0x24, 0x03, 0x01, 0x11, 0x01, 0x02, 0x01, 0x00, 0x09, 0x24, 0x03, 0x01, 0x13, 0x01, 0x04, 0x01, 0x00, 0x09, 0x24, 0x03, 0x01, 0x15, 0x01, 0x06, 0x01, 0x00, 0x09, 0x24, 0x03, 0x01, 0x17, 0x01, 0x08, 0x01, 0x00, 0x09, 0x24, 0x03, 0x01, 0x19, 0x01, 0x0A, 0x01, 0x00, 0x09, 0x24, 0x03, 0x01, 0x1B, 0x01, 0x0C, 0x01, 0x00, 0x09, 0x24, 0x03, 0x01, 0x1D, 0x01, 0x0E, 0x01, 0x00, 0x09, 0x24, 0x03, 0x01, 0x1F, 0x01, 0x10, 0x01, 0x00, 0x09, 0x24, 0x03, 0x02, 0x12, 0x01, 0x01, 0x01, 0x00, 0x09, 0x24, 0x03, 0x02, 0x14, 0x01, 0x03, 0x01, 0x00, 0x09, 0x24, 0x03, 0x02, 0x16, 0x01, 0x05, 0x01, 0x00, 0x09, 0x24, 0x03, 0x02, 0x18, 0x01, 0x07, 0x01, 0x00, 0x09, 0x24, 0x03, 0x02, 0x1A, 0x01, 0x09, 0x01, 0x00, 0x09, 0x24, 0x03, 0x02, 0x1C, 0x01, 0x0B, 0x01, 0x00, 0x09, 0x24, 0x03, 0x02, 0x1E, 0x01, 0x0D, 0x01, 0x00, 0x09, 0x24, 0x03, 0x02, 0x20, 0x01, 0x0F, 0x01, 0x00, 0x09, 0x05, 0x01, 0x02, 0x40, 0x00, 0x00, 0x00, 0x00, 0x0C, 0x25, 0x01, 0x08, 0x11, 0x13, 0x15, 0x17, 0x19, 0x1B, 0x1D, 0x1F, 0x09, 0x05, 0x81, 0x02, 0x40, 0x00, 0x00, 0x00, 0x00, 0x0C, 0x25, 0x01, 0x08, 0x01, 0x03, 0x05, 0x07, 0x09, 0x0B, 0x0D, 0x0F,
};
```
```cpp
// 16 virtual cables
constexpr static uint8_t config_descriptor[581] {
  0x09, 0x02, 0x45, 0x02, 0x02, 0x01, 0x00, 0x80, 0x32, 0x09, 0x04, 0x00, 0x00, 0x00, 0x01, 0x01, 0x00, 0x00, 0x09, 0x24, 0x01, 0x00, 0x01, 0x09, 0x00, 0x01, 0x01, 0x09, 0x04, 0x01, 0x00, 0x02, 0x01, 0x03, 0x00, 0x00, 0x07, 0x24, 0x01, 0x00, 0x01, 0xE7, 0x01, 0x06, 0x24, 0x02, 0x01, 0x01, 0x00, 0x06, 0x24, 0x02, 0x01, 0x03, 0x00, 0x06, 0x24, 0x02, 0x01, 0x05, 0x00, 0x06, 0x24, 0x02, 0x01, 0x07, 0x00, 0x06, 0x24, 0x02, 0x01, 0x09, 0x00, 0x06, 0x24, 0x02, 0x01, 0x0B, 0x00, 0x06, 0x24, 0x02, 0x01, 0x0D, 0x00, 0x06, 0x24, 0x02, 0x01, 0x0F, 0x00, 0x06, 0x24, 0x02, 0x01, 0x11, 0x00, 0x06, 0x24, 0x02, 0x01, 0x13, 0x00, 0x06, 0x24, 0x02, 0x01, 0x15, 0x00, 0x06, 0x24, 0x02, 0x01, 0x17, 0x00, 0x06, 0x24, 0x02, 0x01, 0x19, 0x00, 0x06, 0x24, 0x02, 0x01, 0x1B, 0x00, 0x06, 0x24, 0x02, 0x01, 0x1D, 0x00, 0x06, 0x24, 0x02, 0x01, 0x1F, 0x00, 0x06, 0x24, 0x02, 0x02, 0x02, 0x00, 0x06, 0x24, 0x02, 0x02, 0x04, 0x00, 0x06, 0x24, 0x02, 0x02, 0x06, 0x00, 0x06, 0x24, 0x02, 0x02, 0x08, 0x00, 0x06, 0x24, 0x02, 0x02, 0x0A, 0x00, 0x06, 0x24, 0x02, 0x02, 0x0C, 0x00, 0x06, 0x24, 0x02, 0x02, 0x0E, 0x00, 0x06, 0x24, 0x02, 0x02, 0x10, 0x00, 0x06, 0x24, 0x02, 0x02, 0x12, 0x00, 0x06, 0x24, 0x02, 0x02, 0x14, 0x00, 0x06, 0x24, 0x02, 0x02, [... cut ...] x25, 0x27, 0x29, 0x2B, 0x2D, 0x2F, 0x31, 0x33, 0x35, 0x37, 0x39, 0x3B, 0x3D, 0x3F, 0x09, 0x05, 0x81, 0x02, 0x40, 0x00, 0x00, 0x00, 0x00, 0x14, 0x25, 0x01, 0x10, 0x01, 0x03, 0x05, 0x07, 0x09, 0x0B, 0x0D, 0x0F, 0x11, 0x13, 0x15, 0x17, 0x19, 0x1B, 0x1D, 0x1F,
};
```
The 16-cable one seems to result in a hardfault, I'm not quite sure why, maybe the hardware and/or software doesn't support such long descriptors. I don't have much time to look into it, and I don't have a debugger handy to investigate.