tttapa/Control-Surface #755: Setting ССPotentiometer for Joystick Module

Wallirec, 2022-06-05:
Hello, thanks for the great library! Can you please tell me if it is possible to configure the CCPotentiometer in my sketch so that it sends CC1 with the value 0 (now 62) in the middle position and increases the value to 127 in the extreme left and right positions?
`
#include <Control_Surface.h> // Include the Control Surface library

// Instantiate a MIDI over USB interface.
USBMIDI_Interface midi;

// Instantiate a PBPotentiometer object
PBPotentiometer potentiometer {
  A1,        // Analog pin connected to potentiometer
  CHANNEL_1, // MIDI Channel 1
};
// Instantiate a CCPotentiometer object
CCPotentiometer potentiometer2 {
  A0, {MIDI_CC::Modulation_Wheel, CHANNEL_1}
};

// Instantiate a CCButton object
CCButton button {
  // Push button on pin 5:
  5,
  // Damper_Pedal on MIDI channel 1:
  {MIDI_CC::Damper_Pedal, CHANNEL_1},
};

void setup() {
  Control_Surface.begin(); // Initialize Control Surface
}

void loop() {
  Control_Surface.loop(); // Update the Control Surface
}`
Thank you!

Wallirec, 2022-06-05:
I seem to understand that the best solution would be the work of the CC Potentiometer at the touch of a button, tell me how to do it please!

tttapa, 2022-06-05:
You can use the `map()` function: https://tttapa.github.io/Control-Surface-doc/Doxygen/d3/d26/CCPotentiometer-Map_8ino-example.html

Wallirec, 2022-06-05:
Thank you! Your sketch works as it should!

closed 2022-06-05 by Wallirec