tttapa/Control-Surface #761: Do you have schematic for midi 5 pin with node mcu v3 (8266)?

e36910, 2022-06-15:
How to solder midi 5 pin to arduino?

e36910, 2022-06-16:
I found this pattern. Does she fit that project? If suitable, which pin on the Arduino 8266 board to use for midi data? 
[image]

tttapa, 2022-06-16:
No, the only officially correct schematic is the one I posted a link to earlier.

The pin you should use for the output is the UART TX pin. But as mentioned earlier, you probably want a 5V buffer in between.

e36910, 2022-06-16:
I don't understand wich pin I need used in my board?
2=GDN
4=R200+Vin
5=R220+TX ?

e36910, 2022-06-25:
This is correct schematic for midi 5 pin. In this example we need used - USBMIDI_Interface midi;
It does work.

closed 2022-06-25 by e36910