tttapa/Control-Surface #764: How to properly use the code for HairlessMIDI and midi 5 pin togeather?

e36910, 2022-06-25:
I have 2 exaple 
HairlessMIDI_Interface midi; // for usb with HairlessMIDI app
and
USBMIDI_Interface midi; // for midi 5 pin

Separately it works, but when both lines are left, then there is a conflict. What is the right way to write this part of the code so that 2 functions work together?

tttapa, 2022-06-25:
This depends on the board you're using. On boards without native USB support, both the Hairless and the USBMIDI interfaces use the same `Serial` UART by default.

>  USBMIDI_Interface midi; // for midi 5 pin

This doesn't make much sense, if you want an interface for 5-pin MIDI, use the `HardwareSerialMIDI_Interface` (or any of the other available Serial interfaces: https://tttapa.github.io/Control-Surface-doc/Doxygen/dc/df0/group__MIDIInterfaces.html).

If both work separately, that probably means that you connected your 5-pin MIDI connector to the same UART that is also connected to the on-board UART-to-USB adapter. You cannot use that one UART for both purposes at the same time, you'll have to use a different UART for 5-pin MIDI.

e36910, 2022-06-25:
I have Node mcu V3 8266. I would used one device for PC via USB and for Helix Rack via 5pin midi out.

tttapa, 2022-06-25:
Then you'll have to use UART1 for 5-pin DIN and UART0 for Hairless.

e36910, 2022-06-26:
How to use it?

tttapa, 2022-06-26:
Use Serial1: https://arduino-esp8266.readthedocs.io/en/latest/reference.html#serial

e36910, 2022-06-27:
OK thanks. How to correctly write line UART0 for usb and UART1 for midi5 din?

tttapa, 2022-07-01:
```cpp
HardwareSerialMIDI_Interface my_midi_uart0 {Serial};
HardwareSerialMIDI_Interface my_midi_uart1 {Serial1};
```

Renlong2020, 2024-11-07:
For specific code examples, I use arduino pro micro board to use SoftwareSerialMIDI_Interface. In order to ensure that the upload program does not occupy tx rx, I use 8 and 9 as rx and tx. But how does the code replace usbinterface? Hope to have concrete code examples