tttapa/Control-Surface #787: transposer not working

latimarkim, 2022-08-27:
Hello. I uploaded ‘transposer.ino’ example to my arduino micro as it is except changing the pin number. However, transposer function didn’t work. On the other hand, every buttons for playing notes were working well. I tried ‘encoder selector’ with rotary encoder to transpose before but it sometimes worked, sometimes didn’t. My buttons including transposing ones were physically ok (I checked with my button checking code). I really wonder what is wrong. I put 1k resistors each to pull my buttons down, can it be a problem??

latimarkim, 2022-08-27:
my own code is this;

//Pin Number
// toggle 1-6: 2-7, toggle 7-12: A0-A5, button: 9, 10 (transpose) 11 (led on/off), LED: 12

//libraries
#include <Control_Surface.h> // Include the Control Surface library
#include <Adafruit_NeoPixel.h>
#ifdef __AVR__
#include <avr/power.h>

//LED
#endif
#define LED_button 11
#define LED_pin 12
#define LED_COUNT 8
boolean current_button = LOW;
boolean last_button = LOW;
boolean LED_on = false;
Adafruit_NeoPixel strip(LED_COUNT, LED_pin, NEO_GRB + NEO_KHZ800);

//Toggle
#define toggle_number     12
int t_active[toggle_number];
int t_pins[] = {2, 3, 4, 5, 6, 7, A0, A1, A2, A3, A4, A5};

// Instantiate a MIDI over USB interface.
USBMIDI_Interface midi;
using namespace MIDI_Notes;

//Transpose (button1: +1 octave, button2: -1 octave, together: reset)
Transposer < -5, +5 > transposer(12);

IncrementDecrementSelector<transposer.getNumberOfBanks()> selector {
  transposer,
  {9, 10},
  Wrap::Clamp,
};

//Toggle Note
Bankable::NoteButton buttons[] {
  {transposer, 2, MIDI_Notes::C(5)}, {transposer, 3, MIDI_Notes::Db(5)},
  {transposer, 4, MIDI_Notes::D(5)}, {transposer, 5, MIDI_Notes::Eb(5)},
  {transposer, 6, MIDI_Notes::E(5)}, {transposer, 7, MIDI_Notes::F_(5)},
  {transposer, A0, MIDI_Notes::Gb(5)}, {transposer, A1, MIDI_Notes::G(5)},
  {transposer, A2, MIDI_Notes::Ab(5)}, {transposer, A3, MIDI_Notes::A(5)},
  {transposer, A4, MIDI_Notes::Bb(5)}, {transposer, A5, MIDI_Notes::B(5)},
};

void setup() {
  Control_Surface.begin(); // Initialize Control Surface

  //LED
  pinMode(LED_pin, INPUT_PULLUP);
  strip.begin();
  strip.show();
}

void loop() {
  Control_Surface.loop(); // Update the Control Surface

  // Check for Keypress
  current_button = digitalRead(LED_button);
  delay (100);
  if (last_button == LOW && current_button == HIGH) {
    LED_on = !LED_on;
  }
  last_button = current_button;

  //LED
  if (LED_on) {
    int r = random(1, 255);
    int g = random(1, 255);
    int b = random(1, 255);
    int location = random(0, 7);
    uint32_t c = strip.Color(r, g, b);
    strip.setBrightness(127);

    //led function
    for (int i = 0; i < toggle_number; i++) {
      t_active[i] = bool(analogRead(t_pins[i]));
      if (t_active[i]) {
        strip.setPixelColor(location, c);
        strip.fill (c, location, 2);
        strip.show();
      }
    }
  }
  else {
    strip.fill(0);
    strip.show();
  }
}

tttapa, 2022-08-28:
Please open a single thread for each issue.

For future reference, see https://github.com/tttapa/Control-Surface/discussions/786