tttapa/Control-Surface #795: MIDI_Pipes-Routing V2 request

Petrus125, 2022-09-12:
Hello Pieter, i'm working on a kind of midi-thru project so that incoming midi data from serial 1 can be sent to serial2 or v.v and also to USB midi bus.. Now the question is if virtual serial port are compatible and how to implement. The example of MIDI_Pipes-Routing would be great if a virtual serial 2 can be added and  re-pipped. Probably a Teensy with host capabilities is the way to go, but far more expensive and powerfull for the needs.
An example would on how to build this would be great. 
Greets

[image]

Veroledez, 2022-09-12:
Hola, solo tienes que crear todas las istancias midi tanto seriales como la usb, luego haces las conexiones en el setup  y como quieres y listo

Petrus125, 2022-09-13:
> Hola, solo tienes que crear todas las istancias midi tanto seriales como la usb, luego haces las conexiones en el setup y como quieres y listo

Thank you for your reply. SAure has to be simple but I just  could not make it. Can you help me on this ?