tttapa/Control-Surface #809: switching programs on two buttons

Wallirec, 2022-10-01:
Hello! Tell me how to switch programs on two buttons, I can’t find an example.
The first button increases the program number and the second decreases it.

tttapa, 2022-10-01:
The following resources should help:
- [FAQ:  Can I change the address, controller number, MIDI channel, etc. at runtime?](https://tttapa.github.io/Control-Surface-doc/Doxygen/da/dc1/FAQ.html#faq-change-address-runtime)
- [PCButton](https://tttapa.github.io/Control-Surface-doc/Doxygen/d6/d5c/classBankable_1_1PCButton.html)
- [ManyAddressesPCButton.ino](https://tttapa.github.io/Control-Surface-doc/Doxygen/dc/db4/ManyAddressesPCButton_8ino-example.html)
- [IncrementDecrementSelector](https://tttapa.github.io/Control-Surface-doc/Doxygen/d2/deb/classIncrementDecrementSelector.html)

Wallirec, 2022-10-02:
Yes, thanks, I found all these links myself. There is switching banks, binding the program number to the button ...
But I did not find (not enough knowledge) how to switch programs with two buttons.
My midi keyboard has no control buttons and I need to switch programs in the Raspberry sampler.