tttapa/Control-Surface #810: Error

alan27tom, 2022-10-05:
Hello guys
I'm new to arduino, when I try to add the library (control surface) this error appears

Error: Request installZip failed with message: 13 INTERNAL: Library install failed: extracting archive: Create link C:\Users\Cultura no Morro\AppData\Local\Temp\arduino-cli-lib-2603712153\Control-Surface-main\doxygen\pages\xbm.md: symlink ....\src\Display\Bitmaps\README.md C:\Users\Cultura no Morro\AppData\Local\Temp\arduino-cli-lib-2603712153\Control-Surface-main\doxygen\pages\xbm.md: A required privilege is not held by the client.

Can someone help me?

Trickster-git, 2022-10-24:
Hi
You need add zip library
 sketch-> include library-> add zip library

per1234, 2025-04-25:
This was fixed by https://github.com/tttapa/Control-Surface/commit/ed57f9b22f9e656a9cb7363c812ea9f56e92d5ae