tttapa/Control-Surface #823: How connect arduino 8266(node mcu) with zoom g1 xon via usb?

e36910, 2022-10-22:
What code do I need to write to connect the device via usb to the zoom pedal and control it?
Example in video [https://youtu.be/XSJGG0wdzXA](https://youtu.be/XSJGG0wdzXA)

laps78, 2022-10-23:
Hi!
Sorry but I can't help you with it for I just only forked this library. I don't have practice usin it yet.

You should ask for help by community in original repo. Visit home page
https://github.com/tttapa/Control-Surface

e36910, 2022-10-23:
We are in this community

tttapa, 2022-10-23:
The ESP8266 does not have USB support, you'll need a different microcontroller, one with USB Host capabilities.  
Alternatively, you could use some kind of “USB Host shield”, but this can be cumbersome.

e36910, 2022-10-23:
Would an esp32 board or an arduino nano board be suitable for this?

e36910, 2022-10-23:
Wich arduino boards support usb midi?

tttapa, 2022-10-26:
The high-end Teensies (3.5, 3.6, 4.1), some models of ESP32, (although these are not officially supported by Control Surface, so you'll need some coding skills to get it to work), the RP2040 (again, coding required), and probably many more.

I don't have enough experience with any of them to recommend a specific one for your use case.

Once you have a board and a library with USB Host support, using it with Control Surface should be very straightforward, you just have to supply functions for reading and writing USB MIDI packets, see e.g. [Custom-USB-MIDI-Backend.ino](https://tttapa.github.io/Control-Surface-doc/Doxygen/d7/d5f/Custom-USB-MIDI-Backend_8ino-example.html).