tttapa/Control-Surface #828: MIDI_Notes::F not working

tzenn, 2022-10-28:
all of the "midi.sendNoteOff({ MIDI_Notes::__, CHANNEL_1 }, velocity);" work except for note F?

the error:
no matching function for call to 'CS::USBMIDI_Interface::sendNoteOff(<brace-enclosed initializer list>, const uint8_t&)

the code:
if (strumbutton1.getState()) {
      midi.sendNoteOff({ MIDI_Notes::C(4), CHANNEL_1 }, velocity);
      midi.sendNoteOff({ MIDI_Notes::D(4), CHANNEL_1 }, velocity);
      midi.sendNoteOff({ MIDI_Notes::E(4), CHANNEL_1 }, velocity);
      midi.sendNoteOff({ MIDI_Notes::F(4), CHANNEL_1 }, velocity);
      midi.sendNoteOff({ MIDI_Notes::G(4), CHANNEL_1 }, velocity);
      midi.sendNoteOff({ MIDI_Notes::A(4), CHANNEL_1 }, velocity);
      midi.sendNoteOff({ MIDI_Notes::B(4), CHANNEL_1 }, velocity);
}

tttapa, 2022-10-28:
See the FAQ: <https://tttapa.github.io/Control-Surface-doc/Doxygen/da/dc1/FAQ.html#faq-midi-note-f>

tzenn, 2022-11-01:
ohhh thank you so much

closed 2022-11-01 by tzenn