tttapa/Control-Surface #850: Can I use arduino pro micro with zoom g1xon via usb without host shield?

e36910, 2022-11-19:
Wich library will be to connect for working zoom devices?
[image]

tttapa, 2022-11-19:
You cannot connect two USB devices to each other, you need at least one host. The ATmega32U4 only supports device mode.

The Zoom's manual also doesn't mention MIDI at all, it seems that the USB connection only supports firmware upgrades, are you sure it supports MIDI over USB?

e36910, 2022-11-19:
Yes, I`m sure it supports MIDI over USB. Look this video - [https://youtu.be/rHc8-jrd__c?t=220](url)

tttapa, 2022-11-19:
Okay, that's good. Either way, you'll need a USB host (unless the Zoom supports USB OTG, which I doubt).

e36910, 2022-11-19:
How do I check the zoom for USB OTG support?

tttapa, 2022-11-19:
You'll have to check the manual/documentation or contact the manufacturer. Or you can just try an OTG adapter. But I'm 99% sure it doesn't support OTG, otherwise they would advertise it.

e36910, 2022-11-19:
How to connect USB host shield with arduino pro micro?

tttapa, 2022-11-19:
I'm afraid that's not something I can help you with. I'm sure you can find more information online, but if you need more help after reading the online instructions, you could open a thread on https://forum.arduino.cc/.