tttapa/Control-Surface #851: Change encoder multiplier

Trickster-git, 2022-11-20:
Hi, it's possible change encoder multiplier by changing variable? 
I want to make it in variables for bank i have other idea.
Board pi pico.

#include <Control_Surface.h> // Include the Control Surface library
#include <Arduino_Helpers.h>
#include <AH/Hardware/Button.hpp>

// Instantiate a MIDI over USB interface. 31250
USBMIDI_Interface usbmidi;
HardwareSerialMIDI_Interface serialmidi = {Serial, 256000};//2000000

MIDI_PipeFactory<10> pipes;

// enc 1fst ard
  int Enc1  = 0 ;
  int Enc2  = 1 ;
  int Enc3  = 2 ;
  int Enc4  = 3 ;
  int Enc5  = 4 ;
  int Enc6  = 5 ;
  int Enc7  = 6 ;
  int Enc8  = 7 ;
  int Enc9  = 8 ;
  int Enc10 = 9 ;
  int Enc11 = 10;
  int Enc12 = 11;
  int Enc13 = 12;
  int Enc14 = 13;
  int Enc15 = 14;
  int Enc16 = 15;
// end enc 1fst ard
  int Pot1  = 0 +16;
  int Pot2  = 1 +16;
  int Pot3  = 2 +16;
  int Pot4  = 3 +16;
  int Pot5  = 4 +16;
  int Pot6  = 5 +16;
  int Pot7  = 6 +16;
  int Pot8  = 7 +16;
  int Pot9  = 8 +16;
  int Pot10 = 9 +16;
  int Pot11 = 10+16;
  int Pot12 = 11+16;
  int Pot13 = 12+16;
  int Pot14 = 13+16;
  int Pot15 = 14+16;
  int Pot16 = 15+16;

Button qpuls_button = {7};  // the number of the pushbutton pin
Button qpuls_button2 = {8};  // the number of the pushbutton pin
int8_t ledPin = 25;
// const int buttonPin = 7;

int16_t muti    = 1;
int16_t qpuls   = 4;
int8_t  qflug   = 0;

// on first ard
CD74HC4067 mux = {
  {A0},       
  {18, 19, 20, 21,} // Address pins S0, S1, S2
};
CCPotentiometer volumePotentiometers[] = {
 {mux.pin( 0), {Pot1 , CHANNEL_2}},
 {mux.pin( 1), {Pot2 , CHANNEL_2}},
 {mux.pin( 2), {Pot3 , CHANNEL_2}},
 {mux.pin( 3), {Pot4 , CHANNEL_2}},
 {mux.pin( 4), {Pot5 , CHANNEL_2}},
 {mux.pin( 5), {Pot6 , CHANNEL_2}},
 {mux.pin( 6), {Pot7 , CHANNEL_2}},
 {mux.pin( 7), {Pot8 , CHANNEL_2}},
 {mux.pin( 8), {Pot9 , CHANNEL_2}},
 {mux.pin( 9), {Pot10, CHANNEL_2}},
 {mux.pin(10), {Pot11, CHANNEL_2}},
 {mux.pin(11), {Pot12, CHANNEL_2}},
 {mux.pin(12), {Pot13, CHANNEL_2}},
 {mux.pin(13), {Pot14, CHANNEL_2}},
 {mux.pin(14), {Pot15, CHANNEL_2}},
 {mux.pin(15), {Pot16, CHANNEL_2}},
};
// Bankable::CCAbsoluteEncoder<>
// enc 1fst ard
////////////////////////////////////////////////////////////////////////////////////////////////
//////HERE i change variable /qpuls/ 
/////////////////////////////////////////////////////////////////////////////////////////////
  CCAbsoluteEncoder Enc9led  = {{10, 11}, {Enc9  , CHANNEL_2, CABLE_1}, qpuls,};
  CCAbsoluteEncoder Enc10led = {{12, 13}, {Enc10 , CHANNEL_2, CABLE_1}, qpuls,};
  // CCAbsoluteEncoder Enc10led = {{12, 13}, {Enc10 , CHANNEL_2, CABLE_1}, muti, qpuls,};
// end enc 1fst ard

// Bankable::CCAbsoluteEncoder<size_t NumBanks>

  CCLEDPWM ledEnc9  = {14, {Enc9 ,CHANNEL_2,}};
  CCLEDPWM ledEnc10 = {15, {Enc10,CHANNEL_2,}};
// CCValueFastLED
// CCLED
// CCValue

void setup() {
  // serialmidi | pipes | usbmidi;
  // Control_Surface | pipes | usbmidi;
  // Control_Surface | pipes | serialmidi;  

  serialmidi >> pipes >> usbmidi; 
  serialmidi << pipes << usbmidi; 
  serialmidi << pipes << serialmidi; // if main usb interface need coment

  Control_Surface >> pipes >> usbmidi;
  Control_Surface << pipes << usbmidi;
  Control_Surface >> pipes >> serialmidi;
  Control_Surface << pipes << serialmidi;

  // pinMode(buttonPin, INPUT_PULLUP);
  pinMode(ledPin, OUTPUT);
  qpuls_button.begin();
  qpuls_button2.begin();

  usbmidi.begin();
  serialmidi.begin();
  Control_Surface.begin(); // Initialize Control Surface
  // Serial.begin(115200);
}

void loop() {
//   static bool ledState = LOW;
//   // int buttonState = 0; 
// if (qpuls_button.update() == Button::Falling) {
//   ledState = !ledState; 
//   digitalWrite(ledPin, ledState ? HIGH : LOW);
  //  ( ledState ? (qpuls   = 4) : (qpuls   = 1));
// }
///////////////////////////////////////////////////////////////////////////////
/// HERE two buttons to switch encoder multiplier, it work for led but not for multiplier. 
///////////////////////////////////////////////////////////////////////////////////////
if (qpuls_button.update() ) {
  if (qpuls_button.getState() == Button::Released){
    }
  if (qpuls_button.getState() == Button::Falling){
     qpuls   = 4    ;
     digitalWrite(ledPin , LOW )  ;
    }
}
if (qpuls_button2.update() ) {
  if (qpuls_button2.getState() == Button::Released){
    }
  if (qpuls_button2.getState() == Button::Falling){
     qpuls   = 1    ;
     digitalWrite(ledPin , HIGH ) ;
    }
}

  // if ((buttonState == HIGH) && (qflug  == 0)) {
  //   qpuls   = 4;
  //   qflug   = 1;
  //   }
  // if ((buttonState == HIGH) && (qflug  == 1)) { 
  //   qpuls   = 1;
  //   qflug   = 0;
  //   }

// static bool ledState = LOW;
// uint8_t     qpuls_stat = 4;
  // if (qpuls_button.update() == Button::Falling) {
  //   ledState = !ledState; // Invert the state of the LED
  //   digitalWrite(ledPin, ledState ? HIGH : LOW);
  // if (ledState)  {qpuls=4;}
  // else{  qpuls=1;  }
    // qpuls ? 4 : 1;
    // digitalWrite(ledPin, ledState ? qpuls=4 : qpuls=1);
    // qpuls(ledState ? 4 : 1);
  

  // if (qpuls_button.update() == Button::Falling && qflug==1) {qpuls=1; qflug=0; }
  // else if (qpuls_button.update() == Button::Falling && qflug==0) {qpuls=4; qflug=1;}

  usbmidi.update();
  serialmidi.update();
  Control_Surface.loop(); // Update the Control Surface
  // Serial.println(qpuls);
  // Serial.println("qpuls");

}

tttapa, 2022-11-20:
> Hi, it's possible change encoder multiplier by changing variable?

No, but you can just use the [`CCAbsoluteEncoder::setSpeedMultiply()`](https://tttapa.github.io/Control-Surface-doc/Doxygen/d8/d2f/classCCAbsoluteEncoder.html#a3081636a6e8673c41e59a82c024450ff) function.

Trickster-git, 2022-11-20:
Thanx, but haw i can use [CCAbsoluteEncoder::setSpeedMultiply()](https://tttapa.github.io/Control-Surface-doc/Doxygen/d8/d2f/classCCAbsoluteEncoder.html#a3081636a6e8673c41e59a82c024450ff) can you provide example.

tttapa, 2022-11-20:
It's an ordinary member function: https://www.learncpp.com/cpp-tutorial/classes-and-class-members

E.g. `Enc9led.setSpeedMultiply(4);`.

closed 2022-11-21 by Trickster-git