tttapa/Control-Surface #877: Rotary Encoder not compiling for esp32

TrailerParkerBeats, 2023-01-15:
**Describe the bug**
I Seem to be unable to get the basic example sketch for the rotary encoder to work for a NodeMCU-32S esp32, it just returns an error saying "Error compiling for board NodeMCU-32S."

**To Reproduce**
if you try to compile the code for a NodeMCU-32S it returns the error.

**Expected behavior**
I was expecting now that esp32 and esp8266 are incudled in the comments as compatible boards the sketch would compile.

**Compile Error**
```cpp
/var/folders/0v/_w2_vrmd62d5kd_5p0b3sqtr0000gn/T/arduino_build_722896/sketch/RotaryEncoder.ino.cpp.o: In function `CS::AHEncoder::update()':
Multiple libraries were found for "Control_Surface.h"
 Used: /Users/joshparker/Documents/Arduino/libraries/Control-Surface-main
 Not used: /Users/joshparker/Documents/Arduino/libraries/Control-Surface-1.2.0
/Users/joshparker/Documents/Arduino/libraries/Control-Surface-main/src/MIDI_Constants/Chords/Chords.hpp:16:(.iram1.16[CS::AHEncoder::update()]+0x3c): dangerous relocation: l32r: literal placed after use: .literal._ZN2CS9AHEncoder6updateEv
collect2: error: ld returned 1 exit status
```

```cpp
// Your code here
/** 
 * This example demonstrates the use of relative MIDI CC rotary encoders that
 * can be used for changing effect parameters, volumes, pan and balance
 * controls, etc.
 *
 * @boards  AVR, AVR USB, Nano Every, Due, Nano 33 IoT, Nano 33 BLE, Pi Pico, Teensy 3.x, ESP32, ESP8266, ESP8266
 * 
 * Connections
 * -----------
 * 
 * - 27: pin A of the encoder
 * - 14: pin B of the encoder
 * 
 * Connect the common terminal of the encoder to ground. The built-in pullup
 * resistors will be enabled.
 * 
 * Behavior
 * --------
 * 
 * - When you turn the encoder, you should receive incremental MIDI Control
 *   Change messages.
 * 
 * Mapping
 * -------
 * 
 * Select the Arduino as a custom MIDI controller in your DAW, and use the 
 * relative CC mode.  
 * There are three main modes for sending relative messages, make sure you
 * use the same one in this sketch as in your DAW.
 * 
 * @see relativeCCmode
 * 
 * Written by PieterP, 2019-12-18  
 * https://github.com/tttapa/Control-Surface
 */

#include <Control_Surface.h> // Include the Control Surface library

// Instantiate a MIDI over USB interface.
USBMIDI_Interface midi;

// Instantiate a CCRotaryEncoder object
CCRotaryEncoder enc {
  {27, 14},       // pins
  MCU::V_POT_1, // MIDI address (CC number + optional channel)
  1,            // optional multiplier if the control isn't fast enough
};

void setup() {
  // Select the correct relative MIDI CC mode.
  // Options:
  //   - TWOS_COMPLEMENT (default)
  //   - BINARY_OFFSET
  //   - SIGN_MAGNITUDE
  //   - NEXT_ADDRESS
  // Aliases:
  //   - REAPER_RELATIVE_1
  //   - REAPER_RELATIVE_2
  //   - REAPER_RELATIVE_3
  //   - TRACKTION_RELATIVE
  //   - MACKIE_CONTROL_RELATIVE
  //   - KORG_KONTROL_INC_DEC_1
  RelativeCCSender::setMode(relativeCCmode::MACKIE_CONTROL_RELATIVE);
  Control_Surface.begin(); // Initialize Control Surface
}

void loop() {
  Control_Surface.loop(); // Update the Control Surface
}
```

tttapa, 2023-01-15:
This seems more difficult than expected: https://stackoverflow.com/questions/19532826/what-does-a-dangerous-relocation-error-mean

Do you get the same error for the `main` branch?

TrailerParkerBeats, 2023-01-15:
Oh that does seem like a fairly substantial issue. I am relatively new to coding, I have rotary encoder sketches that I can use to handle the reading of the encoder. I am struggling with how to then send them in relative mode so that I can increment bpm by 1 bpm per click.

TrailerParkerBeats, 2023-01-15:
And this is using the main branch as far as I'm aware.

Vahen1981, 2023-09-17:
Hello!
Could you solve this problem? Now I'm experimenting the same... no problem compiling for arduino uno or pro micro, but this same problem with ESP32