tttapa/Control-Surface #880: Raspberry Pi Pico showing up as USB serial, not as USB MIDI

DJLevel3, 2023-01-28:
**Describe the bug**
As stated in the title, I am trying to use a Raspberry Pi Pico as a MIDI controller and rather than getting a USB MIDI device when I plug in the Pi, I get a USB serial device.

**To Reproduce**
On either the Arduino or the Philhower libraries, create a MIDI device (in my case, a simple box with 6 analog inputs on 2 multiplexers). Write the appropriate code to drive said analog inputs and upload to the board. Congratulations! You get the data over a serial connection when a USB MIDI connection is expected. The data sends correctly in USBDebugMIDI_Interface mode, but it's still a serial connection in USBMIDI_Interface

**Expected behavior**
I expect a MIDI device to show up in Windows, instead a USB serial device shows up

**Code**
```cpp
#include <Control_Surface.h>

USBMIDI_Interface midi;

AnalogMultiplex<2> mux1 {
  A0,
  {0, 1}
};
AnalogMultiplex<2> mux2 {
  A1,
  {0, 1}
};

CCPotentiometer ccPots[] {
  {mux1.pin(0), {MIDI_CC::General_Purpose_Controller_1, CHANNEL_1 } },
  {mux1.pin(1), {MIDI_CC::General_Purpose_Controller_1, CHANNEL_2 } },
  {mux1.pin(2), {MIDI_CC::General_Purpose_Controller_1, CHANNEL_3 } },
  {mux2.pin(0), {MIDI_CC::General_Purpose_Controller_1, CHANNEL_4 } },
  {mux2.pin(1), {MIDI_CC::General_Purpose_Controller_1, CHANNEL_5 } },
  {mux2.pin(2), {MIDI_CC::General_Purpose_Controller_1, CHANNEL_6 } },
};

void setup() {
  // put your setup code here, to run once:
  Control_Surface.begin();
}

void loop() {
  // put your main code here, to run repeatedly:
  Control_Surface.loop();
}
```

tttapa, 2023-01-29:
Are you using the `main` version of Control Surface? Version 1.2.0 does not support the Pi Pico yet.

I've just tested Control Surface `main` (e5757ebd8d923565095766940f1d2c1e7c1c254b) with `mbed_rp2040/3.5.4` and it showed up as a MIDI device on Windows 11 and Ubuntu 22.04, sending and receiving MIDI as expected.

DJLevel3, 2023-01-29:
I might not be, I'll check when I get home. I thought I was on the latest
but I'm not actually 100% sure

matthias-glatthorn, 2024-06-02:
Any updates on this? I have the exact same problem ...

matthias-glatthorn, 2024-06-03:
Oh, solved it :) had to switch to "USB Stack: Adafruit Tiny USB"