tttapa/Control-Surface #882: PBpotentiometer resolution

simoniddqd, 2023-02-01:
My DAW reads PBpotentiometer with 2 db increments/decrements, and I would like to have it more fine toothed. Using a Pro Micro

Just instatiating a PBpotentiometer works great, but I wanted a finer resolution, so I tried the FilteredAnalog approach with 14 bit and then calling midi.sendBitchBend in void loop like so: 
```cpp
static Timer<millis> timer = 1; // ms
 if (timer && analog.update())
 midi.sendPitchBend(MCU::VOLUME_7, analog.getValue());``` 

This too works great, however at certain levels very small fluctuations are read and sent and causes problems with my DAW. I have tried all possible filtering shift factors, but the problem remains. Is there a way to further filter the read input? I am certain I am missing something here

This is the full code

```cpp
#include <Control_Surface.h>

USBMIDI_Interface midi;
FilteredAnalog<14,      // Output precision in bits
               8,       // The amount of filtering
               uint32_t // The integer type for the calculations
               >
  analog = A0;

void setup() {

  analog1.setupADC();

  analog1.resetToCurrentValue();

     RelativeCCSender::setMode(relativeCCmode::MACKIE_CONTROL_RELATIVE);
  Control_Surface.begin();
}
void loop() {
  Control_Surface.loop();
  static Timer<millis> timer = 1; // ms
  if (timer && analog.update())
        midi.sendPitchBend(MCU::VOLUME_1, analog.getValue());
          static Timer<millis> timer2 = 1; // ms
}
```

tttapa, 2023-02-02:
You could try using a wider `AnalogType` as well (e.g. use `FilteredAnalog<14, 8, uint32_t, uint32_t>`). This will give you some more bits of filter precision, and this makes the hysteresis more effective. By default, the `FilteredAnalog` class increases the analog resolution to the following value: https://tttapa.github.io/Control-Surface-doc/Doxygen/d1/db3/structAH_1_1MaximumFilteredAnalogIncRes.html#a6d3bdcf9951ef5b48e2aefcb317f7b9a

That being said, the ADC of the Pro Micro only has a resolution of 10 bits. The 10-bit readings already have 1 or 2 bits of noise, but with software oversampling in software, FilteredAnalog can give you a bit more resolution. However, going from a noisy 10-bit resolution to 14 noise-free bits might be a bit of a stretch.  
Commercial control surfaces like the Mackie MCU only use the top 10 bits of the Pitch Bend values for that reason.

simoniddqd, 2023-02-02:
Thank you very much Pieter, I am afraid I don't understand your answer. Just dived into programming with this beautiful library. I am not sure what is meant by using a wider AnalogType. Using `FilteredAnalog<14, 8, uint32_t, uint32_t>` gives me _"Compilation error: two or more data types in declaration of 'type name'"._

Is there a way to make `analog.update` only update if the value change is greater than say 10 ?

tttapa, 2023-02-04:
> I am not sure what is meant by using a wider AnalogType

`AnalogType` is the fourth template argument to the `FilteredAnalog` class: https://tttapa.github.io/Control-Surface-doc/Doxygen/d3/d6e/classAH_1_1FilteredAnalog.html

> Using `FilteredAnalog<14, 8, uint32_t, uint32_t>` gives me _"Compilation error: two or more data types in declaration of 'type name'"._

Please post the full code and the full error message you get.

> Is there a way to make `analog.update` only update if the value change is greater than say 10 ?

`FilteredAnalog` already does this internally using the `Hysteresis` class: https://tttapa.github.io/Control-Surface-doc/Doxygen/d2/dcb/classHysteresis.html

You can add some extra logic on top of that, but only updating the value only when the change is larger than x actually lowers the resolution, and makes it harder to reach the boundaries of the range.

simoniddqd, 2023-02-05:
Hi Pieter, I was missing a comma in regards to the compilation error. I am in awe of the great work you do here, thank you so much.

closed 2023-02-05 by simoniddqd

NumericSpace, 2023-02-05:
Colleagues! I tried this code! Cool! The fader moves very smoothly! Could you show how this code is written for multiple faders? Thank you!