tttapa/Control-Surface #883: 2 different Banks saving data on SD with Json Structure

danilobelpedio, 2023-02-02:
First of everything,  Pieter thanks for your wonderful job.
I Am Danilo from Italy and I have to ask you help for an ideas that arrived from problems during music live performance .
I am not a musician (hobby), I am not a programmer (hobby), and I have few times to do a lot of thing that I have in my head.
I programmed at university and sometimes at work but when I programmed first time there was the "basic" program language on so I m new in this kind of wonderful system like Arduino.
Well, it's the first time I write here and I solved a lot of problems reading here, so thanks also to everyone.
My problem is:
how I can use two different banks (like bank1 and bank2) for two arrays of encoders saving data on a json structure and alter on a sd?

My code is a mix of your code and other code around the network...
I modified also your code (sorry don't be angry and I hope that I done good job...)

Anyways the code below is only for one encoder one bank: (I USE an ARDUINO MEGA... because the code need more memory )

#include <Control_Surface.h> // Include the Control Surface library
#include <SPI.h> 
#include <ArduinoJson.h>
#include <SD.h>

#define SS 53     //Pin del SS del SD Reader

// variabile per la definizione del bottone di write e read
#define write 3          //Pin del pulsante per scrivere i data su SD
#define read 2          //Pin del pulsante per leggere i data salvati su SD

// definisco i Pin del RGBLed per la SD
#define RgbLedRedPin 54   //A0 analogPin
#define RgbLedGreenPin 55 //A1 analogPin
#define RgbLedBluePin 56  //A2 analogPin

//variabili per il debounce dei pulsanti
long Debounce_delay = 100;
long t1 = 0;

const char *filename = "MONITORS.TXT";  // <- SD library uses 8.3 filenames

// Creo l'oggetto JSON 
DynamicJsonDocument doc(1024);
JsonObject obj;

File myFileSDCart;

JsonObject getJSonFromFile(DynamicJsonDocument *doc, String filename, bool forceCleanONJsonError = true ) 
  {
    myFileSDCart = SD.open(filename);     // open the file for reading:
    if (myFileSDCart) {                   // read from the file until there's nothing else in it:
        DeserializationError error = deserializeJson(*doc, myFileSDCart);
        if (error) { 
            Serial.print(F("Error parsing JSON ")); // if the file didn't open, print an error:
            Serial.println(error.c_str());
            if (forceCleanONJsonError){
                return doc->to<JsonObject>();
            }
        }
        myFileSDCart.close();     // close the file:
        return doc->as<JsonObject>();
    } else {
        Serial.print(F("Error opening (or file not exists) "));   // if the file didn't open, print an error:
        Serial.println(filename);
        Serial.println(F("Empty json created"));
        return doc->to<JsonObject>();
    }
}

// Custom callback to handle output for a selector.
class MySelectorCallback {
  uint8_t pickedColor[10][3] ={ // defined Bank Colors R,G,B
    {255,  0,  0},     //rosso
    {200,100, 53},     //giallo limone
    {  0,  0,150},     //blu
    {  0,100,  0},     //verde
[... 279 lines cut ...]
// Handy type alias
using  CCAbsEnc1 = Bankable::CCAbsoluteEncoder<9>; 

CCAbsEnc1 enc1[] = {
  {  
  {bank1, BankType::CHANGE_ADDRESS},
  .....
};

probably I need something like that
but my problem is

  for (int j=0; j < 8; j++) {
      enc[0].setValue(obj[F("encVals")][j],j);
    }
I read values from the json structure

with this code
    for (int j = 0; j < 8; j++) {                          
    obj[F("encVals")][j] = enc[0].getValue(j);                            
      }      
how I can change this code to identify the two different banks?

Thanks for the help...
thanks a lot

tttapa, 2023-02-02:
> how I can change this code to identify the two different banks?

Which two banks?

JSON sounds like overkill for this task, to be honest. You could consider using the internal EEPROM instead.  
As a side note, you can get rid of the switch statements for the color selection, since each case is the same.

danilobelpedio, 2023-02-02:
Thanks for the answer Pieter, 
I mean that i need something like that:

Bank<8> bankMonitors(11); // 8 banks for 11 encoders
Bank<9> bankMixer(8); // 9 banks for 8 encoders

I used json but of course could be better to write on the EEPROM...
Anyways what you means that i can remove the switch statment to pick the color? Because i need 3 different colors taken from the array...but I don't understand what you mean about that are the same...

tttapa, 2023-02-04:
> Bank<8> bankMonitors(11); // 8 banks for 11 encoders
> Bank<9> bankMixer(8); // 9 banks for 8 encoders

It's not clear to me where you're stuck exactly. Is it the data format to save? How to get the current settings out of the Control Surface object? How to set them again when loading from the SD card?

> I had problems with the SD library because in that library you write on the SD only chars not integer and i didn't find a good solutions to convert datas...so i tryed json an solved...

Internally, the Arduino represents all data as 8-bit bytes. Your storage medium doesn't need to have the ability to store arbitrary types, if it also supports 8-bit bytes, you can store anything.

> Anyways what you means that i can remove the switch statment to pick the color? Because i need 3 different colors taken from the array...but I don't understand what you mean about that are the same...

```cpp
    uint8_t getColorRed(setting_t setting) {
        if (setting >= 8)
            return 0;
        return pickedColor[setting][0];
    }
```
You also don't need a function for each color, but let's not get ahead of ourselves.