tttapa/Control-Surface #887: Shared / borrowed encoder

simoniddqd, 2023-02-06:
In my project I am sharing a rotary encoder between an instance of PBAbsoluteEncoder and an instance of CCRotaryEncoder by physically switching pins with a switch. However I would like to make this more simple, perhaps with a small button switch or maybe one of the multipurposebutton commands, and wonder if it could be done with code. I've looked at [this](https://github.com/tttapa/Control-Surface/issues/76#issuecomment-567554695) but it seems to be outdated. Any help is greatly appreciated

simoniddqd, 2023-02-08:
This is the compilation error I get when trying to compile the example from the thread

`/private/var/folders/mx/kjcsv4q907jblll2vbk5807w0000gn/T/.arduinoIDE-unsaved202318-2467-1jcxd3g.ygz3k/sketch_feb8a/sketch_feb8a.ino:16:1: error: could not convert '{enc, CS::MCU::V_POT_1, 1, 4, <brace-enclosed initializer list>()}' from '<brace-enclosed initializer list>' to 'CS::BorrowedMIDIRotaryEncoder<CS::ContinuousCCSender> {aka CS::GenericMIDIRotaryEncoder<CS::AHEncoder&, CS::ContinuousCCSender>}'
 };
 ^
/private/var/folders/mx/kjcsv4q907jblll2vbk5807w0000gn/T/.arduinoIDE-unsaved202318-2467-1jcxd3g.ygz3k/sketch_feb8a/sketch_feb8a.ino:23:1: error: could not convert '{enc, CS::CHANNEL_1, 127, 4, <brace-enclosed initializer list>()}' from '<brace-enclosed initializer list>' to 'CS::BorrowedMIDIAbsoluteEncoder<CS::PitchBendSender<14> > {aka CS::GenericMIDIAbsoluteEncoder<CS::AHEncoder&, CS::PitchBendSender<14> >}'
 };
 ^
/private/var/folders/mx/kjcsv4q907jblll2vbk5807w0000gn/T/.arduinoIDE-unsaved202318-2467-1jcxd3g.ygz3k/sketch_feb8a/sketch_feb8a.ino: In function 'void loop()':
/private/var/folders/mx/kjcsv4q907jblll2vbk5807w0000gn/T/.arduinoIDE-unsaved202318-2467-1jcxd3g.ygz3k/sketch_feb8a/sketch_feb8a.ino:37:11: error: 'CS::BorrowedMIDIAbsoluteEncoder<CS::PitchBendSender<14> > {aka class CS::GenericMIDIAbsoluteEncoder<CS::AHEncoder&, CS::PitchBendSender<14> >}' has no member named 'resetPositionOffset'
     pbenc.resetPositionOffset();
           ^~~~~~~~~~~~~~~~~~~
/private/var/folders/mx/kjcsv4q907jblll2vbk5807w0000gn/T/.arduinoIDE-unsaved202318-2467-1jcxd3g.ygz3k/sketch_feb8a/sketch_feb8a.ino:41:11: error: 'CS::BorrowedMIDIRotaryEncoder<CS::ContinuousCCSender> {aka class CS::GenericMIDIRotaryEncoder<CS::AHEncoder&, CS::ContinuousCCSender>}' has no member named 'resetPositionOffset'
     ccenc.resetPositionOffset();
           ^~~~~~~~~~~~~~~~~~~
Multiple libraries were found for "MIDIUSB.h"
  Used: /Users/simon_mbpm1/Documents/Arduino/libraries/MIDIUSB
  Not used: /Users/simon_mbpm1/Documents/Arduino/libraries/MIDIUSB-master
exit status 1

Compilation error: could not convert '{enc, CS::MCU::V_POT_1, 1, 4, <brace-enclosed initializer list>()}' from '<brace-enclosed initializer list>' to 'CS::BorrowedMIDIRotaryEncoder<CS::ContinuousCCSender> {aka CS::GenericMIDIRotaryEncoder<CS::AHEncoder&, CS::ContinuousCCSender>}'`

tttapa, 2023-02-08:
Untested, but this should compile using the latest version (02374bca8512a299cce8a2424fe34a7d0f0856cd):

```cpp
#include <Control_Surface.h>

USBDebugMIDI_Interface midi;

Button button {4};

AHEncoder enc {2, 3};
BorrowedMIDIRotaryEncoder<ContinuousCCSender> ccenc {
  enc, 
  MCU::V_POT_1,
  1,   // multiplier
  4,   // pulses per step
  {},  // MIDI sender
};
BorrowedMIDIAbsoluteEncoder<PitchBendSender<14>> pbenc {
  enc, 
  CHANNEL_1,
  127, // multiplier
  4,   // pulses per step
  {},  // MIDI sender
};

void setup() {
  button.begin();
  Control_Surface.begin();
  pbenc.disable();
}

void loop() {
  Control_Surface.loop();

  auto buttonState = button.update();
  if (buttonState == Button::Falling) {
    ccenc.disable();
    pbenc.resetPositionOffset();
    pbenc.enable();
  } else if (buttonState == Button::Rising) {
    pbenc.disable();
    ccenc.resetPositionOffset();
    ccenc.enable();
  } 
}
```

simoniddqd, 2023-02-08:
Thank you Pieter for looking into this, it sure does compile, but nothing comes up in the serial monitor. I've double checked my wiring, and checked if the sketch works with each encoder object individually, and it does.  However, when introducing both encoder objects at the same time while disabling one of them, nothing shows in serial monitor.

`#include <Control_Surface.h>

USBDebugMIDI_Interface midi;

AHEncoder enc {2, 3};
BorrowedMIDIRotaryEncoder<ContinuousCCSender> ccenc {
  enc, 
  MCU::V_POT_1,
  1,   // multiplier
  4,   // pulses per step
  {},  // MIDI sender
};
BorrowedMIDIAbsoluteEncoder<PitchBendSender<14>> pbenc {
  enc, 
  CHANNEL_1,
  127, // multiplier
  4,   // pulses per step
  {},  // MIDI sender
};

void setup() {

  Control_Surface.begin();
  pbenc.disable();
}

void loop() {
  Control_Surface.loop();
}`

I am really at the boundaries of my knowledge here, so any help or pointers in the right direction is hugely appreciated :)

tttapa, 2023-02-08:
I think the issue that the encoder is initialized twice, and the second time, it raises an error because the interrupts are already in use. You can try moving the `pbenc.disable()` line before `Control_Surface.begin()`, so Control Surface won't initialize `pbenc`. 

Alternatively, you can try 39182bbab79487236c11579db3acfa336813bf54, which should make initializing encoders multiple times no longer fail.

simoniddqd, 2023-02-09:
Fantastic!

closed 2023-02-09 by simoniddqd