tttapa/Control-Surface #888: Help no Codigo

davidgauze, 2023-02-07:
Prezado, a tempo atras havia escrito esse codigo com sua ajuda,  porem faz 1 ano que não atualizei mais o mesmo codigo,
estou colocando esse mesmo comando num arduino mega usando Multiplexador 16 Canais Analógico Digital CD74HC4067, estou tendo poblemas para carregar no arduino, dando erros de encoder, notes  e analog  multiplexer.

gostaria de saber se pode me ajudar com esse codigo, lembrando que semprei no inicio o mesmo codigo e funcionava 
[image]

linhas de comando das notas nos mux
[image]
 estou ainda criando conforme tabelas.

// Include the Control Surface library.
#include <Control_Surface.h>

#include <Encoder.h> // Include the Encoder library.
// This must be done before the Control Surface library.

// Instantiate a MIDI over USB interface.
//USBMIDI_Interface midi;

// Instantiate a multiplexer with three address lines.

AnalogMultiplex<1> mux1= {3, {10, 11, 12, 13 } };
AnalogMultiplex<2> mux2= {4, {10, 11, 12, 13 } };
AnalogMultiplex<3> mux3= {5, {10, 11, 12, 13 } };
AnalogMultiplex<4> mux4= {6, {10, 11, 12, 13 } };
AnalogMultiplex<5> mux5= {7, {10, 11, 12, 13 } };
AnalogMultiplex<6> mux6= {8, {10, 11, 12, 13 } };
AnalogMultiplex<7> mux7= {9, {10, 11, 12, 13 } };

const int speedMultiply = 1; // If the jog wheels or other encoders are too slow in your software, increase this value
                             // (it will be multiplied with the actual speed of the encoder, as the name implies.) Default is 1.

struct RelativeNoteSender {
    static void send(int delta, MIDIAddress address) {
        delta = constrain(delta, -127, 127);
        if (delta > 0) {
          Control_Surface.sendNoteOn(address, delta);
          // Send whatever your software expects
        } else {
          Control_Surface.sendNoteOn(address + 1 /* ??? */, -delta);
          // Send whatever your software expects
        }
    }

};
struct ContinuousNoteSender {
    void send(uint8_t value, MIDIAddress address) {
        Control_Surface.sendNoteOn(address, value);
    }  
    static constexpr uint8_t precision() { return 7; }
};

struct NoteEncoder : MIDIRotaryEncoder<RelativeNoteSender> {
    NoteEncoder(EncoderPinList pins, MIDIAddress address)
      : MIDIRotaryEncoder(pins, address, 1, 4, {}) {}
};

// Instantiate a shift register.

// Instantiate the momentary push buttons that send out MIDI Note events.

//NoteButtonLatching switches[] = {

 NoteButton buttons[] = {

 

AnalogMultiplex<1> mux1= { 3, {10, 11, 12, 13 } };
    { mux1.pin(0), {0x00, CHANNEL_1} }, // digital input pin, note number, [channel]
    { mux1.pin(1), {0x00, CHANNEL_1} }, // digital input pin, note number, [channel]
[... 162 lines cut ...]
    {A8, note(0x0, 1)}, // Fader 
    {A9, note(0x0, 1)}, // Fader 
    {A10, note(0x0, 1)}, // Fader 
    {A11, note(0x0, 1)}, // Fader 

};

NoteEncoder enc[] = {
//Exemplo {{ Qual Pino , Qual Pino }, 0xQual Nota },
  {{ 40, 41 }, 0x0 }, // Encoder A
  {{ 42, 43 }, 0x0 }, // Encoder B
  {{ 43, 44 }, 0x0 }, // Encoder C
  {{ 45, 46 }, 0x0 }, // Encoder D

};

void setup() {
    // Initialize everything.
    Control_Surface.begin();
}

void loop() {
    // Update the control surface: read new MIDI events, and read all inputs.
    Control_Surface.loop();
}

tttapa, 2023-02-07:
The first template argument of the `AnalogMultiplex` class is the number of address lines, see https://tttapa.github.io/Control-Surface-doc/Doxygen/d2/d94/classAH_1_1AnalogMultiplex.html.

If your multiplexers have four address lines, you should use `AnalogMultiplex<4>`.

The rest of the sketch also contains syntax errors, lines like this one are not valid on their own:
```cpp
    { mux1.pin(0), {0x00, CHANNEL_1} }, // digital input pin, note number, [channel]
```
I'd encourage you to start with a smaller example, using only a single multiplexer, and get that working first before duplicating all the code for multiple multiplexers.

davidgauze, 2023-02-08:
Ola , fiz pequeno esboço peço me ajude,  button, encoder e potenciômetros serão enviadas note on quando pressionadas,
 abaixo fiz condigo se possivel me dar help , estta dando erro novamente  se conseguir fazer exemplo pra ter ideia e por 4 encoder, 105 button e 7 fader, 
no software que irei usar é grandma2 onpc, software de iluminação

// Include the library
#include <Control_Surface.h>

#include <Submodules/Encoder/AHEncoder.hpp>

// Instantiate a MIDI Interface to use
USBMIDI_Interface midi;

using namespace ExtIO;
using namespace MIDI_Notes;

// Instantiate an analog multiplexer
CD74HC4067 mux1 {10, {3, 4, 5,6 } };

CD74HC4067 mux2 {11, {3, 4, 5,6 } };

const int speedMultiply = 1; // If the jog wheels or other encoders are too slow in your software, increase this value
                             // (it will be multiplied with the actual speed of the encoder, as the name implies.) Default is 1.

struct RelativeNoteSender {
    static void send(int delta, MIDIAddress address) {
        delta = constrain(delta, -127, 127);
        if (delta > 0) {
          Control_Surface.sendNoteOn(address, delta);
          // Send whatever your software expects
        } else {
          Control_Surface.sendNoteOn(address + 1 /* ??? */, -delta);
          // Send whatever your software expects
        }
    }

};
struct ContinuousNoteSender {
    void send(uint8_t value, MIDIAddress address) {
        Control_Surface.sendNoteOn(address, value);
    }  
    static constexpr uint8_t precision() { return 7; }
};

// Button multiplexadores 
NoteButton buttons[] = {
    { mux1.pin(0), {0x01, CHANNEL_1} }, // digital input pin, note number, [channel]
    { mux1.pin(1), {0x02, CHANNEL_1} }, // digital input pin, note number, [channel]
    { mux1.pin(2), {0x03, CHANNEL_1} }, // digital input pin, note number, [channel]
  
    { mux2.pin(0), {0x04, CHANNEL_1} }, // digital input pin, note number, [channel]
    { mux2.pin(1), {0x05, CHANNEL_1} }, // digital input pin, note number, [channel]
    { mux2.pin(2), {0x06, CHANNEL_1} }, // digital input pin, note number, [channel]

};

//potentiometer 
 
 CCPotentiometer potentiometer [] { 
  {A1, 0x13}, 
  {A2, 0x14}, 
  {A3, 0x15}, 
  
};
     
CCRotaryEncoder enc []{
    {40, 41},MIDI_Notes::A(1), 1, 
    {42, 43},MIDI_Notes::B(2), 1,
    {44, 45},MIDI_Notes::C(3), 1, 
    {46, 47},MIDI_Notes::D(4), 1,

};

void setup() {
  Control_Surface.begin(); // Initialize Control Surface
}

void loop() {
  Control_Surface.loop(); // Update the Control Surface
}

davidgauze, 2023-02-11:
Hello, can you make an example of how to send midi note on the fader and encoders?

#include <Control_Surface.h>

USBMIDI_Interface usbmidi;

using namespace MIDI_Notes;

Notespotentiometer faders[] {
  // {A0, MIDI_Notes::C(4)},
 // {A1, {0x01, CHANNEL_1}},

  };

CCRotaryEncoder enc {
{ 8, 9 }, {0x10},speedMultiplier, 4,  
{ 10, 11}, {0x20},speedMultiplier, 4,  

};

void setup() {
  Control_Surface.begin(); // Initialize Control Surface
}

void loop() {
  Control_Surface.loop(); // Update the Control Surface
}

tttapa, 2023-02-11:
If you run into a problem, try to isolate it first. Strip down your code to the bare minimum required to reproduce the error, and then post the code and the full error message.

> por 4 encoder, 105 button e 7 fader

Start with 1 encoder, 1 button and 1 fader, separately. Only start scaling up once you have the simplest test case working.

davidgauze, 2023-02-11:
sim, me de exemplos de encoder e fader enviando note on quando usada, modo CC software não aceita