tttapa/Control-Surface #891: Stop NoteOff or NotOn from happening

gringow831, 2023-02-09:
I am using NoteButton with ManyAddresses. At first I thought a NoteButtonLatched would be fine but it's not part of ManyAddresses as I've checked on some other post.

When adding a MultiPurposeButton with LongPress the result is 
Note On
Longpress (on)
Longpress (release)
Note Off

That last NoteOff triggers the button so I am trying to stop that from happening if I use long press. 

Is there a way to do that? Otherwise I guess having a NoteButtonLatched would help since it needs two steps?

gringow831, 2023-02-10:
The other approach I thought off is MyNoteButton with ManyAddresses and on release not do anything but I don't seem to find the key element to make it work.

```ruby
#include <Control_Surface.h>

namespace CS { namespace Bankable { namespace ManyAddresses {

  template <setting_t NumBanks>
class MyNoteButton : public MIDIOutputElement {
 public:

  MyNoteButton(const Bank<NumBank> &bank, pin_t pin,
               const Array<MIDIAddress, NumBanks> pin, uint8_t velocity)
    : address(ManyAddresses<NumBanks>, pin_t pin), button(pin), velocity(velocity) {}

 public:

  void begin() final override { button.begin(); }

  void update() final override {
    AH::Button::State state = button.update(); // Read the button
    if (state == AH::Button::Falling) {        // if pressed
      address.lock();
      Control_Surface.sendNoteOn(address.getActiveAddress(), velocity);
    } else if (state == AH::Button::Rising) { // if released, only unlock
       address.unlock();
    }
  }

 private:
  Bankable::ManyAddresses<NumBanks> address;
  AH::Button button;
  uint8_t velocity;
};

}}}

USBMIDI_Interface midi;

Bank<4> bank(12);

IncrementSelector<4> selector {
  bank, // Bank to manage
  4,    // push button pin
};

Bankable::ManyAddresses::MyNoteButton<4> button {
{bank, A0, {MCU::SELECT_1, MCU::SELECT_5, MCU::SELECT_3, MCU::SELECT_2}},
  0x7F,                             // Maximum velocity
};

void setup() {
  Control_Surface.begin();
}

void loop() {
  Control_Surface.loop(); 
```

gringow831, 2023-02-11:
Just realised the NoteOn is the one activating things with 0x7F velocity so the idea would be inverting the NoteButton array and the something like `If LongPress = true {Control_Surface.sendNoteOn(address.getActiveAddress(), 0x00);`

Is there a way to make that happen within the LongPress?

One more thing I realised when trying to make this work is, if using a SmartPotentiometer, even if I disable the smart pot this keeps sending the information so when I release the MultiPurposeButton the first potentiometer will jump to that second potentiometer position no matter what. 
How can I make that correct?

tttapa, 2023-02-11:
It's not clear to me what you're after exactly. Could you describe the behavior in more detail? (E.g. if I press switch A, send MIDI message X, if I turn pot B, send MIDI message Y, etc.)

gringow831, 2023-02-11:
with NoteButton, If I press button A short, act as normal (MCU::SELECT_1). (It can be inverted so OFF > ON)
If I longpress button A, deactivate the active Smart potentiometer 1 and activate potentiometer 2. 
When longpress release, Activate Smart potentiometer 1 and deactivate 2. Don't send NoteOn from the MCU::Select_1 or if that's not possible send NoteOn, 0x00 velocity so nothing happens.

tttapa, 2023-02-11:
I would start by doing everything manually in the main loop, with a multipurpose button and a switch statement, like in one of your snippets. You can then later abstract the logic away in a class if necessary.

The “smart” potentiometers are smart between bank changes, but not if you manually enable/disable them. When enabling them, you'll have to add some logic yourself to notify the smart pot of this event, so it knows it has to lock its state. Basically, you need to execute these lines yourself:

https://github.com/tttapa/Control-Surface/blob/39182bbab79487236c11579db3acfa336813bf54/src/MIDI_Outputs/Bankable/Abstract/SmartMIDIFilteredAnalog.hpp#L80-L85

Alternatively, don't enable/disable them, but use a single 4-bank potentiometer instead, and change the bank yourself instead of using a `IncrementSelector` (using `bank.select(i)` with `0 <= i < 4`).

gringow831, 2023-02-11:
The problem with changing banks is I am trying to use the FakeEncoder solution from #885 as a second potentiometer so changing between banks doesn't work as are two different types of pots.
I am trying to find the solution to read the button within the multipurpose button switch statement to say if it's been press during short press do A if it's pressed during longpress change the velocity of NoteOff for that particular moment.

gringow831, 2023-02-11:
What is the right method to read the value of the NoteButton within the switch statment? I tried if (Bankable::ManyAddresses::NoteButton<2>::getButtonState(buttons) == Button::FALLING{ Control_Surface.sendNoteOn(address, 0x00) //when released
}
Also tried 
  for (auto &button : buttons) // for each button in the buttons array
     Control_Surface.sendNoteOn(address, 0x00);

tttapa, 2023-02-11:
I think trying to use the existing classes for things they weren't designed for is over-complicating things.
I'd take a step back, and implement (parts of) your design by manually using the more low-level features of the library (`Button`, `MultiPurposeButton`, `FilteredAnalog`, `AHEncoder`, `EncoderState`, `Control_Surface.sendNoteOn()`, etc.) rather than using MIDI elements, banks, etc.

gringow831, 2023-02-11:
Sorry, I don't understand what you mean with the low-level features for this issue.

gringow831, 2023-02-11:
Also, I guess when you said the lines myself ```
if (value == previousValue || previousValue == initial) 
     state = Active; 
 else if (value > previousValue) 
     state = Higher; 
 else 
     state = Lower; 
```
First I need to declare locally value, Lower, Higher, Active, etc?

tttapa, 2023-02-11:
The code you posted uses classes like `NoteButton`, `Bank`, `IncrementSelector`, `SmartMIDIFilteredAnalog` etc.  
These classes are fairly simple, for instance `NoteButton` simply wraps a `Button`, and sends a MIDI message if the button was pressed or released. You can easily replace this with an if-statement in your own code, giving you full control over what messages it sends and when. Similarly, a Bank just keeps track of the number of the bank that is active. An IncrementSelector is just a counter that counts up when a button is pressed. 

Again, by implementing those things yourself, you can program much more complicated behavior, like the one you were asking for.

> First I need to declare locally value, Lower, Higher, Active, etc?

No, these are member variables of the SmartMIDIFilteredAnalog class. You'd have to add member functions to that class, or to your own `PBSmartPotentiometer` from the other thread.

> I tried finding examples for the MultiPurposeButton but there are not many showing what is possible to do with it

All MultiPurposeButton does is tell you what happens to a push button (e.g. pressed, released, long press, double press, etc.). To use it, you need to implement your own logic around that using switch- and if-statements, e.g. “if button was released, send a MIDI note off message, else if another button was pressed, change the MIDI address I'm sending to, else ...”.

A very simple example of what I mean:

```cpp
#include <Control_Surface.h>

Button bankButton {2};
Button noteButton {3};
FilteredAnalog<7> potentiometer {A0};

MIDIAddress address = MIDI_Notes::C(4);

USBMIDI_Interface midi;

void setup() {
  midi.begin();
  bankButton.begin();
  noteButton.begin();
  potentiometer.setupADC();
  potentiometer.resetToCurrentValue();
}

void loop() {
  if (bankButton.update() == Button::Falling)
    address += 1;
  
  if (noteButton.update() == Button::Falling)
    midi.sendNoteOn(address, 0x7F);
  else if (noteButton.getState() == Button::Rising)
    midi.sendNoteOff(address, 0x7F);
  
  if (potentiometer.update())
    midi.sendControlChange(0x10, potentiometer.getValue());

  midi.update();
}
```
Notice how I'm manually polling the buttons and analog inputs, and then using if-statements to implement the high-level behavior I want, depending on the events I get.

I'm afraid that providing detailed assistance for implementing that custom logic is beyond the scope of this issue tracker, but you might get some help on the Arduino forum.

gringow831, 2023-02-11:
Thank you! That's very useful. I will properly check it in a bit

tttapa, 2023-02-11:
As a side note, the simple example contains a subtle bug: if you press the note button, then press the bank button, and then release the note button, the note off message will be sent for the wrong note. This is why Control Surface uses `address.lock()/unlock()` for bankable elements.

gringow831, 2023-02-11:
I just thought of something. 
basically the example you sent needs to create a Button example {1}
to also be used if I have NoteButton. 

Is it possible to do something like 
if (example.update() == Button::Falling) the next message will be 
    example.midi.sendNoteOn(address, 0x00);

?

gringow831, 2023-02-12:
I meant it as in, if both buttons are within the same pin. How do I make one cancel the other by telling what to do with that pin?

gringow831, 2023-02-12:
I'm trying a new thing from what you said but it isn't working.
I have created 4-bank with the last one no_channel for the potentiometer.

using bank.select(); isn't actually doing anything and I end up with both of the potentiometer working at the same time.

```ruby
#include <Control_Surface.h> // Include the Control Surface library
#include <Arduino_Helpers.h>
#include <AH/Hardware/MultiPurposeButton.hpp>

USBMIDI_Interface midi;

namespace CS { namespace Bankable { namespace ManyAddresses {

template <setting_t NumBanks>
struct PBSmartPotentiometer : SmartMIDIFilteredAnalog<NumBanks, ManyAddresses<NumBanks>, PitchBendSender<10>> {
  PBSmartPotentiometer(const Bank<NumBanks> &bank, pin_t analogPin, const Array<MIDIChannelCable, NumBanks> &addresses)
    : SmartMIDIFilteredAnalog<NumBanks, ManyAddresses<NumBanks>, PitchBendSender<10>> {
    {bank, AH::copyAs<MIDIAddress>(addresses)}, analogPin, {}
  } {}
};

}}
}

constexpr auto no_address = MIDIAddress::invalid();
constexpr auto no_channel = MIDIChannelCable::invalid();

Bank<4> bank(1);
IncrementSelector<4> bankselect = {bank, 4};

Bankable::ManyAddresses::PBSmartPotentiometer<4> potentiometer[] {
  {bank, A0, {CHANNEL_1, CHANNEL_2, CHANNEL_3, no_channel}},
};

Bankable::ManyAddresses::PBSmartPotentiometer<4> potentiometer2[] {
  {bank, A0, {CHANNEL_4, CHANNEL_2, CHANNEL_1, no_channel}},
};

Bankable::ManyAddresses::NoteButton<4> buttons[] {
  {bank, 5, {MCU::SELECT_1, MCU::SELECT_2, MCU::SELECT_3, no_channel}},
  };

MultiPurposeButton btn{5};
Button button1 {5};

void setup() {
    Bankable::ManyAddresses::CCPotentiometer<2>::disable(potentiometer2);
  
RelativeCCSender::setMode(relativeCCmode::MACKIE_CONTROL_RELATIVE);
  Control_Surface.begin();
  btn.setLongPressDelay(1000);
  btn.setMultiPressDelay(400);
  btn.begin();
  button1.begin();
}

void loop() {
 Control_Surface.loop();
  auto longPressTime = btn.getLongPressedTime();

switch (btn.update()) {
case btn.None:  break;
case btn.PressStart: break;
case btn.ShortPressRelease:break;
case btn.LongPress: 
if (button1.getState() == Button::Falling){
  if (bank.getSelection() == 0){ 
  bank.select(3);
}}
   Bankable::ManyAddresses::CCPotentiometer<2>::enable(potentiometer2);

  break;
case btn.LongPressRelease:
if (button1.getState() == Button::Rising){
  if (bank.getSelection() == 0){ 
  bank.select(0);
}}
  Bankable::ManyAddresses::CCPotentiometer<2>::disable(potentiometer2);
 
//STOP buttons NOTEOFF from happening if LongPressReelased

  break;
    
}
}
```

tttapa, 2023-02-12:
> I meant it as in, if both buttons are within the same pin

You shouldn't use two button objects on the same pin. If you want custom behavior, create one `Button` and implement the behavior yourself, without `NoteButton` or any other high-level class.

> How do I make one cancel the other by telling what to do with that pin?

You can't.

> using bank.select(); isn't actually doing anything

Are you sure that these lines actually get executed?

Either way, the code you posted is way too complex for a simple proof of concept. I'd highly recommend starting with something simpler.

gringow831, 2023-02-13:
I'll try to simplify. I managed to find why it wasn't  getting executed and got it working now. I'm sorry for the hassle.

I might have found a bug, It seems using `buttons[0].setVelocity(0x00);` within LongPress manage to eliminate Note On. Unless I shouldn't be using it for that purpose, I manage to change the velocity of the Note but now there is no more Note On, only off. Is this the normal behaviour?

for example
```ruby
#include <Control_Surface.h>
#include <Arduino_Helpers.h> // https://github.com/tttapa/Arduino-Helpers

#include <AH/Hardware/MultiPurposeButton.hpp>

USBMIDI_Interface midi;

Bank<2> bank(4);
SwitchSelector selector {bank, 4};

Bankable::NoteButton buttons[] {
  {{bank, BankType::CHANGE_ADDRESS}, 6, MCU::MUTE_1},
};

MultiPurposeButton btn{5};

void setup() {
  Control_Surface.begin();
  btn.setLongPressDelay(1000);
  btn.setMultiPressDelay(400);
  btn.begin();
}

void loop() {
  Control_Surface.loop();
  auto longPressTime = btn.getLongPressedTime();
  switch (btn.update()) {
    case btn.None: break;
    case btn.PressStart:  break;
    case btn.ShortPressRelease:  break;
    case btn.LongPress:  break;
    case btn.LongPressRelease:
    
    buttons[0].setVelocity(0x00);
   
     break;
 
  }
}
```
From Midi Monitor t, this is what it shows 
02:11:41.280	From Port 1	Note On	1	E-1	127
02:11:41.408	From Port 1	Note Off	1	E-1	127
02:11:42.726	From Port 1	Note On	1	E-1	127
02:11:42.842	From Port 1	Note Off	1	E-1	127 <<After Longpress
02:11:45.513	From Port 1	Note Off	1	E-1	0
02:11:45.600	From Port 1	Note Off	1	E-1	127
02:11:46.645	From Port 1	Note Off	1	E-1	0
02:11:46.729	From Port 1	Note Off	1	E-1	127

while with USBDebugMIDI_Interface still shows
Note On          Channel: 1	Data 1: 0x10	Data 2: 0x7f
Note Off         Channel: 1	Data 1: 0x10	Data 2: 0x7f
Note On          Channel: 1	Data 1: 0x10	Data 2: 0x7f
Note Off         Channel: 1	Data 1: 0x10	Data 2: 0x7f <<After longpress
Note On          Channel: 1	Data 1: 0x10	Data 2: 0x00
Note Off         Channel: 1	Data 1: 0x10	Data 2: 0x7f
Note On          Channel: 1	Data 1: 0x10	Data 2: 0x00
Note Off         Channel: 1	Data 1: 0x10	Data 2: 0x7f

closed 2023-02-23 by gringow831