tttapa/Control-Surface #893: help send midi note on the fader and encoders ?

davidgauze, 2023-02-11:
Hello, can you make an example of how to send midi note on the fader and encoders?

#include <Control_Surface.h>

USBMIDI_Interface usbmidi;

using namespace MIDI_Notes;

Notespotentiometer faders[] {
// {A0, MIDI_Notes::C(4)},
// {A1, {0x01, CHANNEL_1}},

};

CCRotaryEncoder enc {
{ 8, 9 }, {0x10},speedMultiplier, 4,
{ 10, 11}, {0x20},speedMultiplier, 4,

};

void setup() {
Control_Surface.begin(); // Initialize Control Surface
}

void loop() {
Control_Surface.loop(); // Update the Control Surface
}

tttapa, 2023-02-11:
> how to send midi note on the fader and encoders?

You'll first have to define exactly what that means. How should the position value or delta be encoded in note messages?

davidgauze, 2023-02-11:
ola. estou fazendo controladora midi usb , software reconhece eventos  Note on e Note Off, não reconhecer control change ou CC,
porem quero todos fader e encoder mandar note on quando utilizadas. ,

tttapa, 2023-02-11:
Like I said, you'll have to define the data format first. How should the 7-bit or 14-bit value be converted to a note message? How are relative deltas encoded?

davidgauze, 2023-02-12:
tenho uma programação que voce criou para em abril de 2022,  onde solicitei conversão midi control change para note on, 
porem hoje em dia nao estou conseguindo fazer o mesmo funcionar, dando erro de NotePotentiometer pots.

https://github.com/tttapa/Control-Surface/issues/161

#include <Encoder.h> // Include the Encoder library.
// This must be done before the Control Surface library.
#include <Control_Surface.h> // Include the Control Surface library

// Instantiate a MIDI over USB interface.
USBMIDI_Interface midi;

struct RelativeNoteSender {
    static void send(int delta, MIDIAddress address) {
        delta = constrain(delta, -127, 127);
        if (delta > 0) {
          Control_Surface.sendNoteOn(address, delta);
          // Send whatever your software expects
        } else {
          Control_Surface.sendNoteOn(address + 1 /* ??? */, -delta);
          // Send whatever your software expects
        }
    }
};

struct ContinuousNoteSender {
    void send(uint8_t value, MIDIAddress address) {
        Control_Surface.sendNoteOn(address, value);
    }  
    static constexpr uint8_t precision() { return 7; }
};

struct NotePotentiometer : MIDIFilteredAnalogAddressable<ContinuousNoteSender> {
    NotePotentiometer(pin_t analogPin, MIDIAddress address)
      : MIDIFilteredAnalogAddressable(analogPin, address, {}) {}
};

USBMIDI_Interface midi;

using namespace MIDI_Notes;
NotePotentiometer pots[] = {
  {A0, note(C, 4)},
  {A1, note(Db, 4)},
  {A2, note(D, 4)},
   {A3, note(Eb, 4)},
};

struct NoteEncoder : MIDIRotaryEncoder<RelativeNoteSender> {
    NoteEncoder(EncoderPinList pins, MIDIAddress address)
      : MIDIRotaryEncoder(pins, address, 1, 4, {}) {}
};

NoteEncoder enc = {
  {2, 3}, // pins
  0x01,   // MIDI address (note number + optional channel)
};

void setup() {
  Control_Surface.begin(); // Initialize Control Surface
}

void loop() {
  Control_Surface.loop(); // Update the Control Surface
}

tttapa, 2023-02-12:
Please use code fences when posting code, and post the errors as well (as text, not as screenshots of text).

Try this (untested):

```cpp
#include <Control_Surface.h> // Include the Control Surface library

struct ContinuousNoteSender {
    void send(uint8_t value, MIDIAddress address) {
        Control_Surface.sendNoteOn(address, value);
    }  
    static constexpr uint8_t precision() { return 7; }
};

struct NotePotentiometer : MIDIFilteredAnalog<ContinuousNoteSender> {
    NotePotentiometer(pin_t analogPin, MIDIAddress address)
      : MIDIFilteredAnalog(analogPin, address, {}) {}
};

struct RelativeNoteSender {
    static void send(int delta, MIDIAddress address) {
        delta = constrain(delta, -127, 127);
        if (delta > 0) {
          Control_Surface.sendNoteOn(address, delta);
          // Send whatever your software expects
        } else {
          Control_Surface.sendNoteOn(address + 1 /* ??? */, -delta);
          // Send whatever your software expects
        }
    }
};

struct NoteEncoder : MIDIRotaryEncoder<RelativeNoteSender> {
    NoteEncoder(AHEncoder &&enc, MIDIAddress address)
      : MIDIRotaryEncoder<RelativeNoteSender>(std::move(enc), address, 1, 4, {}) {}
};

USBMIDI_Interface midi;

using namespace MIDI_Notes;
NotePotentiometer pots[] = {
    {A0, note(C, 4)},
    {A1, note(Db, 4)},
    {A2, note(D, 4)},
    {A3, note(Eb, 4)},
};

NoteEncoder enc = {
  {2, 3}, // pins
  0x01,   // MIDI address (note number + optional channel)
};

void setup() {
  Control_Surface.begin(); // Initialize Control Surface
}

void loop() {
  Control_Surface.loop(); // Update the Control Surface
}
```