tttapa/Control-Surface #898: Erratic movements and values in Audition

Roots98, 2023-03-01:
Hi Tttapa, you have an amazing library! 
I noticed erratic values with general purpose controller and Adobe Audition. the value moves right or left, sometimes inverted and sometimes starts moving in the same direction with jumps or when you get to the end it jumps to the beginning. 
You can see the problem in the video uploaded here https://youtu.be/ZSiZ8HuoJKQ
From the second 20 is very clear to see 

```cpp
#include <Control_Surface.h>

USBMIDI_Interface midi;

CCPotentiometer GPC {
  18, {MIDI_CC::General_Purpose_Controller_1}
};

void setup() {
RelativeCCSender::setMode(MACKIE_CONTROL_RELATIVE);
Control_Surface.begin();
}

void loop() {
Control_Surface.loop();
}
```

Trickster-git, 2023-03-01:
First try midi monitor and see on values from your pot.

Roots98, 2023-03-01:
@Trickster-git It works fine with midi monitor so don't understand the error. Have you tried it to see if you can mimic the problem?

Roots98, 2023-03-02:
I noticed too that the potentiometer centre is not the analog centre when in Audition but on the midi monitor looks fine. I use B10K potentiometers. Is there a way to fix it?

Trickster-git, 2023-03-02:
Try other daw 
try use example https://tttapa.github.io/Control-Surface-doc/Doxygen/de/de4/Control-Change-Potentiometer_8ino-example.html
You can understand it is hardwire problem or not.
check audition midi settings.

Roots98, 2023-03-02:
I am using a Teensy 4.0 and have tried the Control-Change-Potentiometer_8ino in Reaper, Pro Tools and Audition. All with the same result. Have also tried an Arduino Leonardo with and same results.
Tried with pitch bend too and it's always the same situation although Midi Monitor shows the correct number. 
The potentiometers used should be at the centre when getting to 0 and it seems you have to go to 70% more or less for the 0. Have tried different brands just in case and all being linear always have the same result.
Using the General purpose also gives erratic values like in the video posted and the pots are connected to ground left, 3.3v right and A0, A1 and so on centre

Trickster-git, 2023-03-03:
Maybe its problem in usb driver.

One my story, when i try connecting display i have big latency, with daw cc value and value on display, and when i stop pot display carry on counting value, i think it same behaver as your case. 
I don`t try it on other pc or mac, my dev os is win10, after i clean win by ccleaner and advanced system optimizer, my problem with display latency gone. 

Can you try on another mac or pc. and use Teensy and Leonardo?

And one idea maybe your pot is bad, can you check using multimeter.
You use pot b10k, right?

Roots98, 2023-03-03:
I have tried different computers with the basics of the code. 
My first thought it was the potentiometer so I used different ones. Have 3 different brands all with the same result. 

Just went and measure them with a multimeter, and seems fine 0 to 10 and middle 5 ohm(decimals here and there but normal).
I saw #478 but I don't think it's correct for me. It turns 4 times or I some other errors

Roots98, 2023-03-08:
I realised that while I am using a linear potentiometer scale Auditon uses a DBFS scale then the middle of the DBFS equals around 12400 of the linear scale. 
I used the code found [https://github.com/tttapa/MIDI_controller/issues/109#issuecomment-698823253](https://github.com/tttapa/MIDI_controller/issues/109#issuecomment-698823253) and modified it to the approximate scale but there is a lot of noise when on the lower end to the left.  Is there a better way to convert the linear pots to the DBFS used on many DAW?

Also, I still haven't found why 
```
CCPotentiometer GPC {
  18, {MIDI_CC::General_Purpose_Controller_1}
};
``` 
Doesn't work properly