tttapa/Control-Surface #900: Compile error on ESP32-S2. "esp_gap_ble_api.h: No such file or directory"

Gershy13, 2023-03-02:
**Describe the bug**
Trying to use Control Surface for a university project, but i cannot get anything to compile on my ESP32-S2. Error is "esp_gap_ble_api.h: No such file or directory." I think this might be due to the fact that my ESP32-S2 does not have bluetooth. Is there anyway to disable the bluetooth part of the library?

**To Reproduce**
Open any Control Surface sketch/example and compile for ESP32-S2 (I am using ESP32S2 DevKit).

**Expected behavior**
The sketch to compile??

**Code**
Any control surface example or even just including the library in a blank sketch.

Gershy13, 2023-03-03:
Anyone with any ideas?

TigeyJewellAlibhai, 2023-03-06:
Also having this problem, please let me know if you found a solution. Thanks!

Gershy13, 2023-03-06:
> Also having this problem, please let me know if you found a solution. Thanks!

Unfortunately still have not. I can only assume its because the S2 does not have BLE, and its trying to include a file which does not exist. If there was a way to disable the bluetooth part of Control Surface, it would probably work.

TigeyJewellAlibhai, 2023-03-06:
I got it to at least compile! In control-surface-main/src I took out a couple lines in Control_Surface.h so it no longr includes Bluetooth MIDI Interface. Then I had to delect the BLEMIDI folder in interfaces

TigeyJewellAlibhai, 2023-03-06:
130-132 from Control_Surface.h. I just commented them out actually

Gershy13, 2023-03-06:
> 130-132 from Control_Surface.h. I just commented them out actually

Thanks! Ill give it a try, did you manage to get anything working on the esp32s2? Im trying to implement Adafruit TinyUSB for MIDI-USB support.

TigeyJewellAlibhai, 2023-03-06:
Nothing working yet :(   I got the control surface code to compile and write for the s2 but it isn't showing up on a midi monitor. I may be doing something else wrong though

Gershy13, 2023-03-06:
Are you using USB? If you are I had an issue when i was trying to use the standard MIDI library where USB didn't work. I had to roll back my esp32 boards version to 2.0.6 from 2.0.7. Then it started working. I think the latest adafruit TinyUSB fixes this tho.

TigeyJewellAlibhai, 2023-03-06:
Just realized that the original ESP32 doesn't support native USB, so I'm guessing this library isnt designed to have any esp32 connect using usb, just ble. Not sure if thats how it works

TigeyJewellAlibhai, 2023-03-06:
I am using usb, updated to the newest tinyUSB. Anything else I should have to do to get control surface code working?

Gershy13, 2023-03-06:
> I am using usb, updated to the newest tinyUSB. Anything else I should have to do to get control surface code working?

I haven't given it a try yet, but there is something about using a custom usb stack in the documentation, I think we have to use that and assign the usb implementation to adafruit tinyusb.

Are you getting a midi device showing in windows?

TigeyJewellAlibhai, 2023-03-07:
I am not getting a device showing in windows, or on a midi monitor application. I did a little digging in to the code and there is the option to use adafruit tiny usb in MIDI_Interfaces/USBMIDI in USBMIDI_Adafruit_TinyUSB.hpp and changing a few lines in USBMIDI.hpp *should* force the controller to use tinyUSB. However no luck actually getting a working midi controller from that yet

Gershy13, 2023-03-07:
I'd first make sure the usb driver is working.

So use the Arduino midi library (midi.h) and an example sketch from the tinyusb library (there's a midi one)

Try that first to make sure it's not your device.

Are you using Arduino IDE? Have you enabled USB-CDC.

Gershy13, 2023-03-08:
> Thanks! Didn't realize I needed USB-CDC. Now I'm getting an error seemingly inside tinyusb saying it undefined reference to TU_LOG_PTR. Ill investigate further later. Did you get it working?

I haven't had time to try control surface yet. My MIDI over usb was working fine with TinyUSB and Arduino's MIDI tho. Even with the latest update now.

Gershy13, 2023-03-10:
> Thanks! Didn't realize I needed USB-CDC. Now I'm getting an error seemingly inside tinyusb saying it undefined reference to TU_LOG_PTR. Ill investigate further later. Did you get it working?

Did you manage to get it to work? Mine compiles and i can see a midi interface in windows, but i cant see any midi data from my rotary encoder. (not sure if this is on the encoder side or the usb side not working)

TigeyJewellAlibhai, 2023-03-11:
I finally got the original MIDI example with tinyusb working, but not with the control surface library. Did you use the code changes I suggested for the control surface library? You could try to send a fake "value" that is generated in the code to confirm its not just your encoder

Gershy13, 2023-03-11:
> I finally got the original MIDI example with tinyusb working, but not with the control surface library. Did you use the code changes I suggested for the control surface library? You could try to send a fake "value" that is generated in the code to confirm its not just your encoder

Yeah, i got it to compile and it seems like its accepting the usb interface. Yeah maybe ill try that. I could try the serial monitoring to see if its receiving any midi from my computer.

Gershy13, 2023-09-06:
For anyone wanting to know if i managed to get it to work, It seems to be working perfectly over USB on a ESP32-S2.

To get it to run, i followed the instructions that @TigeyJewellAlibhai said to get it to compile.

1. Delete/comment lines 130-132 in /src/Control_Surface.h
2. Delete the BLEMIDI folder in /src/MIDI_Interfaces
3. Delete the 2 files BluetoothMIDI_Interface.cpp and .hpp in /src/MIDI_Interfaces

It will work now but without USB support. I could not get it to work with the already included Adafruit TinyUSB backend, maybe there is some way to manually switch it on? To add USB support, I created a custom USB backend using the example given. I copied the code from the USBMIDI_Adafruit_TinyUSB.hpp (also in the /src/MIDI_Interfaces folder) file and adapted it to the custom usb backend. Ive linked the sample sketch i used below.

https://github.com/Gershy13/ESP32-S2-Control-Surface-USB-Test/

This was a way i got it to work, im sure there are much cleaner and better ways of doing it, as the adafruit tinyusb backend is already integrated into control surface, its just not switched on for ESP32.

Maybe @tttapa could help with this, it seems as simple as disabling the BLE interface specifically on the ESP32 S2 as it doesnt have bluetooth, and then getting USB to work would be as simple as enabling the Adafruit TinyUSB backend when a ESP32-S2 is selected.

tttapa, 2023-09-06:
Thank for describing the procedure!

I'm a bit busy now, and I don't have an ESP32-S2 to test things, but the problem with BLE on the S2 is definitely something I intend to fix.

Gershy13, 2023-09-06:
> Thank for describing the procedure!
> 
> I'm a bit busy now, and I don't have an ESP32-S2 to test things, but the problem with BLE on the S2 is definitely something I intend to fix.

No worries, there's a way to get it working currently as I posted, so until official support arrives I'm sure anyone who wants to try the project can use this method as it's not too complicated!

I'd be happy to help with testing on my S2 if and when you update the files!

tttapa, 2023-10-15:
Should be fixed in a711863ecd074d667e6f41173e3e9a866901946e.