tttapa/Control-Surface #901: PlatformIO + Pico Compile Issues

runinit, 2023-03-05:
Moved to a Pico recently and was trying to get a simple USB Midi Interface running. Looked through some discussions here and I believe im following the best practices. 

platformio.ini has lib_ignore = MIDIUSB

Testing the Midi Interface Example:

```
#include <Arduino.h>
#include <Control_Surface.h> // Include the Control Surface library

// The MIDI over USB interface to use
USBMIDI_Interface midi;

void setup() {
  midi.begin(); // Initialize the MIDI interface
}

// MIDI note number, channel, and velocity to use
const MIDIAddress address {MIDI_Notes::C(4), CHANNEL_1};
const uint8_t velocity = 0x7F;

void loop() {
  midi.sendNoteOn(address, velocity);
  delay(500);
  midi.sendNoteOff(address, velocity);
  delay(500);

  midi.update(); // Handle or discard MIDI input
}
```

Get some errors here:
```

Linking .pio/build/pico/firmware.elf
/Users/cdilkie/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: .pio/build/pico/libae6/libControl-Surface.a(PluggableUSBMIDI.cpp.o): in function `CS::PluggableUSBMIDI::out_callback()':
PluggableUSBMIDI.cpp:(.text._ZN2CS16PluggableUSBMIDI12out_callbackEv+0x74): undefined reference to `__atomic_fetch_add_4'
/Users/cdilkie/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: .pio/build/pico/libae6/libControl-Surface.a(PluggableUSBMIDI.cpp.o): in function `CS::PluggableUSBMIDI::send_in_callback(unsigned long)':
PluggableUSBMIDI.cpp:(.text._ZN2CS16PluggableUSBMIDI16send_in_callbackEm+0x12): undefined reference to `__atomic_exchange_4'
/Users/cdilkie/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: .pio/build/pico/libae6/libControl-Surface.a(PluggableUSBMIDI.cpp.o): in function `CS::PluggableUSBMIDI::in_callback()':
PluggableUSBMIDI.cpp:(.text._ZN2CS16PluggableUSBMIDI11in_callbackEv+0x20): undefined reference to `__atomic_exchange_4'
/Users/cdilkie/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: PluggableUSBMIDI.cpp:(.text._ZN2CS16PluggableUSBMIDI11in_callbackEv+0x4c): undefined reference to `__atomic_exchange_4'
/Users/cdilkie/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: PluggableUSBMIDI.cpp:(.text._ZN2CS16PluggableUSBMIDI11in_callbackEv+0x66): undefined reference to `__atomic_exchange_4'
/Users/cdilkie/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: PluggableUSBMIDI.cpp:(.text._ZN2CS16PluggableUSBMIDI11in_callbackEv+0x92): undefined reference to `__atomic_exchange_4'
/Users/cdilkie/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: .pio/build/pico/libae6/libControl-Surface.a(PluggableUSBMIDI.cpp.o):PluggableUSBMIDI.cpp:(.text._ZN2CS16PluggableUSBMIDI16timeout_callbackEv+0xe): more undefined references to `__atomic_exchange_4' follow
/Users/cdilkie/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: .pio/build/pico/libae6/libControl-Surface.a(PluggableUSBMIDI.cpp.o): in function `CS::PluggableUSBMIDI::timeout_callback()':
PluggableUSBMIDI.cpp:(.text._ZN2CS16PluggableUSBMIDI16timeout_callbackEv+0x1c): undefined reference to `__atomic_exchange_1'
/Users/cdilkie/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: PluggableUSBMIDI.cpp:(.text._ZN2CS16PluggableUSBMIDI16timeout_callbackEv+0x34): undefined reference to `__atomic_compare_exchange_4'
/Users/cdilkie/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: .pio/build/pico/libae6/libControl-Surface.a(PluggableUSBMIDI.cpp.o): in function `CS::PluggableUSBMIDI::send_now_impl_nonblock(unsigned long)':
PluggableUSBMIDI.cpp:(.text._ZN2CS16PluggableUSBMIDI22send_now_impl_nonblockEm+0x10): undefined reference to `__atomic_exchange_4'
/Users/cdilkie/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: PluggableUSBMIDI.cpp:(.text._ZN2CS16PluggableUSBMIDI22send_now_impl_nonblockEm+0x32): undefined reference to `__atomic_exchange_1'
/Users/cdilkie/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: PluggableUSBMIDI.cpp:(.text._ZN2CS16PluggableUSBMIDI22send_now_impl_nonblockEm+0x52): undefined reference to `__atomic_compare_exchange_4'
/Users/cdilkie/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: PluggableUSBMIDI.cpp:(.text._ZN2CS16PluggableUSBMIDI22send_now_impl_nonblockEm+0x88): undefined reference to `__atomic_exchange_4'
/Users/cdilkie/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: .pio/build/pico/libae6/libControl-Surface.a(PluggableUSBMIDI.cpp.o): in function `CS::PluggableUSBMIDI::read()':
PluggableUSBMIDI.cpp:(.text._ZN2CS16PluggableUSBMIDI4readEv+0x54): undefined reference to `__atomic_fetch_sub_4'
/Users/cdilkie/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: PluggableUSBMIDI.cpp:(.text._ZN2CS16PluggableUSBMIDI4readEv+0x60): undefined reference to `__atomic_exchange_1'
/Users/cdilkie/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: .pio/build/pico/libae6/libControl-Surface.a(PluggableUSBMIDI.cpp.o): in function `CS::PluggableUSBMIDI::write_impl(unsigned long const*, unsigned long, bool)':
PluggableUSBMIDI.cpp:(.text._ZN2CS16PluggableUSBMIDI10write_implEPKmmb+0xbc): undefined reference to `__atomic_compare_exchange_4'
/Users/cdilkie/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: PluggableUSBMIDI.cpp:(.text._ZN2CS16PluggableUSBMIDI10write_implEPKmmb+0x110): undefined reference to `__atomic_exchange_4'
```

The same code works in the Arduino IDE though.

BenChung, 2023-07-04:
I was able to fix this by changing PluggableUSBMIDI to use mstd::atomic instead of std::atomic. This feels like a brittle solution, however. I suspect that the normal Arduino IDE is using a different definition of std::atomic that is tantamount to mstd::atomic, but I don't know how exactly this is set up.

AriESQ, 2025-02-08:
Please close stale issue.