tttapa/Control-Surface #907: Arduino Nano Every not working with MIDI USB

blockiyt, 2023-03-25:
Hey there,
I am trying to get my Nano Every (Atmega4809) running with this libary and MIDI over USB.
As far as I know the Nano Every has native USB Support, and also the [Example](https://github.com/tttapa/Control-Surface/blob/main/examples/3.%20MIDI%20Interfaces/MIDI-Output/MIDI-Output.ino) I am using has in it's description the Nano Every mentioned as working.
Am I doing something wrong, or is it not compatible?

tttapa, 2023-03-25:
The Nano Every does not have native USB support, it has a secondary ATSAMD11 chip that translates between USB and serial, see https://docs.arduino.cc/resources/datasheets/ABX00028-datasheet.pdf#3-3-usb-bridge.

While you _can_ reprogram the SAMD11 with MIDI firmware, the main ATmega4809 that your Arduino sketch is running on does not support (MIDI over) USB. 

MIDI over USB documentation: https://tttapa.github.io/Control-Surface-doc/Doxygen/d8/d4a/md_pages_MIDI-over-USB.html

blockiyt, 2023-03-26:
Oh okay, I see, then I was wrong.
Thank you.

Does anyone know if the Seeed XIAO SAMD21 does work?

[This](https://forum.arduino.cc/t/midi-controller-3-pots-samd21-seeeduino-xiao/1013385/3) forum post says yes.

closed 2023-04-15 by blockiyt