tttapa/Control-Surface #920: Arduino Leonardo freezing on start with MCP23017 rotary encoders examples codes

tainalo2, 2023-04-28:
**Describe the bug**
When starting Leonardo with MCP23017 examples code implement, if SDA linked then arduino freezing.

**To Reproduce**
- Link an MCP23017 to an arduino Leonardo as explain in documentation
- Push official MCP23017-RotaryEncoder-No-Interrupts.ino code without change
- Reset Arduino Leonardo after upload
- Arduino is freezing and not appearing in PC USB device
- Unlink SDA and Reset
- Arduino appearing on PC USB but you can't use the MCP23017

**Expected behavior**
Expecting not freezing and using rotary encoders over MCP23017

**Code**
Link to example test : https://github.com/tttapa/Control-Surface/blob/main/examples/1.%20MIDI%20Output/3.%20Rotary%20Encoders/MCP23017-RotaryEncoder-No-Interrupts/MCP23017-RotaryEncoder-No-Interrupts.ino

**Precision**
I have test my same installation with the Adafruit_MCP23X17.h and a button and it's working, to verify if my MCP23017 is working.

(thank you for your amazing library !)

tttapa, 2023-05-01:
Thanks for the detailed report! I can't seem to reproduce the issue you describe with my setup (Official Arduino Leonardo, Arduino AVR Core 1.8.6, Control surface `main` 39182bbab79487236c11579db3acfa336813bf54).

Could you list the versions of the libraries, cores and tools you're using?

You can try removing the `Wire.setClock()` line, I don't think the ATmega32U4 supports anything above 400 kHz.

tainalo2, 2023-05-02:
Hello,
SOLVED : i  have comment "Wire.setClock()" line
Very thanks !

More precision : 
- Using main branch of this git at 02/05/2023 15H30
- Arduino IDE : 2.1.0
- Not official Leonardo, but a clone from Keyestudio (https://www.keyestudio.com/products/keyestudio-atmega32u4-leonardo-r3-development-board-1m-micro-usb-cable-for-diy-project)

Now i will try to create a prototype of 16 rotary encoders with two MCP23017