tttapa/Control-Surface #921: Board Support: Sparkfun Pro Micro (ATmega32U4 )

ledeniz, 2023-04-29:
I stumbled over this project while trying to implement receiving SysEx with my Sparkfun Pro Micro (clone) and I'm not sure the board is supported by this library (I had trouble compiling an example with PlatformIO), so I open an issue like suggested in the README. I think the board would be a great addition as it is small, cheap and available with USB-C:
- https://www.sparkfun.com/products/12640
- https://www.sparkfun.com/products/15795

Thanks so much for working on Control-Surface, this project is amazing!

Trickster-git, 2023-04-29:
ATmega32U4 work with Control-Surface, try arduino ide.

tttapa, 2023-04-29:
Using the Arduino IDE, and using version 1.1.13 of the “SparkFun AVR Boards” core (see https://learn.sparkfun.com/tutorials/qwiic-pro-micro-usb-c-atmega32u4-hookup-guide#mac_linux_boardaddon), the random selection of Control Surface examples I tried compiled without issues.

What error did you get in PlatformIO?

ledeniz, 2023-04-29:
Ahh you are both so right! Just tried it too, works fine. Sorry I didn;t think about that earlier! PlatformIO seems to use some compiler flags or something that makes it more restrictive than the Arduino IDE, for example you can't call functions before they were defined earlier in the file etc. I need to look into that more, but completely different topic.

Thanks!

closed 2023-04-29 by ledeniz