tttapa/Control-Surface #928: regarding ESP8266 examples

jose-otero-rodriguez, 2023-06-01:
Hi:
I have a question regarding the ESP8266 note on/off examples. I'm using a node mcu v3. I'm able to compile, upload and check serial debug messages in both cases. The softAP sort of works, using AUM in the iPad I can see some blinking signals, but no sound (I think the note off comes too quickly for the selected patches, it is not a sketch bug). In the case of the other example, the ESP is not displayed in AUM, as happens with the example for ESP32 (an ESP-EYE, check screenshot).
Anyway, thanks for you work! I have an arduino BLE sense, I'll check shortly if it works with your library
Regards
[image]

jose-otero-rodriguez, 2023-06-06:
I'm closing this too, same reason, waste of time. I'm going to buy an ESP32 instead. Regards

closed 2023-06-06 by jose-otero-rodriguez

Trickster-git, 2023-06-07:
As alternative you can try RP Pi pico