tttapa/Control-Surface #929: NoteButton example only works with button attached to D4, regardless of parameter

jose-otero-rodriguez, 2023-06-04:
Using a ESP8266 (node mcu) unable to read button state other than the one connected to D4 when changing original parameter (5) to whatever value and rewiring accordingly

To reproduce wire a button to a different pin and change accordingly the parameter, it will not work. Conversely, keep the button connected to D4 and change the parameter, the button will keep working

The expected behavior is the opposite. Change the parameter without rewiring should not work. Rewire and change accordingly the parameter should work

```cpp

#include <Control_Surface.h> // Include the Control Surface library

// Instantiate a MIDI over USB interface.
USBMIDI_Interface midi;

// Instantiate a NoteButton object
NoteButton button {
  5,                       // Push button on pin 5
  {MIDI_Notes::C(4), CHANNEL_1}, // Note C4 on MIDI channel 1
};

void setup() {
  Control_Surface.begin(); // Initialize Control Surface
}

void loop() {
  Control_Surface.loop(); // Update the Control Surface
}
```

tttapa, 2023-06-04:
Which board are you using?  
The pin numbers prefixed with a D are different from the actual GPIO pin numbers.  
See e.g. https://www.wemos.cc/en/latest/d1/d1_mini_pro.html#pin.

The pin numbers you specify in your code are the GPIO pin numbers, see https://arduino-esp8266.readthedocs.io/en/latest/reference.html#digital-io.

jose-otero-rodriguez, 2023-06-06:
> 

Hi, thanks for your answer, I apologize for my answer delay. 
I'm using a node mcu V3 https://cyberblogspot.com/nodemcu-v3-esp8266-pinout-and-configuration/
Your code works when the button is connected to D4 regardless of the parameter (tested 4 and 3). From the pinout, then D4 is GPIO2 and D1 is GPIO5. This makes things even more puzzling, if I connect the button to the later, it doesn't work. Check the picture with the working wiring.
Regards
[image]

jose-otero-rodriguez, 2023-06-06:
I think that ESP8266 board is the less suitable for this task. I have several of these, but I'm thinking about we better close this to save you some time and use the nano instead. 
Regards, thanks for your time

jose-otero-rodriguez, 2023-06-06:
Finally, I think it is a waste of time dealing with this board. Thanks