tttapa/Control-Surface #944: FastLed VU and VPotRing  ( ws2811 )

Jhenesson, 2023-07-26:
Hi @tttapa  ! I want to thank you for the great work, it's all excellent.

After a long time testing and planning my controller, I'm close to success.

But only one function is missing: VPotRing using a ws2811 led ring (to read the MCU position and represent the potentiometer position)

I'm not a programming expert, I've been gathering user codes here and trying to assemble, this was the last code I got.
The VU is working, but I was not successful in VPotRing.

If you can help me, I would be grateful (I imagine that several other users are also interested in this)

```
#include <FastLED.h>
#include <Control_Surface.h>

#define F_PIN1  3
#define F_PIN2  4
#define F_PIN3  5
#define NUM_LEDS_RING  12
#define NUM_LEDS    12
#define LED_TYPE    WS2812
#define COLOR_ORDER GRB
#define BANKS 4 
 
 
USBMIDI_Interface midi;
const int tracks = 8/BANKS;

// Banks conf.
Bank<BANKS> bank(8/BANKS);  
IncrementDecrementSelector<BANKS> bankselector = {bank, {7, 8}, Wrap::Wrap};

// VU meters
unsigned int decayTime = MCU::VUDecay::Default ;
MCU::Bankable::VU<BANKS> vu[] = {
  {bank, 1,decayTime},
  {bank, 2,decayTime},
}; 

// Vpot
MCU::Bankable::VPotRing<BANKS> vpot[] = {
  {bank, 1},
  {bank, 2},
}; 

//---- FastLED config. ---//

DEFINE_GRADIENT_PALETTE(bgyr_gp)         // Gradient Palette for FastLED
{
    0,   0,  0,200,
   80,   0,200,  0,
  176,  80, 80,  0,
  230, 200,  0,  0, 
  255, 220,  0,  0  
};

CRGB leds1[NUM_LEDS];
CRGB leds2[NUM_LEDS];
CRGB ledsRing[NUM_LEDS_RING];
uint8_t gBrightness = 70;
CRGBPalette16 currentPalette;
TBlendType    currentBlending;

void FastLEDsetup() {  
 FastLED.addLeds<LED_TYPE, F_PIN1, COLOR_ORDER>(leds1, NUM_LEDS).setCorrection(TypicalLEDStrip);

  FastLED.addLeds<LED_TYPE, F_PIN2, COLOR_ORDER>(leds2, NUM_LEDS).setCorrection(TypicalLEDStrip);

 FastLED.addLeds<LED_TYPE, F_PIN3, COLOR_ORDER>(ledsRing, NUM_LEDS_RING).setCorrection(TypicalLEDStrip);

  FastLED.setBrightness(gBrightness);
  currentBlending = LINEARBLEND;
  currentPalette = bgyr_gp;
} 

void updateVU(){
  FastLED.clear();
  for (int i = 0; i < tracks; i++)
  {
    if (vu[i].getDirty()) 
    {
        for(int pix = 0; pix < vu[0].getValue(); pix++) 
        { 
          leds1 [pix] = ColorFromPalette( currentPalette, pix*(255/NUM_LEDS), gBrightness, currentBlending);                       
        }
        for(int pix = 0; pix < vu[1].getValue(); pix++) 
        { 
          leds2 [pix] = ColorFromPalette( currentPalette, pix*(255/NUM_LEDS), gBrightness, currentBlending);                                    
        }
      vu[i].clearDirty();      
      FastLED.show();
    }
  }   
}

void updateVPotRing(){
  
  for (uint8_t i = 0; i < tracks; i++)
  {
    if (vpot[i].getDirty()) 
    {      
        for(uint8_t seg = 0; seg < NUM_LEDS_RING ; seg++) 
        { 
          if ( seg >= vpot[i].getStartOn() && seg < vpot[i].getStartOff() )
          {
            ledsRing[i] [seg] = CRGB::Red ; 
          }
          else 
          {
            ledsRing[i] [seg] = CRGB::Black; 
          }
        }
      vpot[i].clearDirty();      
      FastLED.show();
    }
  }   
}

void setup() {  
  FastLEDsetup();
  Control_Surface.begin();
}

 
void loop() {
  Control_Surface.loop();
  
  updateVU(); 
}

```

tttapa, 2023-07-27:
Could it be because the `updateVPotRing` function is never called?

Jhenesson, 2023-07-27:
I really do not know. I'm not a programmer and I have only light notions. The code I got from some user who posted it here.

#335 

Could you help me with this?

tttapa, 2023-07-28:
I'm afraid that general programming support is beyond the scope of this issue tracker.  

I'd recommend writing up a specific question on https://forum.arduino.cc (be sure to read the [guidelines](https://forum.arduino.cc/t/how-to-get-the-best-out-of-this-forum/679966) and include your code - in code tags - and a link to the source where you got it).